openhwfoundation/cvw #17: Is this synthesizable?

zhanghongce, 2023-01-22:
Hi I was wondering if such delay in "assign" statements is actually synthesizable?

Does it matter if the delay is removed? 

https://github.com/openhwgroup/cvw/blob/1fed4b16cc31d3fd3e9d4a9e4e27729ddaecba62/pipelined/src/hazard/hazard.sv#L91

Thanks!

MikeOpenHWGroup, 2023-01-22:
The short answer to your question @zhanghongce is that delayed assignments are synthesizable.  Most synthesis tools will simple ignore the delay.  Of course, this yields to differences in behavior between RTL and gate-level simulations.

In general this coding style is discouraged and there is a clause in the [lowRISC Style Guide](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md#delay-modeling) which prohibts them.

Having said all that, there are often good reasons for violating guidelines, so I will defer to @davidharrishmc on this one, even though `git blame` says only one of these is his.  😉

ross144, 2023-01-22:
Yes, the delay is ignored by synthesis tools.  I've confirmed with this Vivado, DC shell, and genus.  Our flip-flops also have a 1 unit time delay after the clock edge.  In general this is avoid any strange delta time step issues with modelsim.  For example modelsim will add a delta time step if the clock is renamed between hierarchies. This can lead to some modules seeing a late arriving clock causing the design to fail similar to a hold timing violation.  I agree in general it is best to avoid rtl delays, but sometimes they are important.

I am uncertain of the delays delays in the hazard unit however.

davidharrishmc, 2023-01-23:
We have these delays in selected places to help visualize causality during simulation.

David

closed 2023-01-23 by davidharrishmc