openhwfoundation/cvw #47: ecall is reported as retiring, and increments minstret

eroom1966, 2023-02-02:
The ecall instruction does not retire, and it should not increment minstret
from the manual

> As ECALL and EBREAK cause synchronous exceptions, they are not considered to retire, and should not increment the minstret CSR.

The attached (gzipped) elf can reproduce the issue

[ref.elf.gz](https://github.com/openhwgroup/cvw/files/10569808/ref.elf.gz)

ross144, 2023-02-02:
This is a great find and has implications for our textbook.  For reference this is section 3.3.1 of the privileged specification, last paragraph.

eroom1966, 2023-02-02:
Hi @ross144 
This is actually masking a few issues later in the execution at the moment when running some tests - but I think I can work around this in the short term

ross144, 2023-02-02:
I'll have a fix pushed in a few minutes.

closed 2023-02-02 by eroom1966

ross144, 2023-08-25:
Fixed by PR #48