openhwfoundation/cvw #50: Instruction page fault raised incorrectly

eroom1966, 2023-02-02:
Hi @ross144 

I am running more of the riscv-dv tests, it looks like we have an issue with an 'Instruction Page Fault' raised, and we do not think it should be.

I am attaching the elf file as before.

The symptom is with the fetch of instruction at (virtual) address 0x74fc
It appears that the test repeatedly flushes the TLB entries using an sfence.vma, pages get remapped but at instruction 23648 we have the fault occurring in the RTL

I am not sure of the size of the TLB in the RTL but the last sequence we observe is a TLB ENTRY creation for VA 0x7000 - 0x7FFF
we see a number of instructions execute, then instruction at address 0x74fc takes the 'Instruction Page Fault'

below you can see that the RTL takes the fault handler whereas the REF execute the instruction which the RTL reported as the faulting address

```
# [NOTE] testbench.idv_trace2log @ 1156711: RET,0,23647,000074f8,"0000100f fence.i                ",,,,CSRb00(mcycle)=000000000001c3d2 CSRb02(minstret)=0000000000005c5f,
# Info 23647: 'refRoot/cpu', 0x00000000000074f8(sub_3_86_0_t+3a): Supervisor 0000100f fence.i
# Info   MEMX 0x74f8 0x800074f8 2 100f
# Info   MEMX 0x74fa 0x800074fa 2 0000
# [NOTE] testbench.idv_trace2log @ 1157001: RET,0,23648,8000f000,"1df1     c.addi  x27,x27,-4     ",x27=000000008003fcf4,,,CSR300(mstatus)=0000000a00040800 CSR341(mepc)=00000000000074fc CSR342(mcause)=000000000000000c CSR343(mtval)=00000000000074fc CSRb00(mcycle)=000000000001c3ef CSRb02(minstret)=0000000000005c60,
# Info 23648: 'refRoot/cpu', 0x00000000000074fc(sub_3_70_1_t): Supervisor 987d     c.andi  x8,x8,-1
# Info   MEMX 0x74fc 0x800074fc 2 987d
# Error (IDV) PC mismatch (HartId:0, PC:0x00000000000074fc ):
# Info (IDV)  0>
# Info (IDV)   . dut:0x000000008000f000
# Info (IDV)   . ref:0x00000000000074fc
# Error (IDV) Insn. bit pattern mismatch (HartId:0, PC:0x00000000000074fc ):
# Info (IDV)  1>
# Info (IDV)   . dut:1df1     c.addi  x27,x27,-4
# Info (IDV)   . ref:987d     c.andi  x8,x8,-1
```
[ref.elf.gz](https://github.com/openhwgroup/cvw/files/10571714/ref.elf.gz)

Thx
Lee

ross144, 2023-02-02:
@eroom1966 Hmm. I'm getting an error much eariler at instruction 195.  It appears we aren't generating an instruction access fault.
```
# Warning (RISCV_PTWE) CPU 'refRoot/cpu': Page table entry load failed [address=0x1e0 PTEAddress=0x80013000 access=X]
# Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x000001e0 0000     c.illegal: No access permission for fetch (0x1e0)
# Info   mstatus 0000000a00040080 -> 0000000a00040800 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:1 MPRV:0 XS:0(Off) FS:0(Off) MPP:1 VS:0(Off) SPP:0 MPIE:0 UBE:0 SPIE:0 MIE:0 SIE:0]
# Info   mcause 0000000000000000 -> 0000000000000001 [Interrupt:0 Code:1(Instruction access fault)]
# Info   mtval 0000000000000000 -> 00000000000001e0
# Warning (IDV) skipping instruction binary comparison due to fetch access fault
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x00000000000001e0 ):
# Info (IDV)  0> CSR 300 (mstatus)
# Info (IDV)   . dut:0x0000000a00040080 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:1 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0
# Info (IDV)   . ref:0x0000000a00040800 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:1 MPRV:0 XS:0(Off) FS:0(Off) MPP:1 VS:0(Off) SPP:0 MPIE:0 UBE:0 SPIE:0 MIE:0 SIE:0
# Info (IDV)  1> CSR 342 (mcause)
# Info (IDV)   . dut:0x0000000000000000 Interrupt:0 Code:0(Instruction address misaligned)
# Info (IDV)   . ref:0x0000000000000001 Interrupt:0 Code:1(Instruction access fault)
# Info (IDV)  2> CSR 343 (mtval)
# Info (IDV)   . dut:0x0000000000000000 
# Info (IDV)   . ref:0x00000000000001e0 
# [ERROR] testbench.idv_trace2api.state_compare @ 4461: MISMATCH
```

eroom1966, 2023-02-03:
Hi Ross
Are you able to run without imperasdv, to se the behaviour of the rtl ?

ross144, 2023-02-03:
Yes. That was a great idea.  I should be able to debug this quickly.

ross144, 2023-02-03:
I found the bug. I hope to have a fix soon.
The specific issue is the instruction spills across cache lines. (It is a 2 byte instruction, but the IFU treats it as a spill. There is room for some optimization here).  For reason the spill logic stopped stalling the pipeline after fetch of the first cache line.

ross144, 2023-02-03:
The above is wrong.
It does handle the spill correctly.  However for reason the ITLB is marking each instruction fetched after the fence as an instruction page fault.

ross144, 2023-02-03:
@davidharrishmc David I may need your help to understand what's happening.
The address 0x47FC matches two ITLB keys 19 and 21.  the Keys do not match for 19, but instead the PageType = 2 (Gigapage?).
I am wondering if we need some additional logic to prioritize TLB entries with larger but overlapping pages.
I'll need to check the spec on this one.

ross144, 2023-02-03:
I wanted to be sure this wasn't an issue with Wally not flushing the ITLB.  Wally clearly invalidates the TLB and fills the TLB first with the Kilo page in Key 19 and then the giga page in key 21.

Either this is 
1. A software error,
2. Wally's HPTW fetched the wrong PTE,
3. Or software is allowed to have overlapping pages when mixing sizes.

davidharrishmc, 2023-02-03:
I sent a zoom link if you want to sort through this right now.

It shouldn’t be possible to have two PTEs for overlapping virtual pages.  The gigapage would terminate the page table walking, so we’d never get to a PTE for a kilopage so there can’t be a valid kilopage PTE.

If software rewrites the page table, changing from a kilopage to a gigapage, and then doesn’t do a sfence.vma, the TLB could have the stale kilopage as well as the fresh gigapage.  This would be a software bug to use a new page table without fencing first.

There are some complicated interactions going on here, so it may take some work to figure out how the TLB gets these overlapping keys.

David

ross144, 2023-02-03:
I found the root cause. The second page table entry is read incorrectly from memory.  There is an interlock failure between the HPTW and FENCE.I.  If the D-cache is flushed at the same time as an ITLB miss (possibly also DTLB miss, but I think this is impossible by construction) the HPTW begins issuing the walk but the cache gets into the flush state and does not actually return valid data to the HPTW.

ross144, 2023-02-03:
Looks like I've fixed it.  We were just missing the logic to suppress the dcache flush during a hptw.  This is fixed with my latest commit.

ross144, 2023-02-03:
One further comment.  David and I discussed a potential issue with the TLB implementation.  If two pages of different sizes overlapped our implementation will say two TLB entries match.   We temporarily believed our logic to update the TLB key was incorrect.  I tried modifying the VPN key to suppress writing the smaller VPNx when filling a Mega, Giga, Tera page.  For example a Giga page should not have a VPN0 or VPN1. This change would only impact writing the PTE to the TLB.  However the logic to lookup the PTE already contained the necessary check to disregard the VPN1 and VPN0 checks.  The effect was no change in behavior.

eroom1966, 2023-02-03:
> @eroom1966 Hmm. I'm getting an error much eariler at instruction 195. It appears we aren't generating an instruction access fault. Is it possible ImperasDV is updated and I'm using an old version? When I look at the objdump there is a valid instruction at address 0x8000_01e0 and wally fetches and executes this instruction. The first part of the error message warns of no target at this address.

Referring back to this comment - I think this is related to the non-zero DAU bits in a non-leaf PTE
I think I may not have committed the option in the model to disable this check, or I have not updated the published reference model. I have this in my local imperas.ic, I will make sure it is consistent in all the repos today.

> --override cpu/ignore_non_leaf_DAU=1

eroom1966, 2023-02-06:
This is now fixed

closed 2023-02-06 by eroom1966