openhwfoundation/cvw #55: fsgnjx.s producing Double Precision NaN value

eroom1966, 2023-02-03:
We have a scenario captured below that a Single Precision FP operation is producing a Double precision result
The result should be a NaN, but it should be a single precision NaN

```
# fsgnjx.s <NAN ?> producing a DP NaN
#
# Info   f4 = 7ff0000000000001
# Info   f9 = 7ff8000000000000
#
# [NOTE] testbench.idv_trace2log @ 31731: RET,0,944,00008a36,"2044a653 fsgnjx.s f12,f9,f4     ",,f12=fff8000000000000,,CSRb00(mcycle)=0000000000000c60 CSRb02(minstret)=00000000000003b0,
# Info 944: 'refRoot/cpu', 0x0000000000008a36(sub_3_stack_p+7a8): User 2044a653 fsgnjx.s f12,f9,f4
# Info   MEMX 0x8a36 0x80008a36 2 a653
# Info   MEMX 0x8a38 0x80008a38 2 2044
# Info   f12 ffffffffbb372670 -> ffffffff7fc00000
# Error (IDV) FPR register value mismatch (HartId:0, PC:0x0000000000008a36 ):
# Info (IDV)  0> FPR f12
# Info (IDV)   . dut:0xfff8000000000000
# Info (IDV)   . ref:0xffffffff7fc00000

```

davidharrishmc, 2023-02-18:
Fixed with PR #93 
https://github.com/openhwgroup/cvw/pull/93
Lee, please confirm fix.

eroom1966, 2023-02-21:
Fixed

closed 2023-02-21 by eroom1966