openhwfoundation/cvw #56: fsgnjx.s producing Double Precision value instead of Single Precision NaN

eroom1966, 2023-02-03:
In the example below the register f4 is not NaN boxed to represent a single precision value
It looks like the RTL is injecting the sign bit, and producing a Double Precision result

The inputs should be treated as Single Precision, and the result should be Single Precision NaN

```
# fsgnjx.s <VAL ?>
# Info   f4  = cfa695b1047553b1 invalid SP value 047553b1 NaN Box fffffff........
# Info   f14 = ffffffff7fc00000
#
# [NOTE] testbench.idv_trace2log @ 4951: RET,0,297,800003bc,"20e229d3 fsgnjx.s f19,f4,f14    ",,f19=cfa695b1047553b1,,CSRb00(mcycle)=00000000000001ea CSRb02(minstret)=0000000000000129,
# Info 297: 'refRoot/cpu', 0x00000000800003bc(main+26): Machine 20e229d3 fsgnjx.s f19,f4,f14
# Info   MEMX 0x800003bc 0x800003bc 2 29d3
# Info   MEMX 0x800003be 0x800003be 2 20e2
# Info   f19 ffffffff80000000 -> ffffffff7fc00000
# Error (IDV) FPR register value mismatch (HartId:0, PC:0x00000000800003bc main+38):
# Info (IDV)  0> FPR f19
# Info (IDV)   . dut:0xcfa695b1047553b1
# Info (IDV)   . ref:0xffffffff7fc00000
```

davidharrishmc, 2023-02-18:
Fixed with PR #93 
Lee, please confirm fix.

eroom1966, 2023-02-21:
fixed

closed 2023-02-21 by eroom1966