openhwfoundation/cvw #57: fsgnjn.s using sign interpreted from Double Precision on input to assign Single Precision on result

eroom1966, 2023-02-03:
In this example, it looks like the sign difference produced is using the Double Precision format
not the Single Precision format, so inverting the SP sign bit of the result based upon the DP sign bit of the input

```
# fsgnjn.s <VAL ?> Sign difference using double
# Info   f0  = fffffffffb3754ef
# Info   f15 = 7fefffffffffffff
#
# [NOTE] testbench.idv_trace2log @ 12551: RET,0,562,0000064e,"20f01f53 fsgnjn.s f30,f0,f15    ",,f30=ffffffff7b3754ef,,CSRb00(mcycle)=00000000000004e2 CSRb02(minstret)=0000000000000232,
# Info 562: 'refRoot/cpu', 0x000000000000064e(main_17+94): Supervisor 20f01f53 fsgnjn.s f30,f0,f15
# Info   MEMX 0x64e 0x8000064e 2 1f53
# Info   MEMX 0x650 0x80000650 2 20f0
# Info   f30 b2b7b12cf6c6f830 -> fffffffffb3754ef
# Error (IDV) FPR register value mismatch (HartId:0, PC:0x000000000000064e ):
# Info (IDV)  0> FPR f30
# Info (IDV)   . dut:0xffffffff7b3754ef
# Info (IDV)   . ref:0xfffffffffb3754ef
```

davidharrishmc, 2023-02-18:
Fixed with PR #93 
Lee, please confirm fix.

eroom1966, 2023-02-21:
fixed

closed 2023-02-21 by eroom1966