openhwfoundation/cvw #58: fclass.s incorrectly setting bit 5 of result (rs1 is a positive subnormal number)

eroom1966, 2023-02-03:
The RTL interprets f14 as being 'positive subnormal number'
I think again, this is because it has interpreted this as a Double Precision number rather than Single Precision, in which case it should be a Single Precision (quiet) NaN

```
# fclass
# Info   x2 0000000000000000 -> 0000000000000009
# Info   f14 0000000000000000 -> 7ff0000000000001 : ffffffff........
#
# Info 473: 'refRoot/cpu', 0x0000000000000538(main+2): Supervisor e0071153 fclass.s x2,f14
# Info   MEMX 0x538 0x80000538 2 1153
# Info   MEMX 0x53a 0x8000053a 2 e007
# Info   x2 0000000000000009 -> 0000000000000200
# Error (IDV) GPR register value mismatch (HartId:0, PC:0x0000000000000538 ):
# Info (IDV)  0> GPR x2
# Info (IDV)   . dut:0x0000000000000220
# Info (IDV)   . ref:0x0000000000000200
```

davidharrishmc, 2023-02-18:
Fixed with PR #93 
Lee, please confirm fix.

eroom1966, 2023-02-21:
fixed

closed 2023-02-21 by eroom1966