openhwfoundation/cvw #59: illegal instruction seems to execute on the RTL

eroom1966, 2023-02-03:
I am not quite sure what is happening here, but the instruction bit pattern which should be illegal '0x407b273b'
seems to execute and retire in the RTL producing a value written to x14.
For the REF model, this takes an illegal instruction exception

```
# [NOTE] testbench.idv_trace2log @ 32151: RET,0,631,00000828,"407b273b undef                  ",x14=0000000000000001,,,CSRb00(mcycle)=0000000000000c8a CSRb02(minstret)=0000000000000277,
# Info 631: 'refRoot/cpu', 0x0000000000000828(main_20_0_t+49e): User 407b273b undef
# Info   MEMX 0x828 0x80000828 2 273b
# Info   MEMX 0x82a 0x8000082a 2 407b
# Warning (RISCV_UDEC) CPU 'refRoot/cpu' 0x00000828 407b273b undef: Undecoded instruction
#
```

davidharrishmc, 2023-02-03:
Lee, I added this instruction to the beginning of the add-01 tests, and it raised an illegal instruction fault.  So the underlying failure mode must be more complicated.  Can you share the elf that triggers the issue?

Thanks!

eroom1966, 2023-02-03:
Hi @davidharrishmc 

OK, I do see that in the full testcase, this instruction is executing in USER mode, that may be significant ?

I attach the elf file, the instruction is at count 631
[ref.elf.gz](https://github.com/openhwgroup/cvw/files/10580026/ref.elf.gz)

```
# [NOTE] testbench.idv_trace2log @ 32151: RET,0,631,00000828,"407b273b undef                  ",x14=0000000000000001,,,CSRb00(mcycle)=0000000000000c8a CSRb02(minstret)=0000000000000277,
# Info 631: 'refRoot/cpu', 0x0000000000000828(main_20_0_t+49e): User 407b273b undef
```

Please let me know if I can help further

davidharrishmc, 2023-02-03:
This instruction has op = 0111011, which puts it in the RW category of 64-bit ALU instructions operating on words.  The funct3 field is 010 and the funct7 field is 0100000, which is not a defined RW instruction.  However, the ALU will look at these fields and do a set less than operation on 32-bit inputs and produce some result.

We have a lazy decoder that only issues Instruction faults on illegal opcodes.  This op is illegal in RV32, which is why it gave me a fault in my first test, but is legal in RV64.  

Detecting all the illegal combinations of funct3 and funct7 for a legal op is a moderate amount of extra hardware.  We have omitted this hardware intentionally so far.  

I haven't found a section in the spec yet that requires that all unimplemented instructions cause Illegal Instruction faults.  Ross/Lee, any thoughts about how important it is to devote hardware to trap on every illegal instruction?

ross144, 2023-02-03:
The RISC-V profiles https://github.com/riscv/riscv-profiles/blob/main/profiles.adoc strongly suggest all implemented instructions generate an illegal instruction. However I can't find anything in the spec which says this is mandatory.  We don't conform to these profiles, but yesterday we were considering the implementation cost.  I think I'm on board with modifying the hardware.

"Implementations are strongly recommended to raise illegal-instruction exceptions on attempts to execute unimplemented opcodes."

davidharrishmc, 2023-02-03:
This raises another question about "opcodes."  According to the unprivileged spec, the opcode is bits 6:0 of the instruction.  We already trap on unimplemented opcodes, but not on illegal funct3/funct7 fields within an implemented opcode.  

Detecting illegal functions3 and funct7 combinations may have a noticeable cost for rv32e.

ross144, 2023-02-03:
Maybe we could make it a configuration option?

eroom1966, 2023-02-03:
@davidharrishmc 
`We have a lazy decoder that only issues Instruction faults on illegal opcodes. This op is illegal in RV32, which is why it gave me a fault in my first test, but is legal in RV64.`

I am not sure I understand - are you saying that this bit pattern maps to a legal instruction ?
if so can you tell me what instruction that should be - I need to investigate.

davidharrishmc, 2023-02-03:
The op field (bits 6:0]) is legal.  It’s an RV64 W-type ALU instruction.  The funct3 type of instruction is not legal.

davidharrishmc, 2023-03-19:
Added more decoding to controller.sv to throw more precise illegal instruction exceptions in configurations with ZICSR_SUPPORTED.  This elf agrees with ImperasDV now.  I believe this issue can be closed.

closed 2023-03-27 by davidharrishmc