openhwfoundation/cvw #66: Instruction Page Fault exception - raised incorrectly

eroom1966, 2023-02-06:
I am not sure if this issue is related to the issue I closed earlier https://github.com/openhwgroup/cvw/issues/50
But it looks kind of similar.

Here are the files to reproduce
[ref.tar.gz](https://github.com/openhwgroup/cvw/files/10664077/ref.tar.gz)

The symptoms appear the same, we are seeing a half word aligned, 32bit instruction (lhu) straddling a page boundary raising an Instruction Page Fault
The RTL fails to map the new page from the Walker and takes an Instruction Page Fault exception

```
# [NOTE] testbench.idv_trace2log @ 497641: RET,0,10489,0000affa,"fcaf5183 lhu     x3,-54(x30)    ",x3=000000000000a6a7,,,CSRb00(mcycle)=000000000000c25f CSRb02(minstret)=00000000000028f9,
# Info 10489: 'refRoot/cpu', 0x000000000000affa(sub_2_j4+3620): Supervisor fcaf5183 lhu     x3,-54(x30)
# Info   MEMX 0xaffa 0x8000affa 2 5183
# Info   MEMX 0xaffc 0x8000affc 2 fcaf
# Info   MEMR 0x244ec 0x800244ec 2 a6a7
# Info (RISCV_TLB) CPU 'refRoot/cpu': CREATE TLB ENTRY:
# Info (RISCV_TLB) CPU 'refRoot/cpu': VA 0x0000b000:0x0000bfff PA 0x8000b000:0x8000bfff rwx U=0 G=0 A=1 D=1 ASID=0
# Info   x3 000000000000a4a5 -> 000000000000a6a7
# [NOTE] testbench.idv_trace2log @ 497891: RET,0,10490,80019000,"0400006f j       80019040       ",,,,CSR300(mstatus)=0000000a00180800 CSR341(mepc)=000000000000affe CSR342(mcause)=000000000000000c CSR343(mtval)=000000000000affe CSRb00(mcycle)=000000000000c278 CSRb02(minstret)=00000000000028fa,
# Info 10490: 'refRoot/cpu', 0x000000000000affe(sub_2_j4+3624): Supervisor 79a65203 lhu     x4,1946(x12)
# Info   MEMRM 0x8001d000 0x8001d000 8 00000000200078c1 L2 (64-bit Supervisor PMA unified)
# Info   MEMRM 0x8001e000 0x8001e000 8 00000000200080c1 L1 (64-bit Supervisor PMA unified)
# Info   MEMRM 0x80020058 0x80020058 8 0000000020002ccf L0 (64-bit Supervisor PMA unified)
# Info   MEMX 0xaffe 0x8000affe 2 5203
# Info   MEMX 0xb000 0x8000b000 2 79a6
# Info   MEMR 0x24948 0x80024948 2 0203
# Info   x4 0000000000003636 -> 0000000000000203
# Error (IDV) PC mismatch (HartId:0, PC:0x000000000000affe ):
# Info (IDV)  0>
# Info (IDV)   . dut:0x0000000080019000
# Info (IDV)   . ref:0x000000000000affe
# Error (IDV) Insn. bit pattern mismatch (HartId:0, PC:0x000000000000affe ):
# Info (IDV)  1>
# Info (IDV)   . dut:0400006f j       80019040
# Info (IDV)   . ref:79a65203 lhu     x4,1946(x12)
# Error (IDV) GPR register value mismatch (HartId:0, PC:0x000000000000affe ):
# Info (IDV)  2> GPR x4
# Info (IDV)   . dut:0x0000000000003636
# Info (IDV)   . ref:0x0000000000000203
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x000000000000affe ):
# Info (IDV)  3> CSR 300 (mstatus)
# Info (IDV)   . dut:0x0000000a00180800 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:1 MXR:1 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:1 VS:0(Off) SPP:0 MPIE:0 UBE:0 SPIE:0 MIE:0 SIE:0
# Info (IDV)   . ref:0x0000000a00180080 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:1 MXR:1 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0
# Info (IDV)  4> CSR 341 (mepc)
# Info (IDV)   . dut:0x000000000000affe
# Info (IDV)   . ref:0x00000000000001de
# Info (IDV)  5> CSR 342 (mcause)
# Info (IDV)   . dut:0x000000000000000c Interrupt:0 Code:12(Instruction page fault)
# Info (IDV)   . ref:0x0000000000000000 Interrupt:0 Code:0(Instruction address misaligned)
# Info (IDV)  6> CSR 343 (mtval)
# Info (IDV)   . dut:0x000000000000affe
# Info (IDV)   . ref:0x0000000000000000
```

ross144, 2023-02-06:
This is fixed with commit hash 54a128491eed219808740ad4a232f5a6c3a4e53b, pr#69
If a load missed at the same time as a spilled instruction fetch with an ITLB miss in the second cache line, the HPTW did not wait for the load miss to finish.  This caused the to use the ReadData from the load's address rather than the one generated by the HPTW.  The solution is to delay the HPTW until after the load is completed.

eroom1966, 2023-02-07:
fixed

closed 2023-02-07 by eroom1966