openhwfoundation/cvw #70: c.fld instruction appears to execute, but mstatus.FS=0 (off)

eroom1966, 2023-02-07:
I guess this is a question first.
What is the power up state of mstatus.FS, I am assuming 0 ?

I have a test which appears to execute an instruction "c.fld   f9,160(x8)" on the RTL, but the mstatus.FS bit is 0, meaning that floating point is currently disabled.

The following test will show that this instruction executes on the RTL rather than taking an illegal instruction exception
[ref.tar.gz](https://github.com/openhwgroup/cvw/files/10675405/ref.tar.gz)

```
# [NOTE] testbench.idv_trace2log @ 54001: RET,0,1192,000035aa,"3044     c.fld   f9,160(x8)     ",,,,CSRb00(mcycle)=0000000000001513 CSRb02(minstret)=00000000000004a8,
# Info 1192: 'refRoot/cpu', 0x00000000000035aa(sub_2_124_0_t+1c): User 3044     c.fld   f9,160(x8)
# Info   MEMX 0x35aa 0x800035aa 2 3044
# Warning (RISCV_ILL) CPU 'refRoot/cpu' 0x000035aa 3044     c.fld   f9,160(x8): Illegal instruction - extension D (double-precision floating point) is disabled
# Info   mstatus 0000000a00040080 -> 0000000a00040000 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:1 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:0 MPIE:0 UBE:0 SPIE:0 MIE:0 SIE:0]
# Info   mepc 00000000000001c8 -> 00000000000035aa
# Info   mcause 0000000000000000 -> 0000000000000002 [Interrupt:0 Code:2(Illegal instruction)]
# Info   mtval 0000000000000000 -> 0000000000003044
```

davidharrishmc, 2023-02-07:
Powerup status of mstatus.FS is 0.

Most float instructions don’t work on startup, but fld may be checked as a load rather than a float right now.  I can make this easy fix.

David

eroom1966, 2023-02-07:
@davidharrishmc I think you are right, I have also spotted a c.fldsp instruction executing when floating point is disabled

eroom1966, 2023-02-07:
> It turns out FS was enabled at reset. I’ve disabled it, which should fix this issue. David

Great, I will await the PR/Merge