openhwfoundation/cvw #71: Write to MSTATUS.FS not reported in SSTATUS.FS

eroom1966, 2023-02-07:
This could be a simple tracer issue, rather than an RTL bug,
In the following example the MSTATUS.FS bits are set to 'b01 from a csr write instruction
This gets reported in MSTATUS, but not SSTATUS

[ref.tar.gz](https://github.com/openhwgroup/cvw/files/10675612/ref.tar.gz)

```
# Info 29: 'refRoot/cpu', 0x0000000080000060(init_machine_mode+c): Machine 300a1073 csrw    mstatus,x20
# Info   MEMX 0x80000060 0x80000060 2 1073
# Info   MEMX 0x80000062 0x80000062 2 300a
# Info   sstatus 0000000200000000 -> 0000000200002000 [SD:0 UXL:2 MXR:0 SUM:0 XS:0(Off) FS:1(Initial) VS:0(Off) SPP:0 UBE:0 SPIE:0 SIE:0]
# Info   mstatus 0000000a00000000 -> 0000000a00003800 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:1(Initial) MPP:3 VS:0(Off) SPP:0 MPIE:0 UBE:0 SPIE:0 MIE:0 SIE:0]
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x0000000080000060 init_machine_mode+12):
# Info (IDV)  0> CSR 100 (sstatus)
# Info (IDV)   . dut:0x0000000200000000
# Info (IDV)   . ref:0x0000000200002000 SD:0 UXL:2 MXR:0 SUM:0 XS:0(Off) FS:1(Initial) VS:0(Off) SPP:0 UBE:0 SPIE:0 SIE:0

```

davidharrishmc, 2023-02-07:
This does look like a tracer issue.  The RTL appears to be correct.

eroom1966, 2023-02-23:
Looks like this is now fixed

closed 2023-02-23 by eroom1966