openhwfoundation/cvw #106: c.fld raises load access fault instead of Illegal Instructions

eroom1966, 2023-02-21:
Hi All,

I have what I think is the following scenario.
The Floating point is disabled mstatus.FS=0 (Off)
The core attempts to execute 
# Info 305: 'refRoot/cpu', 0x0000000000008fb2(sub_2+38): Supervisor 2aa0     c.fld   f8,80(x13)

The REF model raises an illegal instruction exception, due to the mstatus.FS=0
The RTL riases a Load Access Fault, I think this is because it does not register the instruction as illegal, and attempts to perform a Load from an unmapped address, which then causes a jump to the mtvec_handler.

I think we have discussed something similar previously regarding the power up state of mstatus.FS, but as far as I can tell, the RTL also thinks that the floating point is off, in which case I would have thought that the c.fld should produce an illegal instruction exception

eroom1966, 2023-02-21:
Here is the trace from the compare
```
# [NOTE] testbench.idv_trace2log @ 7581: RET,0,305,80013000,"0400006f j       80013040       ",,,,CSR300(mstatus)=0000000a001c0800 CSR341(mepc)=0000000000008fb2 CSR342(mcause)=0000000000000005 CSR343(mtval)=0000000080000050 CSRb00(mcycle)=00000000000002f1 CSRb02(minstret)=0000000000000131,
# Info 305: 'refRoot/cpu', 0x0000000000008fb2(sub_2+38): Supervisor 2aa0     c.fld   f8,80(x13)
# Info   MEMX 0x8fb2 0x80008fb2 2 2aa0
# Warning (RISCV_ILL) CPU 'refRoot/cpu' 0x00008fb2 2aa0     c.fld   f8,80(x13): Illegal instruction - extension D (double-precision floating point) is disabled
# Info   mstatus 0000000a001c0080 -> 0000000a001c0800 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:1 MXR:1 SUM:1 MPRV:0 XS:0(Off) FS:0(Off) MPP:1 VS:0(Off) SPP:0 MPIE:0 UBE:0 SPIE:0 MIE:0 SIE:0]
# Info   mepc 00000000000001e0 -> 0000000000008fb2
# Info   mcause 0000000000000000 -> 0000000000000002 [Interrupt:0 Code:2(Illegal instruction)]
# Info   mtval 0000000000000000 -> 0000000000002aa0
# Info 306: 'refRoot/cpu', 0x0000000080013000(mtvec_handler): Machine 0400006f j       80013040
# Info   MEMX 0x80013000 0x80013000 2 006f
# Info   MEMX 0x80013002 0x80013002 2 0400
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x0000000080013000 mtvec_handler+0):
# Info (IDV)  0> CSR 342 (mcause)
# Info (IDV)   . dut:0x0000000000000005 Interrupt:0 Code:5(Load access fault)
# Info (IDV)   . ref:0x0000000000000002 Interrupt:0 Code:2(Illegal instruction)
# Info (IDV)  1> CSR 343 (mtval)
# Info (IDV)   . dut:0x0000000080000050
# Info (IDV)   . ref:0x0000000000002aa0
```

As can be seen the Floating point is disabled mstatus.FS=0, the instruction at address mepc=0x8fb2 is
` # Info 305: 'refRoot/cpu', 0x0000000000008fb2(sub_2+38): Supervisor 2aa0     c.fld   f8,80(x13)`
The RTL is reporting this as 'Load access fault' whereas this should be an illegal instruction

[ref.tar.gz](https://github.com/openhwgroup/cvw/files/10791561/ref.tar.gz)

eroom1966, 2023-02-23:
@davidharrishmc @ross144 
Hi David, Ross,
I pulled in the latest commits, I am now at this git hash 
```
commit 24ab8d3ea57ecd9a99ee1ac2ce22db63bcc5d9b6 (HEAD -> main, origin/main, origin/HEAD)
Merge: 1f5326a0 fd5b9408
Author: David Harris <74973295+davidharrishmc@users.noreply.github.com>
Date:   Wed Feb 22 10:19:49 2023 -0800
    Merge pull request #109 from ross144/main
    Swapped string to integer parameter.
```

Should this contain the above fix ?
I still see the same failure when running the test
We now have the environment working for @ross144 - could you try this ?

If you download/unpack  the attachement - you should be able to run like this 
ImperasDV-OpenHW/scripts/cvw/run-elf.bash --elf <path-to-dir>/ref/ref.elf

Then take a look at the logfile cvw/sim/vsim.log

davidharrishmc, 2023-03-22:
Wally is now taking the illegal instruction fault rather than the load access fault.  The error now is that mtval contains 0506b407 rather than 2aa0.  2aa0 is the compressed instruction and 0506b407 is the uncompressed equivalent.