openhwfoundation/cvw #120: CSR register sie is not observing the mideleg register

eroom1966, 2023-03-01:
If I run the Wally test WALLY-mtvec-01.S sie is updated with the same (masked contents) as mie, however this should also observe the value of the corresponding bits in mideleg

> Section 3.1.9 of Privileged Specification:
> Restricted views of the mip and mie registers appear as the sip and sie registers for supervisor
> level. If an interrupt is delegated to S-mode by setting a bit in the mideleg register, it becomes
> visible in the sip register and is maskable using the sie register. **Otherwise, the corresponding
> bits in sip and sie are read-only zero**

So if the mideleg bits are 0 (which they are in this case) then the corresponding sie bits should also be 0

elf file attached

[ref.elf.gz](https://github.com/openhwgroup/cvw/files/10863144/ref.elf.gz)

eroom1966, 2023-03-01:
I was going to say, this will require my pull request - but it looks like it already merged :-)

eroom1966, 2023-03-14:
Just to follow up on this Issue
It appears that the RTL allows sip to be written, even though it has not been delegated. This is wrong
What I see is a write to sip[1] (SSI) in Machine mode, resulting in the corresponding mip[1] changing to 1
This should not happen as the mideleg is 0, and hence the sip is read-only zero

davidharrishmc, 2023-03-14:
I’m out backpacking this week. Can you put this info in an issue so I don’t
lose track of it?  Should be an easy fix.

davidharrishmc, 2023-03-18:
I've made the fix and it passes the ref.elf, but it's broken our regression testing. Kip and I will fix the test case before committing.

davidharrishmc, 2023-03-22:
I think this one can be closed.
I've fixed it.  I've had to disable some of our tests until they get fixed.
@eroom1966 are you keeping a set of test cases to confirm errors have been fixed?  If so, it's great to include this one too.  If not, I don't think this one needs special treatment.

closed 2023-03-27 by davidharrishmc