openhwfoundation/cvw #132: PMP not configured before attempting to return to U Mode

eroom1966, 2023-03-08:
@davidharrishmc 

I have now configured the settings forthe PMP, and I am running the following test
> tests/riscof/riscof_work/rv64i_m/privilege/src/WALLY-csr-permission-u-01.S/ref/ref.elf

With the PMP enabled, this test attempts to run in U mode, but none of the PMP registers have been written, so the test immediately takes an exception in the REFERENCE model. The RTL executes the instruction in U Mode.
The RTL is not taking the Instruction access fault exception

I was not seeing this issue previously, as the PMP was not enabled (ie there were no registers present)

this is the failing mismatch I see
```
# [NOTE] testbench.idv_trace2log.process_event @ 6220: RET,0,296,800007c8,"00001f37 lui     x30,0x1        ",x30=0000000000001000,,,CSRb00(mcycle)=0000000000000269 CSRb02(minstret)=0000000000000128,
# Info 297: 'refRoot/cpu', 0x00000000800007c8(trap_handler_end_m+10): User *** FETCH EXCEPTION ***
# Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x800007c8 00001f37 lui     x30,0x1: No access permission for fetch (0x800007c8)
# Info   mstatus 0000000a00000080 -> 0000000a00000000 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:0 MPIE:0 UBE:0 SPIE:0 MIE:0 SIE:0]
# Info   mcause 000000000000000b -> 0000000000000001 [Interrupt:0 Code:1(Instruction access fault)]
# Info   mtval 0000000000000000 -> 00000000800007c8
# Info 298: 'refRoot/cpu', 0x0000000080000400(trap_handler_m): Machine 0300006f j       80000430
# Info   MEMX 0x80000400 0x80000400 2 006f
# Info   MEMX 0x80000402 0x80000402 2 0300
# Info (IDV) Instruction executed prior to mismatch '0x8000051c(trapreturn_finished_m+20): 30200073 mret'
# Error (IDV) PC mismatch (HartId:0, PC:0x0000000080000400 trap_handler_m+0):
# Info (IDV)  0>
# Info (IDV)   . dut:0x00000000800007c8 trap_handler_end_m+16
# Info (IDV)   . ref:0x0000000080000400 trap_handler_m+0
# Error (IDV) Insn. bit pattern mismatch (HartId:0, PC:0x0000000080000400 trap_handler_m+0):
# Info (IDV)  1>
# Info (IDV)   . dut:00001f37 lui     x30,0x1
# Info (IDV)   . ref:0300006f j       80000430
# Error (IDV) GPR register value mismatch (HartId:0, PC:0x0000000080000400 trap_handler_m+0):
# Info (IDV)  2> GPR x30
# Info (IDV)   . dut:0x0000000000001000
# Info (IDV)   . ref:0x00000000ffffffff
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x0000000080000400 trap_handler_m+0):
# Info (IDV)  3> CSR 300 (mstatus)
# Info (IDV)   . dut:0x0000000a00000080 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0
# Info (IDV)   . ref:0x0000000a00000000 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:0 MPIE:0 UBE:0 SPIE:0 MIE:0 SIE:0
# Info (IDV)  4> CSR 342 (mcause)
# Info (IDV)   . dut:0x000000000000000b Interrupt:0 Code:11(Environment call from M-mode)
# Info (IDV)   . ref:0x0000000000000001 Interrupt:0 Code:1(Instruction access fault)
# Info (IDV)  5> CSR 343 (mtval)
# Info (IDV)   . dut:0x0000000000000000
# Info (IDV)   . ref:0x00000000800007c8
```

davidharrishmc, 2023-03-18:
This one is a bit puzzling.  My ref.elf is in a slightly different location, but passes.

The Verilog doesn't enforce PMP checking when none of the PMP entries are active.  However, the spec says that if at least one PMP entry is implemented and none of the entries are active, all S and U-mode accesses should fail.  This agrees with your explanation that this should fail, and I'm not sure why it's passing.  I'll need to change all the privileged tests to initialize a PMP region before going to S/U mode.

harris@vlsi:~/cvw$ run-elf.bash --elf  tests/riscof/work/wally-riscv-arch-test/rv64i_m/privilege/src/WALLY-csr-permission-u-01.S/ref/ref.elf
running /home/harris/cvw/tests/riscof/work/wally-riscv-arch-test/rv64i_m/privilege/src/WALLY-csr-permission-u-01.S/ref/ref.elf more frees than allocs from controlfile.general
Test Passed

davidharrishmc, 2023-03-18:
Also passed in riscof_work when I rebuilt those tests.  I do get a funny message about more frees than allocs.

harris@vlsi:~/cvw$ run-elf.bash --elf  tests/riscof/riscof_work/rv64i_m/privilege/src/WALLY-csr-permission-u-01.S/ref/ref.elf
running /home/harris/cvw/tests/riscof/riscof_work/rv64i_m/privilege/src/WALLY-csr-permission-u-01.S/ref/ref.elf more frees than allocs from controlfile.general
Test Passed

eroom1966, 2023-03-20:
I need to go back and look at this.
I realised over the weekend that we potentially had a setting missing from the PMP configuration

It is an architectural decision on what happens when reading / writing PMP registers which are not included in your set.
In the case of Wally, there are 16 PMP registers, out of a potential 64
The Spec says that attempting to write/read to those registers (16-63) may result in a read of 0 and a write ignored

We have a parameter to control this behavior, so I suspect for your architecture, this should be set to true.

--override cpu/PMP_undefined=T (Boolean) (default=F) (override) Whether accesses to unimplemented PMP registers are undefined (if True) or write ignored and zero (if False)

davidharrishmc, 2023-03-20:
In csrm.sv, access to an unimplemented PMP register should trap.

davidharrishmc, 2023-03-20:
I didn't make any code changes.  I think there is still a bug that PMP errors should occur if we go to user mode without initializing at least one PMP register.  Am I missing something?

I'm not sure why ref.elf is passing right now.

eroom1966, 2023-03-28:
@davidharrishmc 
I reran my privilege tests, and this one failed (the paths will be different in your environment)

/home/moore/git/Imperas/ImperasDV-OpenHW/scripts/cvw/run-elf.bash --elf /home/moore/git/eroom1966/openhwgroup/cvw/priv-elf/cvw/tests/riscof/work/wally-riscv-arch-test/rv64i_m/privilege/src/WALLY-csr-permission-s-01.S/ref/ref.elf --verbose

[ref.elf.gz](https://github.com/openhwgroup/cvw/files/11088942/ref.elf.gz)

the difference occurs when transitioning to supervisor mode
The [NOTE] messages are reports from the RVVI interface of what the RTL has done
All other messages are from the reference model

As can be seen the RTL and REF execute an mret
This should take both into Superisor mode.

The RTL successfully executes the first instruction in Supervisor mode, but the REF complains that there is not access (due to the PMP)

```
# [NOTE] testbench.idv_trace2log.process_event @ 6210: RET,0,295,8000051c,"30200073 mret                   ",,,,CSR300(mstatus)=0000000a00000180 CSRb00(mcycle)=0000000000000268 CSRb02(minstret)=0000000000000127,
# Info 296: 'refRoot/cpu', 0x000000008000051c(trapreturn_finished_m+14): Machine 30200073 mret
# Info   MEMX 0x8000051c 0x8000051c 2 0073
# Info   MEMX 0x8000051e 0x8000051e 2 3020
# Info   mstatus 0000000a00000900 -> 0000000a00000180 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:1 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0]
# [NOTE] testbench.idv_trace2log.process_event @ 6250: RET,0,296,800007c8,"00001f37 lui     x30,0x1        ",x30=0000000000001000,,,CSRb00(mcycle)=000000000000026c CSRb02(minstret)=0000000000000128,
# Info 297: 'refRoot/cpu', 0x00000000800007c8(trap_handler_end_m+10): Supervisor *** FETCH EXCEPTION ***
# Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x800007c8 00001f37 lui     x30,0x1: No access permission for fetch (0x800007c8)
# Info   mstatus 0000000a00000180 -> 0000000a00000900 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:1 VS:0(Off) SPP:1 MPIE:0 UBE:0 SPIE:0 MIE:0 SIE:0]
# Info   mcause 000000000000000b -> 0000000000000001 [Interrupt:0 Code:1(Instruction access fault)]
# Info   mtval 0000000000000000 -> 00000000800007c8
# Info 298: 'refRoot/cpu', 0x0000000080000400(trap_handler_m): Machine 0300006f j       80000430
# Info   MEMX 0x80000400 0x80000400 2 006f
# Info   MEMX 0x80000402 0x80000402 2 0300
# Info (IDV) Instruction executed prior to mismatch '0x8000051c(trapreturn_finished_m+20): 30200073 mret'
# Error (IDV) PC mismatch (HartId:0, PC:0x0000000080000400 trap_handler_m+0):
# Info (IDV)  0>
# Info (IDV)   . dut:0x00000000800007c8 trap_handler_end_m+16
# Info (IDV)   . ref:0x0000000080000400 trap_handler_m+0
# Error (IDV) Insn. bit pattern mismatch (HartId:0, PC:0x0000000080000400 trap_handler_m+0):
# Info (IDV)  1>
# Info (IDV)   . dut:00001f37 lui     x30,0x1
# Info (IDV)   . ref:0300006f j       80000430
# Error (IDV) GPR register value mismatch (HartId:0, PC:0x0000000080000400 trap_handler_m+0):
# Info (IDV)  2> GPR x30
# Info (IDV)   . dut:0x0000000000001000
# Info (IDV)   . ref:0x00000000ffffffff
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x0000000080000400 trap_handler_m+0):
# Info (IDV)  3> CSR 300 (mstatus)
# Info (IDV)   . dut:0x0000000a00000180 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:1 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0
# Info (IDV)   . ref:0x0000000a00000900 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:1 VS:0(Off) SPP:1 MPIE:0 UBE:0 SPIE:0 MIE:0 SIE:0
# Info (IDV)  4> CSR 342 (mcause)
# Info (IDV)   . dut:0x000000000000000b Interrupt:0 Code:11(Environment call from M-mode)
# Info (IDV)   . ref:0x0000000000000001 Interrupt:0 Code:1(Instruction access fault)
# Info (IDV)  5> CSR 343 (mtval)
# Info (IDV)   . dut:0x0000000000000000
# Info (IDV)   . ref:0x00000000800007c8
```

davidharrishmc, 2023-03-28:
I’ll be on zoom in a few minutes.

davidharrishmc, 2023-03-28:
PR 169 fixes this issue.

Now there are 7 failing privileged tests remaining.

eroom1966, 2023-03-28:
Is this after rebuilding the tests ?

davidharrishmc, 2023-03-28:
> Passes for me.
> running /home/harris/cvw/tests/riscof/work/wally-riscv-arch-test/rv64i_m/privilege/src/WALLY-csr-permission-s-01.S/ref/ref.elf Test Passed
> You’ll need to remake the test case for it to pass (the case needs to configure the PMP before going into supervisor mode).  Either run make (slow) or go into tests/riscof and do
> make wally32 wally64 memfiles
> @ross144 <https://github.com/ross144> @davidharrishmc <https://github.com/davidharrishmc>
> Hi Ross, David
> I am a little confused, I see Ross merged the PR, I have updated to the main branch which has this merged - but running this test still fails in the same way ?

Oops - last word should be memfile

eroom1966, 2023-03-28:
@davidharrishmc 
Do you still have a testcase which shows the bad behavior ?
ie a test which attempts to go into Supervisor mode, which **has not configured the PMP**
Otherwise changing the testcase is just masking the problem I think ?

davidharrishmc, 2023-03-28:
Unfortunately not anymore.

That will be a trickier case to write; it will need some new infrastructure to terminate on the failure.  It’s on my long term to do list - do you have a need for it sooner?

I did try the old case before correcting it, and confirmed that it threw an infinite loop of traps matching the ImperasDV behavior.

I think we can close this.

David

eroom1966, 2023-03-28:
> I did try the old case before correcting it, and confirmed that it threw an infinite loop of traps matching the ImperasDV behavior.

this is what did not work for me, after I took the updates, it still mismatched as described earlier in this thread

davidharrishmc, 2023-03-29:
Your case is failing for me too.  It's failing in a strange way.  At address 80000188, it writes 0x1e0 to mepc.  It sets up some virtual memory. Then at 800001dc, it does a mret to enter supervisor mode.

The ref model tries to fetch from 1e0 and gets an instruction access fault.  It looks like something complex is going on with virtual memory.  There is an instruction fetch fault that seems to be related to a bad page table entry.

The dut seems to see mepc of 800001e0.  I'm not sure how it got this instead of 1e0.

It looks like the PMP issue is solved but something else complex is going on here.  I'll dig deeper another day.

`# [NOTE] testbench.idv_trace2log.process_event @ 4260: RET,0,195,800001dc,"30200073 mret                   ",,,,CSR300(mstatus)=0000000a001c0080 CSRb00(mcycle)=00000000000001a5 CSRb02(minstret)=00000000000000c3,
# Info 195: 'refRoot/cpu', 0x00000000800001dc(init_supervisor_mode+4e): Machine 30200073 mret
# Info   MEMX 0x800001dc 0x800001dc 2 0073
# Info   MEMX 0x800001de 0x800001de 2 3020
# Info   mstatus 0000000a001c0800 -> 0000000a001c0080 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:1 MXR:1 SUM:1 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0]
# Warning (MEM_NRT) 8-byte read from address 0x80013000 in domain '64-bit Supervisor Virtual unified' failed (no target at this address)
# Error (IDV) hartDasmInsBin: opProcessorRead from address 80013000 failed
# [NOTE] testbench.idv_trace2log.process_event @ 4510: RET,0,196,80013000,"                                ",,,,CSR300(mstatus)=0000000a001c0800 CSR341(mepc)=00000000800001e0 CSR342(mcause)=0000000000000001 CSR343(mtval)=00000000800001e0 CSRb00(mcycle)=00000000000001be CSRb02(minstret)=00000000000000c4,
# Warning (MEM_NRT) 2-byte read from address 0x1e0 in domain '64-bit Supervisor Virtual unified' failed (no target at this address)
# Info 196: 'refRoot/cpu', 0x00000000000001e0: Supervisor *** FETCH EXCEPTION ***
# Warning (RISCV_PTWE) CPU 'refRoot/cpu': Page table entry load failed [address=0x1e0 PTEAddress=0x80017000 access=X]
# Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x000001e0 0000     c.illegal: No access permission for fetch (0x1e0)
# Info   mstatus 0000000a001c0080 -> 0000000a001c0800 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:1 MXR:1 SUM:1 MPRV:0 XS:0(Off) FS:0(Off) MPP:1 VS:0(Off) SPP:0 MPIE:0 UBE:0 SPIE:0 MIE:0 SIE:0]
# Info   mcause 0000000000000000 -> 0000000000000001 [Interrupt:0 Code:1(Instruction access fault)]
# Info   mtval 0000000000000000 -> 00000000000001e0
# Info 197: 'refRoot/cpu', 0x0000000080013000(mtvec_handler): Machine 0400006f j       80013040
# Info   MEMX 0x80013000 0x80013000 2 006f
# Info   MEMX 0x80013002 0x80013002 2 0400
# Info (IDV) Instruction executed prior to mismatch '0x80013000(mtvec_handler+0): 0400006f j       80013040'
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x0000000080013000 mtvec_handler+0):
# Info (IDV)  0> CSR 341 (mepc)
# Info (IDV)   . dut:0x00000000800001e0
# Info (IDV)   . ref:0x00000000000001e0
# Info (IDV)  1> CSR 343 (mtval)
# Info (IDV)   . dut:0x00000000800001e0
# Info (IDV)   . ref:0x00000000000001e0
# [ERROR] testbench.idv_trace2api.state_compare @ 4510: MISMATCH`

davidharrishmc, 2023-03-31:
I've tracked this one a bit further.  The program sets up a page table but no PMP.  The page table is at 80017000, and  entry for virtual address 0 points to physical address 80000000.  mepc is set to 000001E0.  The program does a mret to enter supervisor mode at virtual address 000001E0 (physical address 800001E0) with  address translation enabled. This is a trampoline; the program would have gone to physical address 800001E0 if it had remained in machine mode.  

The hardware page table walker attempts to translate virtual page 0 by looking up the PPN from 80017000.  However, PMP is not configured, so a Load Access Fault occurs because supervisor mode has no access to physical memory. 

The trap should set EPC to 000001E0.  However, the RTL has a complicated interaction of the flush after the MRET and an LSU stall. PCF is set to 000001E0, but PCM is still at 800001E0 by the time the trap occurs.  When the trap occurs, it saves PCM in EPC, getting the wrong value (800001E0 in RTL vs. 000001E0 in Imperas).

This will need a careful understanding of the stall and flush interactions and how they play with the LSUStall during a MRET and subsequent trap.

davidharrishmc, 2023-06-07:
Tested csr-permission working.  Closing issue.

closed 2023-06-07 by davidharrishmc