openhwfoundation/cvw #143: SSTATUS.UXL not updating

eroom1966, 2023-03-20:
Whilst booting Linux I see the following instruction execution

# Info   x19 0000000000006000
# Info 612105: 'refRoot/cpu', 0x0000000080009f94: Machine 30099073 csrw    mstatus,x19

After the execution, we see a difference in SSTATUS
# Info (IDV)  0> CSR 100 (sstatus)
# Info (IDV)   . dut:0x8000000000006000
# Info (IDV)   . ref:0x8000000200006000 SD:1 UXL:2 MXR:0 SUM:0 XS:0(Off) FS:3(Dirty) VS:0(Off) SPP:0 UBE:0 SPIE:0 SIE:0

it appears the UXL bits are 0 in the RTL
When I looked at the RTL, I saw the following expression for setting SSTATUS.UXL
{`QEMU ? 2'b0 : STATUS_UXL},

looks like QEMU=1, this sounds like this should be disabled for a correct operation ?

davidharrishmc, 2023-03-20:
Yes.  This was a hack to agree with QEMU.

eroom1966, 2023-03-21:
OK, I will work out some way of disabling this in my environment

closed 2023-03-21 by eroom1966