openhwfoundation/cvw #145: PMPADDR containing illegal settings

eroom1966, 2023-03-21:
PMPADDR registers are WARL for bits 63:54, the legal value is 0, but the RTL accepts non-zero values in this field

Whilst booting the Linux, this is what we observed

```
# [NOTE] testbench.idv_trace2log.process_event @ 27041950: RET,0,2489167,80003fbc,"3b159073 csrw    pmpaddr1,x11   ",,,,CSR3b1(pmpaddr1)=ffffffffffffffff CSRb00(mcycle)=000000000029433e CSRb02(minstret)=000000000025fb4f,
# Info 2489168: 'refRoot/cpu', 0x0000000080003fbc: Machine 3b159073 csrw    pmpaddr1,x11
# Info   MEMX 0x80003fbc 0x80003fbc 2 9073
# Info   MEMX 0x80003fbe 0x80003fbe 2 3b15
# Info   pmpaddr1 0000000000000000 -> 003fffffffffffff [address:0xfffffffffffffc]
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x0000000080003fbc):
# Info (IDV)  0> CSR 3b1 (pmpaddr1)
# Info (IDV)   . dut:0xffffffffffffffff
# Info (IDV)   . ref:0x003fffffffffffff address:0xfffffffffffffc
```

davidharrishmc, 2023-03-22:
I think I've fixed this one.  PR #149 pending.

eroom1966, 2023-03-22:
That would be great - I made a modification to mask out the upper bits in the RTL, but did not really want to issue a PR for my changes