openhwfoundation/cvw #148: mret from M->S mode does not take pending interrupt

eroom1966, 2023-03-22:
This may be superseded by recent changes to mideleg by @davidharrishmc - I will follow up later

The scenario I have seen by booting Linux using ImperasDV is as follows

CPU is running in M-Mode with interrupts disabled
There is a pending interrupt delegated to S-Mode
an mret instruction is taken to S-Mode, which immediately makes the delegated pending interrupt enabled 

The REF model sees the pending and enabled interrupt, and takes an interrupt exception

The RTL executes and retired the first instruction of the epc address upon mret
Then takes the pending and enabled interrupt, and takes an interrupt exception

The Privilege spec is clear about this behavior here in section 3.1.9
These conditions for an interrupt trap to occur must be evaluated in a bounded amount of time
from when an interrupt becomes, or ceases to be, pending in mip, and must also be evaluated
**immediately following the execution of an xRET instruction** or an explicit write to a CSR on which
these interrupt trap conditions expressly depend (including mip, mie, mstatus, and mideleg).

22/Mar/2023: I need to take the latest updates and check that this is the same behavior

davidharrishmc, 2023-03-22:
My recent changes don’t affect this behavior.  It is a real bug.

davidharrishmc, 2023-03-27:
I tried reproducing this.  It printed a lot of stuff in the transcript, but I didn't see an error message.  Can you provide the relevant part of the transcript showing the failure?

ross144, 2023-03-27:
I want to make sure I'm reading the instruction trace correctly.  I've put new lines between instructions.  The first instruction is load at ..e30, the second load at ...e34, the third a csr at ...598, the last is a store at ...59c.  If I understand this correctly ImperasDV says the first two loads commit, the csr at 598 is flushed and then the store at 59c commits.  However the interrupt should occur on the second load (e34) and should have flushed that instruction.  598 is the base address of the machine trap handler.  I am thinking something is wrong with our tracer reporting the interrupt an instruction later.  This may be independent of the issue Lee is running into.  

I checked all interrupts between 10.48M and 10.63M instructions retired and none produced the erroneous behavior Lee saw.  There was an initial concern the data cache was marking cache lines dirty on flushed stores, but this was a wild goose chase.

# [NOTE] testbench.idv_trace2log.process_event @ 165196420: RET,0,10513843**,ffffffff80379e30,"0385be03 ld      x28,56(x11)**    ",x28=00000000000001a4,,,CSR344(mip)=0000000000000080 CSRb00(mcycle)=0000000000fc11d5 CSRb02(minstret)=0000000000a06db3,
# Info 10514197: 'refRoot/cpu', 0xffffffff80379e30: Supervisor 0385be03 ld      x28,56(x11)
# Info   MEMX 0xffffffff80379e30 0x80579e30 2 be03
# Info   MEMX 0xffffffff80379e32 0x80579e32 2 0385
# Info   MEMR 0xffffffff81267bb8 0x81467bb8 8 00000000000001a4

# [NOTE] testbench.idv_trace2log.process_event @ 165196430: RET,0,10513844,**ffffffff80379e34,"0405be83 ld      x29,64(x11)**    ",x29=ffffffff8021b428,,,CSRb00(mcycle)=0000000000fc11d6 CSRb02(minstret)=0000000000a06db4,
# Info 10514198: 'refRoot/cpu', 0xffffffff80379e34: Supervisor 0405be83 ld      x29,64(x11)
# Info   MEMX 0xffffffff80379e34 0x80579e34 2 be83
# Info   MEMX 0xffffffff80379e36 0x80579e36 2 0405
# Info   MEMR 0xffffffff81267bc0 0x81467bc0 8 ffffffff8021b428
# Info   MEMRM 0x812a3010 0x812a3010 8 0000000000000000 L2 (64-bit Supervisor PMP unified)
# Warning (MEM_NRT) 8-byte read from address 0x80000598 in domain '64-bit Supervisor Virtual unified' failed (no target at this address)

# Error (IDV) hartDasmInsBin: opProcessorRead from address 80000598 failed

# [NOTE] testbench.idv_trace2log.process_event @ 165196580: RET,0,10513845,80000598,"                                ",x4=000000008001e000,,,CSR300(mstatus)=0000000a00000822 CSR340(mscratch)=ffffffff81230000 CSR341(mepc)=ffffffff80379e38 CSR342(mcause)=8000000000000007 CSRb00(mcycle)=0000000000fc11e5 CSRb02(minstret)=0000000000a06db5,
# Info (OP_NCH) GlobalTime:0.000000 LocalTime:0.000000 Net:refRoot/MTimerInterrupt 0 => 1
# Info (RISCV_IS) CPU 'refRoot/cpu' 0xffffffff80379e38 0485bf03 ld      x30,72(x11):  MIP:00000000->00000080
# Info 10514199: 'refRoot/cpu', 0xffffffff80379e38: Supervisor *** FETCH EXCEPTION ***
# Info   mstatus 0000000a000000a2 -> 0000000a00000822 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:1 VS:0(Off) SPP:0 MPIE:0 UBE:0 SPIE:1 MIE:0 SIE:1]
# Info   mepc ffffffff80202e5c -> ffffffff80379e38
# Info   mcause 0000000000000009 -> 8000000000000007 [Interrupt:1 Code:7(Machine timer interrupt)]
# Info   mip 0000000000000000 -> 0000000000000080 [MEIP:0 SEIP:0 MTIP:1 STIP:0 MSIP:0 SSIP:0]
# Info 10514200: 'refRoot/cpu', 0x0000000080000598: Machine 34021273 csrrw   x4,mscratch,x4
# Info   MEMX 0x80000598 0x80000598 2 1273
# Info   MEMX 0x8000059a 0x8000059a 2 3402
# Info (OP_NCH) GlobalTime:0.000000 LocalTime:0.105142 Net:refRoot/readcsr 983296 => 262976
# Info   x4 ffffffff81230000 -> 000000008001e000
# Info   mscratch 000000008001e000 -> ffffffff81230000

# [NOTE] testbench.idv_trace2log.process_event @ 165196810: RET,0,10513846,8000059c,"04523423 sd      x5,72(x4)      ",,,,CSRb00(mcycle)=0000000000fc11fc CSRb02(minstret)=0000000000a06db6,
# Info 10514201: 'refRoot/cpu', 0x000000008000059c: Machine 04523423 sd      x5,72(x4)
# Info   MEMX 0x8000059c 0x8000059c 2 3423
# Info   MEMX 0x8000059e 0x8000059e 2 0452
# Info   MEMW 0x8001e048 0x8001e048 8 0000000000000000

eroom1966, 2023-03-28:
Here is the issue, after the mret, the RTL should take the pending interrupt delegated to Supervisor mode, but instead it executes the instruction lhu

```
>> RTL <<
# [NOTE] testbench.idv_trace2log.process_event @ 172122210: RET,0,10748776,800006f8,"30200073 mret                   ",,,,CSR300(mstatus)=0000000a000000a2 CSRb00(mcycle)=000000000106a338 CSRb02(minstret)=0000000000a40368,
>> REF <<
# Info 10748948: 'refRoot/cpu', 0x00000000800006f8: Machine 30200073 mret
# Info   MEMX 0x800006f8 0x800006f8 2 0073
# Info   MEMX 0x800006fa 0x800006fa 2 3020
# Info (RISCV_IS) CPU 'refRoot/cpu' 0xffffffff80299622 0024d783 lhu     x15,2(x9):  SIP:00000000->00000020
# Info   mstatus 0000000a00000822 -> 0000000a000000a2 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:1 MIE:0 SIE:1]

>> RTL <<
# [NOTE] testbench.idv_trace2log.process_event @ 172122370: RET,0,10748777,ffffffff80299622,"0024d783 lhu     x15,2(x9)      ",x15=000000000000000b,,,CSRb00(mcycle)=000000000106a348 CSRb02(minstret)=0000000000a40369,
>> REF <<
# Info 10748949: 'refRoot/cpu', 0xffffffff80299622: Supervisor *** FETCH EXCEPTION ***
# Info   sstatus 0000000200000022 -> 0000000200000120 [SD:0 UXL:2 MXR:0 SUM:0 XS:0(Off) FS:0(Off) VS:0(Off) SPP:1 UBE:0 SPIE:1 SIE:0]
# Info   sepc ffffffff802a38a4 -> ffffffff80299622
# Info   mstatus 0000000a000000a2 -> 0000000a000001a0 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:1 MPIE:1 UBE:0 SPIE:1 MIE:0 SIE:0]
# Info 10748950: 'refRoot/cpu', 0xffffffff80200c8c: Supervisor 14021273 csrrw   x4,sscratch,x4
# Info   MEMX 0xffffffff80200c8c 0x80400c8c 2 1273
# Info   MEMX 0xffffffff80200c8e 0x80400c8e 2 1402

# Info (OP_NCH) GlobalTime:0.000000 LocalTime:0.107490 Net:refRoot/readcsr 983809 => 262464
# Info   x4 ffffffff812339c0 -> 0000000000000000
# Info   sscratch 0000000000000000 -> ffffffff812339c0
# Error (IDV) PC mismatch (HartId:0, PC:0xffffffff80200c8c):
# Info (IDV)  0>
# Info (IDV)   . dut:0xffffffff80299622
# Info (IDV)   . ref:0xffffffff80200c8c
# Error (IDV) Insn. bit pattern mismatch (HartId:0, PC:0xffffffff80200c8c):
# Info (IDV)  1>
# Info (IDV)   . dut:0024d783 lhu     x15,2(x9)
# Info (IDV)   . ref:14021273 csrrw   x4,sscratch,x4
# Error (IDV) GPR register value mismatch (HartId:0, PC:0xffffffff80200c8c):
# Info (IDV)  2> GPR x4
# Info (IDV)   . dut:0xffffffff812339c0
# Info (IDV)   . ref:0x0000000000000000
# Info (IDV)  3> GPR x15
# Info (IDV)   . dut:0x000000000000000b
# Info (IDV)   . ref:0x0000000000000000
# Error (IDV) CSR register value mismatch (HartId:0, PC:0xffffffff80200c8c):
# Info (IDV)  4> CSR 100 (sstatus)
# Info (IDV)   . dut:0x0000000200000022 SD:0 UXL:2 MXR:0 SUM:0 XS:0(Off) FS:0(Off) VS:0(Off) SPP:0 UBE:0 SPIE:1 SIE:2
# Info (IDV)   . ref:0x0000000200000120 SD:0 UXL:2 MXR:0 SUM:0 XS:0(Off) FS:0(Off) VS:0(Off) SPP:1 UBE:0 SPIE:1 SIE:0
# Info (IDV)  5> CSR 140 (sscratch)
# Info (IDV)   . dut:0x0000000000000000
# Info (IDV)   . ref:0xffffffff812339c0
# Info (IDV)  6> CSR 141 (sepc)
# Info (IDV)   . dut:0xffffffff802a38a4
# Info (IDV)   . ref:0xffffffff80299622
# Info (IDV)  7> CSR 300 (mstatus)
# Info (IDV)   . dut:0x0000000a000000a2 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:1 MIE:0 SIE:1
# Info (IDV)   . ref:0x0000000a000001a0 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:1 MPIE:1 UBE:0 SPIE:1 MIE:0 SIE:0
# [ERROR] testbench.idv_trace2api.state_compare @ 172122370: MISMATCH
# 
# Dump State GPR
#  0: 0000000000000000  1: ffffffff80299720  2: ffffffff81263aa0  3: ffffffff8109e0d0 
#  4: 0000000000000000  5: ffffffff8129e780  6: 0000000000000228  7: 0000000000000142 
#  8: ffffffff81263ad0  9: ffffffff812243c0 10: ffffffff812243c0 11: 0000000000000081 
# 12: 000000000000002f 13: 000000000000002f 14: 0000000000000040 15: 0000000000000000 
# 16: a30ef767e4dce200 17: ffc0000000000000 18: ffffffff8109e0a8 19: 000000000000002e 
# 20: 339dc50787ccab19 21: ffffffffffffefff 22: fffffffffffffff6 23: fffffffffffff000 
# 24: ffffffff81268020 25: ffffffff81263b98 26: 0000000000000000 27: 000000000000002f 
# 28: 0000000000000220 29: 0000000000000001 30: ffffffff81012748 31: ffffffff81268025 
# 
# Dump State FPR
#  0: ffffffff00000000  1: ffffffff00000000  2: ffffffff00000000  3: ffffffff00000000 
#  4: ffffffff00000000  5: ffffffff00000000  6: ffffffff00000000  7: ffffffff00000000 
#  8: ffffffff00000000  9: ffffffff00000000 10: ffffffff00000000 11: ffffffff00000000 
[... 23 lines cut ...]
#         pmpaddr5: 0000000000000000         pmpaddr6: 0000000000000000         pmpaddr7: 0000000000000000         pmpaddr8: 0000000000000000 
#         pmpaddr9: 0000000000000000        pmpaddr10: 0000000000000000        pmpaddr11: 0000000000000000        pmpaddr12: 0000000000000000 
#        pmpaddr13: 0000000000000000        pmpaddr14: 0000000000000000        pmpaddr15: 0000000000000000         scontext: 0000000000000000 
#          tselect: 0000000000000000           tdata1: 0000000000000000           tdata2: 0000000000000000           tdata3: 0000000000000000 
#            tinfo: 000000000000807d         tcontrol: 0000000000000000         mcontext: 0000000000000000        mscontext: 0000000000000000 
[... cut ...]
#     hpmcounter21: 0000000000000000     hpmcounter22: 0000000000000000     hpmcounter23: 0000000000000000     hpmcounter24: 0000000000000000 
#     hpmcounter25: 0000000000000000     hpmcounter26: 0000000000000000     hpmcounter27: 0000000000000000     hpmcounter28: 0000000000000000 
#     hpmcounter29: 0000000000000000     hpmcounter30: 0000000000000000     hpmcounter31: 0000000000000000        mvendorid: 0000000000000000 
#          marchid: 0000000000000000           mimpid: 0000000000000100          mhartid: 0000000000000000       mconfigptr: 0000000000000000 
# 

```

ross144, 2023-03-28:
Alright. I've got your updated buildroot images running and I hope to trace down this issue today.

ross144, 2023-03-28:
I found the problem.  We use a Committed(F/M) signal to indicate the IFU or LSU has an ongoing cache or bus transaction and should not be interrupted.  At the time of the mret, the IFU is fetching uncacheable invalid instructions asserting CommittedF.  As the IFU finishes the request it unstalls the pipeline but continues to assert CommittedF.  (This is not necessary for the IFU).  In the same cycle the LSU d cache misses.  Because CommittedF is blocking the interrupt the d cache submits a cache line fetch to the EBU.

I am thinking out loud here.    At it's core the Committed(F/M) ensure memory operations are atomic and caches don't get into inconsistent states.  Once the memory operation is completed the LSU/IFU removes the stall but continues to hold Committed(F/M) because the memory operation has completed and it would be wrong to allow an interrupt to occur with a completed load/store.  However this is not true of the IFU.  If we lower CommittedF once the operation is complete then this problem is solved.  The interrupt won't be masked and the LSU will flush the d cache miss.

This requires a minor change in the cachebusfsm and cachefsm.  I will report back after I've confirmed this works.

ross144, 2023-03-28:
Good news. I built the FPGA with the fix and linux still boots and runs applications.  Still waiting on Questa and ImperasDV to finish running.

ross144, 2023-03-28:
The simulation gets past this retired instruction counter.  However the change I made alters when interrupts occur so the instruction stream is different.

PilotAware, 2023-03-28:
Hi Ross
So are you saying that the scenario no longer shows itself with the given change ?

Thx
Lee

ross144, 2023-03-28:
That is unfortunately the situation. I am hoping it shows up later.

ross144, 2023-03-28:
That is unfortunately the situation.

Ross

ross144, 2023-03-29:
This is exciting.  The linux boot is upto 218M instructions in lock step with ImperasDV!

ross144, 2023-03-29:
When it gets to about 490M instructions the kernel outputs a bunch of messages and then the init process launches a login of the root user and runs a few commands.  This is the expected output.

```
[    0.000000] Linux version 5.10.7 (ross@bandicoot) (riscv64-buildroot-linux-gnu-gcc.br_real (Buildroot 2021.05) 10.3.0, GNU ld (GNU Binutils) 2.35.2) #8 Fri Nov 11 17:33:42 CST 2022
[    0.000000] OF: fdt: Ignoring memory range 0x80000000 - 0x80200000
[    0.000000] Initial ramdisk at: 0x(____ptrval____) (44740608 bytes)
[    0.000000] Zone ranges:
[    0.000000]   DMA32    [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000080200000-0x0000000087ffffff]
[    0.000000] software IO TLB: Cannot allocate buffer
[    0.000000] SBI specification v0.2 detected
[    0.000000] SBI implementation ID=0x1 Version=0x9
[    0.000000] SBI v0.2 TIME extension detected
[    0.000000] SBI v0.2 IPI extension detected
[    0.000000] SBI v0.2 RFENCE extension detected
[    0.000000] riscv: ISA extensions acdfimsu
[    0.000000] riscv: ELF capabilities acdfim
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 31815
[    0.000000] Kernel command line: root=/dev/vda ro
[    0.000000] Dentry cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.000000] Sorting __ex_table...
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 61956K/129024K available (1602K kernel code, 2883K rwdata, 2048K rodata, 84K init, 218K bss, 67068K reserved, 0K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]       fixmap : 0xfffffffe3ee00000 - 0xfffffffe3f000000   (2048 kB)
[    0.000000]       pci io : 0xfffffffe3f000000 - 0xfffffffe40000000   (  16 MB)
[    0.000000]      vmemmap : 0xfffffffe40000000 - 0xffffffff3fffffff   (4095 MB)
[    0.000000]      vmalloc : 0xffffffff40000000 - 0xffffffff7fffffff   (1023 MB)
[    0.000000]       lowmem : 0xffffffff80000000 - 0xffffffff87e00000   ( 126 MB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] riscv-intc: 64 local interrupts mapped
[    0.000000] plic: plic@c000000: mapped 53 interrupts with 1 handlers for 2 contexts.
[    0.000000] random: get_random_bytes called from start_kernel+0x30a/0x46c with crng_init=0
[    0.000000] riscv_timer_init_dt: Registering clocksource cpuid [0] hartid [0]
[    0.000000] clocksource: riscv_clocksource: mask: 0xffffffffffffffff max_cycles: 0x8deda1756, max_idle_ns: 440795202166 ns
[    0.000148] sched_clock: 64 bits at 38MHz, resolution 25ns, wraps every 4398046511096ns
[    0.003010] Console: colour dummy device 80x25
[    0.032225] printk: console [tty0] enabled
[    0.034229] Calibrating delay loop (skipped), value calculated using timer frequency.. 76.92 BogoMIPS (lpj=153848)
[    0.036119] pid_max: default: 32768 minimum: 301
[    0.040478] Mount-cache hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.042792] Mountpoint-cache hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.079570] devtmpfs: initialized
[    0.101913] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.104346] futex hash table entries: 256 (order: 1, 12288 bytes, linear)
[    0.191198] clocksource: Switched to clocksource riscv_clocksource
[   31.368815] Unpacking initramfs...
[   41.820657] Freeing initrd memory: 43688K
[   41.837783] workingset: timestamp_bits=62 max_order=15 bucket_order=0
[   43.265197] Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled
[   43.286981] 10000000.uart: ttyS0 at MMIO 0x10000000 (irq = 1, base_baud = 2403875) is a 16550A
[   47.350086] printk: console [ttyS0] enabled
[   47.605522] debug_vm_pgtable: [debug_vm_pgtable         ]: Validating architecture page table helpers
[   47.734166] Freeing unused kernel memory: 84K
[   47.788366] Run /init as init process
init started: BusyBox v1.33.0 (2022-10-25 13:59:42 CDT)
Starting syslogd: OK
Starting klogd: OK
Running sysctl: OK

BusyBox v1.33.0 (2022-10-25 13:59:42 CDT) built-in shell (ash)
Enter 'help' for a list of built-in commands.

Hello this ~/.profile is meant to demonstrate running some basic commands on Wally.
I am root
And I am on WallyHostname
Created myFile.txt and moved it to myDir. It contains:
Hello World!
And farewell!
Created myScript.sh. Running it yields:
Hello this is another example script
Created symLinkToMyScript.sh. Running it yields:
Hello this is another example script
Now let's remove all these example files and scripts
Here is disk usage:
Filesystem                Size      Used Available Use% Mounted on
devtmpfs                 30.3M         0     30.3M   0% /dev
tmpfs                    51.6M         0     51.6M   0% /dev/shm
tmpfs                    51.6M         0     51.6M   0% /tmp
tmpfs                    51.6M      8.0K     51.6M   0% /run
And here are the current processes:
  PID USER       VSZ STAT COMMAND
    1 root      2796 S    init
    2 root         0 SW   [kthreadd]
    3 root         0 IW   [kworker/0:0-eve]
    4 root         0 IW<  [kworker/0:0H]
    5 root         0 IW   [kworker/u2:0]
    6 root         0 IW<  [mm_percpu_wq]
    7 root         0 SW   [ksoftirqd/0]
    8 root         0 SW   [kdevtmpfs]
    9 root         0 SW   [khungtaskd]
   10 root         0 SW   [oom_reaper]
   11 root         0 IW<  [writeback]
   12 root         0 SW   [kcompactd0]
   13 root         0 IW   [kworker/0:1-eve]
   14 root         0 SW   [kswapd0]
   35 root      2796 S    /sbin/klogd -n
   44 root      2796 S    -/bin/sh
   60 root      2864 R    ps
And finally a login prompt.

```

davidharrishmc, 2023-03-31:
Is this ready to close?

closed 2023-05-30 by davidharrishmc