openhwfoundation/cvw #174: WALLY-mie-01 test fails because stimecmp is not enabled in ImperasDV

davidharrishmc, 2023-03-29:
The WALLY-mi-01 test mismatches because stimecmp is not yet enabled in ImperasDV.

run-elf.bash --elf ../tests/riscof/work/wally-riscv-arch-test/rv64i_m/privilege/src/WALLY-mie-01.S/ref/ref.elf --verbose > tmp

>  [NOTE] testbench.idv_trace2log.process_event @ 5760: RET,0,287,800004a0,"14de1073 csrw    stimecmp,x28   ",,,,CSRb00(mcycle)=000000000000023b CSRb02(minstret)=000000000000011f,
# Info 287: 'refRoot/cpu', 0x00000000800004a0(cause_s_time_interrupt+1c): Machine 14de1073 csrw    stimecmp,x28
# Info   MEMX 0x800004a0 0x800004a0 2 1073
# Info   MEMX 0x800004a2 0x800004a2 2 14de
# Warning (RISCV_CSR_UNIMP) CPU 'refRoot/cpu' 0x800004a0 14de1073 csrw    stimecmp,x28: Unimplemented CSR
# Info   mstatus 0000000a00000008 -> 0000000a00001880 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:3 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0]
# Info   mepc 0000000000000000 -> 00000000800004a0
# Info   mcause 0000000000000000 -> 0000000000000002 [Interrupt:0 Code:2(Illegal instruction)]
# Info   mtval 0000000000000000 -> 0000000014de1073
# Info 288: 'refRoot/cpu', 0x0000000080000600(trap_handler_m): Machine 0300006f j       80000630
# Info   MEMX 0x80000600 0x80000600 2 006f
# Info   MEMX 0x80000602 0x80000602 2 0300
# Info (IDV) Instruction executed prior to mismatch '0x8000049c(cause_s_time_interrupt+24): 01c38e33 add     x28,x7,x28'
# Error (IDV) PC mismatch (HartId:0, PC:0x0000000080000600 trap_handler_m+0):
# Info (IDV)  0>
# Info (IDV)   . dut:0x00000000800004a0 cause_s_time_interrupt+28
# Info (IDV)   . ref:0x0000000080000600 trap_handler_m+0

MikeOpenHWGroup, 2023-03-29:
I took the liberty of editing the comment - those "#" marks are interpreted as Markdown headers and I felt we were getting yelled at by GitHub.

eroom1966, 2023-04-04:
Fixed by this PR
https://github.com/openhwgroup/cvw/pull/198

davidharrishmc, 2023-04-04:
Fixed!

closed 2023-04-04 by davidharrishmc