openhwfoundation/cvw #183: WALLY-wfi confuses Imperas reference model

davidharrishmc, 2023-03-31:
WALLY-wfi confuses the Imperas reference model and produces an error.
"The reference model is unable to find a legal set of actions to converge to the provided DUT state."
See the trace below.

@eroom1966 can I get help with this?

To reproduce:

Pull the repository to get the imperas.ic and wallyTracer supporting SSTC necessary to run this case.

run-elf.bash --elf /home/harris/cvw/tests/riscof/work/wally-riscv-arch-test/rv64i_m/privilege/src/WALLY-trap-01.S/ref/ref.elf --verbose

David

` [NOTE] testbench.idv_trace2log.process_event @ 5370: RET,0,281,80000888,"00c08093 addi    x1,x1,12       ",x1=0000000080000890,,,CSRb00(mcycle)=0000000000000214 CSRb02(minstret)=0000000000000119,
 [NOTE] NET,606,MTimerInterrupt,1
 [NOTE] testbench.idv_trace2log.process_event @ 6410: RET,0,282,8000045c,"1e00006f j       8000063c       ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080 CSRb00(mcycle)=000000000000027c CSRb02(minstret)=000000000000011a,
 Error (IDV) The reference model is unable to find a legal set of actions to converge to the provided DUT state.
 Info (IDV) Reference model mode (HartId: 0):
 Info (IDV)   Machine - Machine mode
 Info (IDV) Pending net changes:
 Info (IDV)  - Group 0:
 Info (IDV)  -    MTimerInterrupt => 1,  age = 35 cycles, 0 retires
 Info (IDV) DUT state change (hartId: 0):
 Info (IDV)  - dut.pc               - 0x000000008000045c trap_handler_m+28
 Info (IDV)  - dut.x0               - 0x0000000000000000
 Info (IDV)  - dut.mstatus          - 0x0000000a00001880 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:3 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0
 Info (IDV)  - dut.mepc             - 0x0000000080000890
 Info (IDV)  - dut.mcause           - 0x8000000000000007 Interrupt:1 Code:7(Machine timer interrupt)
 Info (IDV)  - dut.mip              - 0x0000000000000080
 Info (IDV)  - dut.mcycle           - 0x000000000000027c
 Info (IDV)  - dut.minstret         - 0x000000000000011a
 Info (IDV) Reference state change evaluation tree (HartId: 0):
 Info (IDV)  +  Net MTimerInterrupt => 0x1 { when:606 }
 Info (IDV)     +  Exception { pc:0x8000088c(nowrap+12), mcause:0x8000000000000007, mstatus:0xa00001880 }
 Info (IDV)        +  Retire { pc:0x8000045c(trap_handler_m+28) }
 Info (IDV)           > dut.mepc         - 0x0000000080000890
 Info (IDV)             ref.mepc         - 0x000000008000088c
 Info (IDV)          
 Info (IDV)  +  Retire { pc:0x8000088c(nowrap+12) }
 Info (IDV)     > dut.pc           - 0x000000008000045c trap_handler_m+28
 Info (IDV)       ref.pc           - 0x000000008000088c nowrap+12
 Info (IDV)     > dut.mstatus      - 0x0000000a00001880 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:3 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0
 Info (IDV)       ref.mstatus      - 0x0000000a00000008 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:0 MPIE:0 UBE:0 SPIE:0 MIE:1 SIE:0
 Info (IDV)     > dut.mepc         - 0x0000000080000890
 Info (IDV)       ref.mepc         - 0x0000000000000000
 Info (IDV)     > dut.mcause       - 0x8000000000000007 Interrupt:1 Code:7(Machine timer interrupt)
 Info (IDV)       ref.mcause       - 0x0000000000000000 Interrupt:0 Code:0(Instruction address misaligned)
 Info (IDV)    
 Info (IDV) No state change found for reference to converge to DUT state.
 Warning (IDV) The reference is being forced to converge to the DUT state, results that follow should be treated with caution.
 [ERROR] testbench.idv_trace2api.state_compare @ 6410: MISMATCH
`

eroom1966, 2023-04-04:
@davidharrishmc @ross144 
I have been looking at this and thinking about what is being reported over the RVVI interface.

I think the problem is with the tracer, not the RTL, let me explain

The RTL executes this block of code

```
0000000080000880 <nowrap>:
    80000880:   01cea023                sw      t3,0(t4)
    80000884:   00000097                auipc   ra,0x0
    80000888:   00c08093                addi    ra,ra,12 # 80000890 <nowrap+0x10>
    8000088c:   10500073                wfi
```
As each instruction retires, it is reported over the RVVI Tracer as an instruction retirement,
When it gets to the 'wfi' there is effectively a pause, until some condition satisfies the continuation.

What in fact happens in the RTL, is that an MTimerInt occurs, which causes the Processor to take the interrupt

So here is the fundamental question.
When the wfi instruction is paused and the MTimerInt occurs - does the wfi instruction retire ?
I believe it does, as the instruction address placed into the EPC is @wfi + 4 (the next instruction)
also the minstret should increment at this point

over the RVVI intercase we only see the following
```
# [NOTE] testbench.idv_trace2log.process_event @ 5370: RET,0,281,80000888,"00c08093 addi    x1,x1,12       ",x1=0000000080000890,,,CSRb00(mcycle)=0000000000000214 CSRb02(minstret)=0000000000000119,
# [NOTE] testbench.idv_trace2log.process_event @ 6410: RET,0,282,8000045c,"1e00006f j       8000063c       ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080 CSRb00(mcycle)=000000000000027c CSRb02(minstret)=000000000000011a,
```
We do not see the wfi reported as retired.
I have stumbled across this prevously, and there is an interesting post here, please take a look

https://github.com/riscv/riscv-isa-manual/issues/189

What is confusing the reference model, is that it thinks the wfi has not retired.
What do you think ?

eroom1966, 2023-04-04:
Hi David, this PR should fix the issue with Sstc
https://github.com/openhwgroup/cvw/pull/198

It will also show that WFI does not report as retiring

MikeOpenHWGroup, 2023-04-04:
> It will also show that WFI does not report as retiring

Is that technically correct?  I still think the wfi _should_ report as retiring.

davidharrishmc, 2023-04-04:
Ross and I just found and fixed a bug in the RTL that wfi didn't retire when a pending interrupt arrives and is taken.  Now wfi reports retiring.

Between that fix and SSTC support, all but one of the privileged tests is now running.  The last problem is in the WALLY-wfi test, so something else isn't right about WFI.  I'm closing this issue and will work on the new problem.

closed 2023-04-04 by davidharrishmc