openhwfoundation/cvw #197: Missing fmv.h.x and fmv.x.h

davidharrishmc, 2023-04-04:
Need to add fmv.h.x and fmv.x.h to FPU:

Instructions are provided to move bit patterns between the floating-point and integer registers. FMV.X.H moves the half-precision value in floating-point register rs1 to a representation in IEEE 754-2008 standard encoding in integer register rd, filling the upper XLEN-16 bits with copies of the floating-point number’s sign bit.

FMV.H.X moves the half-precision value encoded in IEEE 754-2008 standard encoding from the lower 16 bits of integer register rs1 to the floating-point register rd, NaN-boxing the result.

FMV.X.H and FMV.H.X do not modify the bits being transferred; in particular, the payloads of
non-canonical NaNs are preserved.

BigSwoll01, 2023-04-05:
Hi David Robert Owens III

My partner Atir and I will take this issue over. Thanks

ross144, 2023-04-15:
Are these instructions not implemented or are we just missing coverage?

davidharrishmc, 2023-04-16:
Not implemented. They should be straightforward. If it gets hard let me
know.

davidharrishmc, 2023-06-14:
I'm no longer sure if this is a real bug.  The rv64gc configuration doesn't support Zfh, so I don't think these should be implemented.  This may show up with code coverage later.  Let's put this issue on hold until FPU code coverage rises.

closed 2023-08-25 by davidharrishmc