openhwfoundation/cvw #200: wfi trace mismatch

davidharrishmc, 2023-04-04:
We've fixed the previous bug that wfi didn't retire in Wally.  Now we have a new disagreement with ImperasDV that seems to show Wally retiring a wfi, while ImperasDV doesn't retire it.  Reproduce with

run-elf.bash --elf tests/riscof/work/wally-riscv-arch-test/rv64i_m/privilege/src/WALLY-wfi-01.S/ref/ref.elf

Wally executes a wfi at 8000088c and waits until the MTimerInterrupt fires.  The trace below shows the DUT reporting the wfi retiring, but ref not reporting the wfi retiring.  Hence, DUT.PC = 8000088c when the wfi retires, but ref.PC = 8000045c because it goes straight to the trap handler without reporting WFI retiring. On the next step, DUT.PC goes to 8000045c when the interrupt occurs.

@eroom1966 can you advise on how to keep the trace happy?

Thank you,

David

``
` [NOTE] testbench.idv_trace2log.process_event @ 5370: RET,0,281,80000888,"00c08093 addi    x1,x1,12       ",x1=0000000080000890,,,CSRb00(mcycle)=0000000000000214 CSRb02(minstret)=0000000000000119,
 Info 281: 'refRoot/cpu', 0x0000000080000888(nowrap+8): Machine 00c08093 addi    x1,x1,12
 Info   MEMX 0x80000888 0x80000888 2 8093
 Info   MEMX 0x8000088a 0x8000088a 2 00c0
 Info   x1 0000000080000884 -> 0000000080000890

 [NOTE] NET,606,MTimerInterrupt,1
 [NOTE] testbench.idv_trace2log.process_event @ 6170: RET,0,282,8000088c,"10500073 wfi                    ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080 CSRb00(mcycle)=0000000000000264 CSRb02(minstret)=000000000000011a,
 Error (IDV) The reference model is unable to find a legal set of actions to converge to the provided DUT state.
 Info (IDV) Reference model mode (HartId: 0):
 Info (IDV)   Machine - Machine mode
 Info (IDV) Pending net changes:
 Info (IDV)  - Group 0:
 Info (IDV)  -    MTimerInterrupt => 1,  age = 11 cycles, 0 retires
 Info (IDV) DUT state change (hartId: 0):
 Info (IDV)  - dut.pc               - 0x000000008000088c nowrap+12
 Info (IDV)  - dut.mstatus          - 0x0000000a00001880 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 PRV:0 XS:0(Off) FS:0(Off) MPP:3 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0
 Info (IDV)  - dut.mepc             - 0x0000000080000890
 Info (IDV)  - dut.mcause           - 0x8000000000000007 Interrupt:1 Code:7(Machine timer interrupt)
 Info (IDV)  - dut.mip              - 0x0000000000000080
 Info (IDV)  - dut.mcycle           - 0x0000000000000264
 Info (IDV)  - dut.minstret         - 0x000000000000011a
 Info (IDV) Reference state change evaluation tree (HartId: 0):
 Info (IDV)  +  Net MTimerInterrupt => 0x1 { when:606 }
 Info (IDV)     +  Exception { pc:0x8000088c(nowrap+12), mcause:0x8000000000000007, mstatus:0xa00001880 }
 Info (IDV)        +  Retire { pc:0x8000045c(trap_handler_m+28) }
 Info (IDV)           > dut.pc           - 0x000000008000088c nowrap+12
 Info (IDV)             ref.pc           - 0x000000008000045c trap_handler_m+28
 Info (IDV)           > dut.mepc         - 0x0000000080000890
 Info (IDV)             ref.mepc         - 0x000000008000088c
 Info (IDV)          
 Info (IDV)  +  Retire { pc:0x8000088c(nowrap+12) }
 Info (IDV)     > dut.mstatus      - 0x0000000a00001880 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:3 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0
 Info (IDV)       ref.mstatus      - 0x0000000a00000008 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:0 MPIE:0 UBE:0 SPIE:0 MIE:1 SIE:0
 Info (IDV)     > dut.mepc         - 0x0000000080000890
 Info (IDV)       ref.mepc         - 0x0000000000000000
 Info (IDV)     > dut.mcause       - 0x8000000000000007 Interrupt:1 Code:7(Machine timer interrupt)
 Info (IDV)       ref.mcause       - 0x0000000000000000 Interrupt:0 Code:0(Instruction address misaligned)
 Info (IDV)    
 Info (IDV) No state change found for reference to converge to DUT state.
Warning (IDV) The reference is being forced to converge to the DUT state, results that follow should be treated with caution.
 [ERROR] testbench.idv_trace2api.state_compare @ 6170: MISMATCH
...
 [NOTE] testbench.idv_trace2log.process_event @ 6410: RET,0,283,8000045c,"1e00006f j       8000063c       ",,,,CSRb00(mcycle)=000000000000027c CSRb02(minstret)=000000000000011b,
 Info (IDV) hart0 PC has been resynchronized to 8000045c
 Info   mstatus 0000000a00000008 -> 0000000a00001880 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0->3 VS:0(Off) SPP:0 MPIE:0->1 UBE:0 SPIE:0 MIE:1->0 SIE:0]
 Info   mepc 0000000000000000 -> 0000000080000890
 Info   mcause 0000000000000000 -> 8000000000000007 [Interrupt:1 Code:7(Machine timer interrupt)]
`
> ``

eroom1966, 2023-04-04:
Hi David, I think the trace now looks correct
I will take a detailed look at this tomorrow - apologies for the delay in resolving

eroom1966, 2023-04-05:
Hi @davidharrishmc @ross144 
This is an interesting case, and we have had to think about this very carefully.
In all other testcases (on openhw cores)  I have gone and looked at, whenever a wfi is executed, and surprised to see there are no testcases where interrupts are enabled (as far as I can tell), so in fact the core simply wakes up and executes the next instruction

This is our take on the wfi behavior
When a wfi executes it immediately retires, and updates no CSR state - except for the implicit updates of mcycle/minstret
The core is now in a wait state.

In order to exit this wait state, there are a number of routes, for example
1. wfi is a nop
in this scenario, the next instruction pc+4 is executed, as if the wfi were simply a NOP instruction

2. interrupt (interrupts disabled)
In this scenario, the core resumes, and the next instruction pc+4 will be executed

3. interrupt (interrupts enabled)
In this scenario, the core resumes, and the next instruction will be executed at the trap handler - updating the CSR's appropriately, eg mepc=pc+4

So at the moment we are performing scenario 3, and it looks like this in the wally tracer

```
# [NOTE] testbench.idv_trace2log.process_event @ 6170: RET,0,282,8000088c,"10500073 wfi                    ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080 CSRb00(mcycle)=0000000000000264 CSRb02(minstret)=000000000000011a,
# [NOTE] testbench.idv_trace2log.process_event @ 6410: RET,0,283,8000045c,"1e00006f j       8000063c       ",,,,CSRb00(mcycle)=000000000000027c CSRb02(minstret)=000000000000011b,
```
The changes to mstatus, mepc, mcause are reported in the retirement of wfi, we think this is not correct, and in fact should be as follows

```
# [NOTE] testbench.idv_trace2log.process_event @ 6170: RET,0,282,8000088c,"10500073 wfi                    ",,,,CSRb00(mcycle)=0000000000000264 CSRb02(minstret)=000000000000011a,
# [NOTE] testbench.idv_trace2log.process_event @ 6410: RET,0,283,8000045c,"1e00006f j       8000063c       ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080 CSRb00(mcycle)=000000000000027c CSRb02(minstret)=000000000000011b,
```
In the above, the retirement of wfi updates the implicit CSR's of mcycle, minstret, and enters wait
it immediately resumes, executing the trap handler instruction reporting the current values for mcause, mstatus, mepc,

Optionally (but not required) a TRAP could be reported on the instruction of wfi+4, which reported the updates to mcause, mstatus, mepc
```
# [NOTE] testbench.idv_trace2log.process_event @ 6170: RET,0,282,8000088c,"10500073 wfi                    ",,,,CSRb00(mcycle)=0000000000000264 CSRb02(minstret)=000000000000011a,
# [NOTE] testbench.idv_trace2log.process_event @ 8350: TRAP,0,339,80000890,"0600de37 lui     x28,0x600d     ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080,
# [NOTE] testbench.idv_trace2log.process_event @ 6410: RET,0,283,8000045c,"1e00006f j       8000063c       ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080 CSRb00(mcycle)=000000000000027c CSRb02(minstret)=000000000000011b,
```

If you want to discuss we could have a call to clarify

davidharrishmc, 2023-04-05:
I can chat in 10 minutes if that is still good.

davidharrishmc, 2023-04-11:
@eroom1966 Any thoughts for a fix?

davidharrishmc, 2023-06-08:
This specific test case involves enabling interrupts, setting up a machine timer interrupt to go off 0x60 cycles beyond the present time, then issuing a wfi at 0x8000088c.  The wfi will wait until the interrupt occurs, then jump to the exception handler at 0x8000045c.  We are uncertain whether the wfi is considered to retire.  We believe when this is done that the pc should change from 8000088c to 8000045c and the mepc should become 80000890.

Here is the interesting portion of the log.  dut.mepc becomes 80000890 as expected.  dut.pc is 8000088c rather than 8000045c at the time it is logged.  I wonder if the problem is that pc should be logged as 8000045c instead?  

The log also shows wfi retiring and mistret incrementing from 119 to 11a.  It's unclear to me whether wfi is considered to retire in this circumstance.

```
# [NOTE] testbench.idv_trace2log.process_event @ 6170: RET,0,282,8000088c,"10500073 wfi                    ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080 CSRb00(mcycle)=0000000000000264 CSRb02(minstret)=000000000000011a,
# Error (IDV) The reference model is unable to find a legal set of actions to converge to the provided DUT state.
# Info (IDV) Reference model mode (HartId: 0):
# Info (IDV)   Machine - Machine mode
# Info (IDV) Pending net changes:
# Info (IDV)  - Group 0:
# Info (IDV)  -    MTimerInterrupt => 1,  age = 11 cycles, 0 retires
# Info (IDV) DUT state change (hartId: 0):
# Info (IDV)  - dut.pc               - 0x000000008000088c nowrap+12
# Info (IDV)  - dut.mstatus          - 0x0000000a00001880 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:3 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0
# Info (IDV)  - dut.mepc             - 0x0000000080000890
# Info (IDV)  - dut.mcause           - 0x8000000000000007 Interrupt:1 Code:7(Machine timer interrupt)
# Info (IDV)  - dut.mip              - 0x0000000000000080
# Info (IDV)  - dut.mcycle           - 0x0000000000000264
# Info (IDV)  - dut.minstret         - 0x000000000000011a
# Info (IDV) Reference state change evaluation tree (HartId: 0):
# Info (IDV)  +  Net MTimerInterrupt => 0x1 { when:606 }
# Info (IDV)     +  Exception { pc:0x8000088c(nowrap+12), mcause:0x8000000000000007, mstatus:0xa00001880 }
# Info (IDV)        +  Retire { pc:0x8000045c(trap_handler_m+28) }
# Info (IDV)           > dut.pc           - 0x000000008000088c nowrap+12
# Info (IDV)             ref.pc           - 0x000000008000045c trap_handler_m+28
# Info (IDV)           > dut.mepc         - 0x0000000080000890
# Info (IDV)             ref.mepc         - 0x000000008000088c
# Info (IDV)          
# Info (IDV)  +  Retire { pc:0x8000088c(nowrap+12) }
# Info (IDV)     > dut.mstatus      - 0x0000000a00001880 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:3 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0
# Info (IDV)       ref.mstatus      - 0x0000000a00000008 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:0 MPIE:0 UBE:0 SPIE:0 MIE:1 SIE:0
# Info (IDV)     > dut.mepc         - 0x0000000080000890
# Info (IDV)       ref.mepc         - 0x0000000000000000
# Info (IDV)     > dut.mcause       - 0x8000000000000007 Interrupt:1 Code:7(Machine timer interrupt)
# Info (IDV)       ref.mcause       - 0x0000000000000000 Interrupt:0 Code:0(Instruction address misaligned)
# Info (IDV)    
# Info (IDV) No state change found for reference to converge to DUT state.
# Warning (IDV) The reference is being forced to converge to the DUT state, results that follow should be treated with caution.
# [ERROR] testbench.idv_trace2api.state_compare @ 6170: MISMATCH
# 
# Dump State GPR
#  0: 0000000000000000  1: 0000000080000890  2: 0000000080004008  3: 7fbb6fab7fbb6fab 
#  4: 0000000000000000  5: 00000000ffffffff  6: 0000000080005218  7: 00000000000001f9 
#  8: 5bfddb7d5bfddb7d  9: adfeedbeadfeedbe 10: 0000000000000000 11: 0000000000000000 
# 12: 0000000000000000 13: 0000000000000060 14: f56ff76df56ff76d 15: fab7fbb6fab7fbb6 
# 16: 0000000080005218 17: beadfeedbeadfeed 18: df56ff76df56ff76 19: 6fab7fbb6fab7fbb 
# 20: b7d5bfddb7d5bfdd 21: dbeadfeedbeadfee 22: 6df56ff76df56ff7 23: b6fab7fbb6fab7fb 
# 24: db7d5bfddb7d5bfd 25: edbeadfeedbeadfe 26: 76df56ff76df56ff 27: bb6fab7fbb6fab7f 
# 28: 0000000000000259 29: 0000000002004000 30: 000000000200bff8 31: 0000000000000000 
# 
# Dump State FPR
#  0: 0000000000000000  1: 0000000000000000  2: 0000000000000000  3: 0000000000000000 
#  4: 0000000000000000  5: 0000000000000000  6: 0000000000000000  7: 0000000000000000 
#  8: 0000000000000000  9: 0000000000000000 10: 0000000000000000 11: 0000000000000000 
# 12: 0000000000000000 13: 0000000000000000 14: 0000000000000000 15: 0000000000000000 
# 16: 0000000000000000 17: 0000000000000000 18: 0000000000000000 19: 0000000000000000 
# 20: 0000000000000000 21: 0000000000000000 22: 0000000000000000 23: 0000000000000000 
# 24: 0000000000000000 25: 0000000000000000 26: 0000000000000000 27: 0000000000000000 
# 28: 0000000000000000 29: 0000000000000000 30: 0000000000000000 31: 0000000000000000 
# 
# Dump State CSR
#           fflags: 0000000000000000              frm: 0000000000000000             fcsr: 0000000000000000          sstatus: 0000000200000000 
#              sie: 0000000000000000            stvec: 0000000000000000       scounteren: 0000000000000000          senvcfg: 0000000000000000 
[... cut ...]
# Info 282: 'refRoot/cpu', 0x000000008000045c(trap_handler_m+1c): Machine 1e00006f j       8000063c
# Info   MEMX 0x8000045c 0x8000045c 2 006f
# Info   MEMX 0x8000045e 0x8000045e 2 1e00
# [NOTE] testbench.idv_trace2log.process_event @ 6440: RET,0,284,8000063c,"34011173 csrrw   x2,mscratch,x2 ",x2=0000000080004808,,,CSR340(mscratch)=0000000080004008 CSRb00(mcycle)=000000000000027f CSRb02(minstret)=000000000000011c,
# Info 283: 'refRoot/cpu', 0x000000008000063c(m_time_vector_m): Machine 34011173 csrrw   x2,mscratch,x2
```

ross144, 2023-06-22:
The implication here is that MEPC is always PC+4 for WFI regardless if it retires or not.  This doesn't really make sense to me in the 3rd type.  If the WFI generates an illegal instruction exception and does not retire why doesn't MEPC point to PC?  Wouldn't this confuse the trap handler by indicating the following instruction is the one which causes the illegal instruction? Maybe I'm missing something.

eroom1966, 2023-06-23:
Hi Ross

> The implication here is that MEPC is always PC+4 for WFI regardless if it retires or not.
I messed up my logic, you are correct - this should have read in the case of a timeout MEPC=PC
I have corrected the above

```
3. TW=1 Stalls for some (implementation specific) period, then either retires, 
or takes an illegal instruction exception

if (fetch(PC) == WFI) {
    wait(interrupt || timeout)
    if (interrupt) {
        rvvi.retire(PC)
        if (enabled(interrupt)) {
            MEPC = PC+4
            PC = PC.IRQHANDLER
        } else { 
            PC = PC+4
        }
    } else if (timeout) {
       MEPC = PC
       PC = PC.ILLEGAL_INSTRUCTION
    }
}
```

eroom1966, 2023-06-23:
I went back to look at the Wally tracer, and I think where we have the miscommunication is where the wfi is reported as retiring and where the trap status is reported, currently we see this

```
RET,0,282,8000088c,"10500073 wfi                    ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080 CSRb00(mcycle)=0000000000000264 CSRb02(minstret)=000000000000011a,
```
The WFI is reported as retiring, but updates the state associated with the interrupt, this is not correct

I changed the tracer to re-insert this code
`  assign rvvi.trap[0][0]     = TrapW; // 0; // TODO: IMPERAS TrapW;`

I then saw the following
```
EXC,0,282,8000088c,"10500073 wfi                    ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080 CSRb00(mcycle)=0000000000000264 CSRb02(minstret)=000000000000011a,
RET,0,283,8000045c,"1e00006f j       8000063c       ",,,,CSRb00(mcycle)=000000000000027c CSRb02(minstret)=000000000000011b,
```

this again is not correct, because now the WFI does not get reported as retiring, I think we should be seeing one of the following, either only the instructions that retire and their associated state
```
RET,0,282,8000088c,"10500073 wfi                    ",,,,CSRb00(mcycle)=0000000000000264 CSRb02(minstret)=000000000000011a,
RET,0,283,8000045c,"1e00006f j       8000063c       ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080 CSRb00(mcycle)=000000000000027c CSRb02(minstret)=000000000000011b,

```
or the retired instruction and the intermediate exception
```
RET,0,282,8000088c,"10500073 wfi                    ",,,,CSRb00(mcycle)=0000000000000264 CSRb02(minstret)=000000000000011a,
EXC,0,283,80000890,"1e00006f lui     t3,0x600d      ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080,
RET,0,283,8000045c,"1e00006f j       8000063c       ",,,,CSRb00(mcycle)=000000000000027c CSRb02(minstret)=000000000000011b,
```

There is clearly too much ambiguity regarding the WFI behavior, I tried to begin some clarification here
https://github.com/riscv/riscv-isa-manual/issues/189#issuecomment-1603955297
I think Andrew is in agreement, and this concurs with what I am describing above.

I wonder if this requires a quick conference call so we can come to some agreement ?

ross144, 2023-06-23:
I think I see the problem now, I believe this is a tracer issue. I believe Wally is taking the correct actions and we are in agreement with you and Andrew Waterman.  I've reviewed the simulationn waveform to double check.  MEPC is set to the address after WFI, WFI retires, and PCNextF points to the trap handler.  I'm a bit unsure if I'm reading the imperasDV report correctly, but it seems to me it thinks Wally points PC to the the WFI after the trap.

Info (IDV)  -    MTimerInterrupt => 1,  age = 11 cycles, 0 retires
Info (IDV) DUT state change (hartId: 0):
Info (IDV)  - dut.pc               - **0x000000008000088c nowrap+1**2

I'll take a closer look at the tracer.

ross144, 2023-06-23:
I understand the problem now.  MSTATUS and presumably all CSRs are de-synchronized from when an instruction is indicated committed.  I thought this was handled by the tracer, but it's definitely wrong for WFI.  At the time of the trap, MSTATUS is 0xA000_0008 and then gets updated to 0xA000_1880 on the next cycle.  WFI is in the Memory state during this time.  The CPU then points PCNextF to the trap handler and needs to wait to fill the I$.  Some number of cycles later the WFI commits in the Writeback stage, but now MSTATUS is 0xA000_1800.  At this time the tracer reports the WFI done with the newer MSTATUS.

ross144, 2023-06-23:
I think I'm mistaken.  I may not fully understand what ImperasDV is displaying.  Disregard the previous message.

ross144, 2023-06-23:
After rereading Andrew's message and your description, I'm not convinced our implementation follows ImperasDV's 3 expected paths.  Our implementation looks more like Andrew's message.

"Consider the following algorithm, which combines cases (1, 2) and (3):
    stall the WFI for a bounded amount of time, but retire the instruction and skip the following steps if an interrupt becomes pending
    if the bounded timeout expires, check whether TW should result in an exception; raise the exception if so
    otherwise, retire the WFI, then stall until an interrupt becomes pending"

In our scenario TW = 0 and interrupts are enabled.  Wally executes WFI and stalls the CPU.  Then a timer interrupt fires.  Wally retires WFI and updates the CSRs.  

`CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080 CSRb00(mcycle)=0000000000000264 CSRb02(minstret)=000000000000011a,`

Since the WFI retires MEPC should be PC+4 (0x8000_0890), since a machine timer interrupt occurred mcause= 0x8000...07 and MIP = 0x80.  Mcycle and minstret should increment.

ross144, 2023-06-26:
I've been thinking about this some more after talking with Lee this morning.  I have two major thoughts.
1. Our implementation is not quite power efficient, but I think it's functionally correct according to Andrew's hybrid method.  Our implementation starts by stalling on WFI, but does not retire.  If an interrupt becomes pending, we signal a Trap and WFI retires.  If the bounded timer expires and the priv mode is not machine and TW = 1 then WFI generates an exception.  Similarly if in user mode it generates an exception regardless of TW.  Finally if nether and the timeout occurs then the instruction continues to wait for an interrupt.  However unlike Andrew's description Wally hasn't yet retired the WFI.  Once an interrupt does occur WFI will retire.  What could be improved is not allowing the timer counter to run after expiring.  As this wastes power and will potentially trigger multiple timeouts.

2. I'm not opposed to re-implementing to match ImperasDV as in someways the implementation may be simpler. The hybrid method requires a mux on the trap PC logic to pick either PCM or PCM+4.  Using the Imperas method, the wfi will retire when TW = 0 and we would use wfiW & ~MIP to stall the CPU.  When TW = 1, we need to check wfiM and wait for the timeout.  If it does timeout MEPC = PCM immediate (wfi).  If an interrupt is pending before the timeout, the pipeline will advance one instruction and then generate an interrupt trap.  This part of the hardware will undoubtedly be more awkward that what we already have.  However MEPC = PCM so the PC logic is similar. If the interrupt was disabled no trap occurs but wfi still commits and the CPU unstalls.

ross144, 2023-06-27:
I've been thinking about this more.  Page 47 of the privilege spec V20211203 section 3.3.3 states "If an enabled interrupt is present or later becomes present while the hart is stalled, the interrupt
trap will be taken on the following instruction, i.e., execution resumes in the trap handler and mepc
= pc + 4"

We are partially doing this.  Our implementation generates the interrupt trap while the WFI is in the memory stage and not yet retired.  Wally has an extra mux explained above which sets MEPC to PC+4 so that the trap handler sees the next instruction as the faulting address rather than the WFI.  The WFI does retire as MINSTRET increments.  However this is not exactly consistent with the language on page 47.  One could argue the interrupt really should be delayed until the next instruction. 

I do believe it reasonable to say either implementation is valid as the CSRs are updated to correct values.  The problem for imperas is it doesn't see the WFI retirement  until after the trap occurs.  I believe there are two solutions here.

1. We modify the RTL to behave exactly to the specification and to match imperasDV.
2. We modify the tracer to delay reporting the trap 1 cycle only when wfi is stalled and a non-masked interrupt is pending.
3. 
At this point I'm leaning on 2, because it should be easy and I think I our implementation does follow the ethos of WFI even through it does not strictly follow the loose goose WFI description.

davidharrishmc, 2023-06-29:
Sorry I’m traveling and haven’t followed every nuance.  

The implementation has to exactly match the specification to be correct.  As long as we are satisfied ImperasDV doesn’t have a bug, we also have to exactly match ImperasDV for a suitable tracer output. However, the microarchitecture (and RTL) can do whatever we want to long as a programmer can’t observe any deviation from the specification.  In the context of this issue, I believe the required behavior is when an interrupt fires while the hart is waiting, MINSTRET increments, EPC = PC + 4, and PC goes to the interrupt handler.  I believe this is exactly what happens in cvw. Do you agree?

If we have the correct behavior, then I think option 2 is preferred.  Otherwise we have to fix the RTL to have the correct behavior.

The “interrupt really should be delayed until the next instruction,”  “we are partially doing this” and “modify the RTL to behave exactly to the specification and to match ImpersDV” and “does not stricttly follow the loose goose WFI description” statements are throwing me off.  As far as I understand, the cvw architectural state is always exactly correct, and the discrepancy we are seeing is simply a matter of the interaction of our pipeline and tracer causing a trace to mismatch ImperasDV.  If this is true, we need to fix the tracer.

David

davidharrishmc, 2023-08-24:
Lee says there has been progress on ImperasDV.

eroom1966, 2023-10-05:
we will take a look at this tomorrow, and try to reproduce the testcase

eroom1966, 2023-10-06:
Hi @ross144 @davidharrishmc 

I reviewed this example once again, having updated from IDV and CVW.
I think we may want to have a call to go through this in more detail, but firstly let me try to explain what I think is happening here.

There is still an issue, but I think it is just a reporting issue really, let me try to explain, but also, I think it is worth re-reading this post here first
https://github.com/riscv/riscv-isa-manual/issues/189#issuecomment-1603955297

I think the case we are looking at with this example, is described by case 2
```
if (fetch(PC) == WFI) {
    retire(PC)
    wait(interrupt)
    if (enabled(interrupt)) {
        MEPC = PC+4
        PC = IRQHANDLER
    } else { 
        PC = PC+4
    }
}
```

What currently gets reported is the following
```
# Info (IDV) testbench.idv_trace2log.process_event @ 5370: RET,0,281,80000888,"00c08093 addi    x1,x1,12       ",x1=0000000080000890,,,CSRb00(mcycle)=0000000000000214 CSRb02(minstret)=0000000000000119,
# Info (IDV) NET,606,MTimerInterrupt,1
# Info (IDV) testbench.idv_trace2log.process_event @ 6170: RET,0,282,8000088c,"10500073 wfi                    ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080 CSRb00(mcycle)=0000000000000264 CSRb02(minstret)=000000000000011a,
# Info (IDV) testbench.idv_trace2log.process_event @ 6410: RET,0,283,8000045c,"1e00006f j       8000063c       ",,,,CSRb00(mcycle)=000000000000027c CSRb02(minstret)=000000000000011b,
```

This does not quite match the pseudo code. It appears the retirement of wfi is stalled until an event occurs, in this case the timer interrupt. whereas the wfi should retire unconditionally, then stall awaiting an event to continue to the next instruction. 
The next instruction to be retired will depend upon whether interrupts were enabled, in which case retire first instruction of trap handler
or interrupts were disabled,  in which case retire(retire next instruction (&wfi)+4

If interrupts were enabled, then the state update is communicated by a TRAP event providing the updated registers, or the updated state registers are provided with the retired instruction of the trap handler. 
So I think the events should produce one of the following

```
<< case 1 >>
# Info (IDV) testbench.idv_trace2log.process_event @ 5370: RET,0,281,80000888,"00c08093 addi    x1,x1,12       ",x1=0000000080000890,,,CSRb00(mcycle)=0000000000000214 CSRb02(minstret)=0000000000000119,
# Info (IDV) testbench.idv_trace2log.process_event @ 6170: RET,0,282,8000088c,"10500073 wfi                    ",,,,CSRb00(mcycle)=0000000000000264 CSRb02(minstret)=000000000000011a,
# Info (IDV) testbench.idv_trace2log.process_event @ 6410: RET,0,283,8000045c,"1e00006f j       8000063c       ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080 CSRb00(mcycle)=000000000000027c CSRb02(minstret)=000000000000011b,
```
or
```
<< case 2 >>
# Info (IDV) testbench.idv_trace2log.process_event @ 5370: RET,0,281,80000888,"00c08093 addi    x1,x1,12       ",x1=0000000080000890,,,CSRb00(mcycle)=0000000000000214 CSRb02(minstret)=0000000000000119,
# Info (IDV) testbench.idv_trace2log.process_event @ 6170: RET,0,282,8000088c,"10500073 wfi                    ",,,,CSRb00(mcycle)=0000000000000264 CSRb02(minstret)=000000000000011a,
# Info (IDV) testbench.idv_trace2log.process_event @ 6410: TRAP,0,283,80000890,"0600de37 lui     x28,0x600d     ",,,,CSR300(mstatus)=0000000a00001880 CSR341(mepc)=0000000080000890 CSR342(mcause)=8000000000000007 CSR344(mip)=0000000000000080,
# Info (IDV) testbench.idv_trace2log.process_event @ 6410: RET,0,284,8000045c,"1e00006f j       8000063c       ",,,,CSRb00(mcycle)=000000000000027c CSRb02(minstret)=000000000000011b,

```

ross144, 2023-10-24:
Wally is doing what the tracer reports.  But the timing is skewed because of pipelining, so it is technically correct.
addi (pc=8000_0888) retires
Concurrently wfi (pc=8000_088c) is in the Memory stage and the pipeline is stalled waiting for the interrupt.
The interrupt occurs, then wfi advances and is officially retired.  However it's important to know wfi is an exception compared to other instructions in the situation.  All other instructions would be flushed.  In away wfi is implicitly retired because it cannot be flushed by the interrupt effectively saying it is retired before the interrupt.

This is a bit screwy and I don't really love this solution, but we could modify the tracer to to indicate the wfi retires in the Memory stage.

ross144, 2023-10-24:
This approach does not work. By indicating the wfi retires in the memory stage, the instruction in the writeback stage gets reordered relative to the wfi.

ross144, 2023-10-24:
There is something else in here which does not make sense to me. Why is TrapM delayed 5 cycles?

```
  flopenrc #(1)     IntrFReg (clk, reset, 1'b0, ~StallF, TrapM, IntrF);
  flopenrc #(1)     IntrDReg (clk, reset, FlushD, ~StallD, IntrF, IntrD);
  flopenrc #(1)     IntrEReg (clk, reset, FlushE, ~StallE, IntrD, IntrE);
  flopenrc #(1)     IntrMReg (clk, reset, FlushM, ~StallM, IntrE, IntrM);
  flopenrc #(1)     IntrWReg (clk, reset, FlushW, ~StallW, IntrM, IntrW);
```

ross144, 2023-10-24:
Does ImperasDV ignore rvvi.intr and use the interrupt pending and enabled bits to figure out when an interrupt occurs?  Otherwise I can't see how ImperasDV reports the interrupt occurs before the wfi retirement.

ross144, 2023-10-25:
After a bit of effort I believe I have a good solution.  I think easiest thing to do is change the implementation.
I'm testing and reviewing a modification which has the wfi stall in the Memory stage.  If it times out it behaves exactly as before.  If an interrupt becomes pending, rather than take the TrapM immediately the hazard unit lets the wfi advance to the writeback stage.  Then it lets the TrapM occur.  This has the added advantage of simplifying some of the PCNextF logic as the PCM+4 adder is no longer needed when wfi interrupts.

I'm going to re-review my work tomorrow and then consider merging.

ross144, 2023-10-26:
Fixed by PR #441

closed 2023-10-26 by davidharrishmc