openhwfoundation/cvw #203: MEPC seems to have wrong value on exception

eroom1966, 2023-04-05:
I have a testcase here that is from riscv-dv
it attempts to switch from Machine to User mode, but fails as the PMP has not been setup correctly
The strange thing is, that the mepc value at the point if the mret instruction is 0x1d8

A fetch exception occurs (correctly) in the RTL and the REF, but for some reason the RTL reports the mepc as 0x800001d8 instead of 0x000001d8

elf file attached to re-run

Thx
Lee
[riscv_mmu_stress_test_sdel_0.o.elf.gz](https://github.com/openhwgroup/cvw/files/11158967/riscv_mmu_stress_test_sdel_0.o.elf.gz)

ross144, 2023-04-05:
This is good.  I was not able to reproduce the issue on 132.  Hopefully this will point me to the issue quickly.

davidharrishmc, 2023-04-05:
> I think I see the problem. The IFU fetches 0x1d8 which generates a hptw walker address fault. This fault should be pipelined to the memory stage but immediately asserts TrapM. PCM contains old stall address which just happens to be the physical address 0x8000_01d8.
>
> I think there may also be a second issue. SelHPTW is still asserted when exception is asserted asserted. This will cause the to stall I believe we need to modify the HPTW fsm to lower SelHPTW when the exception occurs.

You’ve made short work of this diagnosis!

ross144, 2023-04-05:
Hmm. This is a bit trickier than I thought.  I made the change to pipeline the HPTWInstrAccessFault; however the hptw fsm does not terminate immediately so the fault is lost.  I thought we already had a test showing this worked for the analogous load hptw pma fault.  I'm not understanding what is fundamentally different.  Either I believe we need to modify the fsm.

ross144, 2023-04-05:
I've fixed the MEPC issue but there is still an issue.   I'll come back to this after class.
```

# [NOTE] testbench.idv_trace2log.process_event @ 4210: RET,0,193,800001d4,"30200073 mret                   ",,,,CSR300(mstatus)=0000000a00080080 CSRb00(mcycle)=00000000000001a0 CSRb02(minstret)=00000000000000c1,
# Warning (MEM_NRT) 8-byte read from address 0x8002f000 in domain '64-bit User Virtual unified' failed (no target at this address)
# Error (IDV) hartDasmInsBin: opProcessorRead from address 8002f000 failed
# [NOTE] testbench.idv_trace2log.process_event @ 4550: RET,0,194,8002f000,"                                ",,,,CSR300(mstatus)=0000000a00080000 CSR342(mcause)=0000000000000001 CSR343(mtval)=00000000000001d8 CSRb00(mcycle)=00000000000001c2 CSRb02(minstret)=00000000000000c2,
# Warning (MEM_NRT) 2-byte read from address 0x1d8 in domain '64-bit User Virtual unified' failed (no target at this address)
# Warning (RISCV_PTWE) CPU 'refRoot/cpu': Page table entry load failed [address=0x1d8 PTEAddress=0x80033000 access=X]
# Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x000001d8 0000     c.illegal: No access permission for fetch (0x1d8)
```

eroom1966, 2023-04-05:
@ross144 
This is not a mismatch, its simply an error in attempted disassembly of inaccessible memory 
If there is no mismatch reported I think you are good to go

ross144, 2023-04-05:
PR #208 fixes this bug. @davidharrishmc please review the changes.  This modifies the hptw fsm a fair bit. I will be running an fpga build to make sure this did not break anything else.

ross144, 2023-04-05:
The FPGA linux boot and benchmarks are still working!

davidharrishmc, 2023-04-05:
Whoo!

ross144, 2023-05-30:
This should be fixed.

davidharrishmc, 2023-06-07:
Tested and it passes.  Closing the issue.

closed 2023-06-07 by davidharrishmc