openhwfoundation/cvw #225: ImperasDV Generates Load Page Fault in Linux Boot

ross144, 2023-04-10:
@eroom1966 Hi Lee,  

The linux boot makes it to 541M instructions before mismatching with ImperasDV with a load page fault.  I am wondering if the issue is how ImperasDV handles access and dirty bits when set to 0.  The trace below shows ImperasDV taking a load page fault, but Wally does not.  Wally's HPTW reads Access = 0 but updates the PTE by setting the access bit in both the TLB and writing back to memory.  Is there a way to configure ImperasDV to match? 

Thanks,

Ross

```
 Info (RISCV_PTWE) CPU 'refRoot/cpu': Page table entry A=0 [address=0x3fdacb7000 PTEAddress=0x802025b8 access=R]
# Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x3fdac1463c 3787b783 ld      x15,888(x15): Page fault at load address (0x3fdacb79b0)
# Info   sstatus 0000000200004022 -> 0000000200004020 [SD:0 UXL:2 MXR:0 SUM:0 XS:0(Off) FS:2(Clean) VS:0(Off) SPP:0 UBE:0 SPIE:1 SIE:0]
# Info   sepc 0000003fdac14628 -> 0000003fdac1463c
# Info   scause 000000000000000c -> 000000000000000d [Interrupt:0 Code:13(Load page fault)]
# Info   stval 0000003fdac14628 -> 0000003fdacb79b0
# Info   mstatus 0000000a000040a2 -> 0000000a000040a0 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:2(Clean) MPP:0 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:1 MIE:0 SIE:0]
# Info 543173024: 'refRoot/cpu', 0xffffffff80200c8c: Supervisor 14021273 csrrw   x4,sscratch,x4
# Info   MEMX 0xffffffff80200c8c 0x80400c8c 2 1273
# Info   MEMX 0xffffffff80200c8e 0x80400c8e 2 1402
# Info (OP_NCH) GlobalTime:0.000000 LocalTime:5.431730 Net:refRoot/readcsr 983296 => 262464
# Info   x4 0000003fdacbb2f0 -> ffffffff81237380
# Info   sscratch ffffffff81237380 -> 0000003fdacbb2f0
# Error (IDV) PC mismatch (HartId:0, PC:0xffffffff80200c8c):
```

davidharrishmc, 2023-04-10:
Ross,

I see two lines commented out in the imperas.ic that might help to uncomment.

#--override cpu/updatePTEA=T
#--override cpu/updatePTED=T

David

ross144, 2023-04-10:
Ahh bummer. I had an older version of imperas.ic which didn't have these two lines.  Well. I'll start up the simulation again.  I don't see a reason to keep this issue open.

eroom1966, 2023-04-10:
Hi Ross
Yes I am sure this is the issue
There is a config option to handle this which is commented out with a ‘#’
in the imperas.ic file
The reason it is commented as I recall is that it conflicts with one of the
privilege tests

Today is a public holiday in the uk
I will check tomorrow which priv test conflicted- unless you can rerun all
the priv tests today ?

Thx
Lee

-- 
====================================================================
The information contained in this electronic mail message and any
attachments hereto
is privileged and confidential information intended only for the use of the
individual or
entity named above or their designee.  If the reader of this message is not
the intended
recipient, you are hereby notified that any dissemination, distribution or
copying of this
communication is strictly prohibited.  If you have received this
communication in error
please immediately notify us by return  message or by telephone and delete
the
original message from your mail system.  Thank you.
====================================================================

davidharrishmc, 2023-04-10:
Reopening this issue.

I found SVADU_SUPPORTED was 1 in the buildroot wally-config and 0 in rv64gc wally-config.

I uncommented the updatePTEA/D lines in imperas.ic and reran iter-elf.
iter-elf.bash --search tests/riscof/work/wally-riscv-arch-test/rv64i_m/privilege/src

WALLY-mmu-sv39 failed.

I then turned SVADU_SUPPORTED = 1 in rv64gc, which is what we said it should be in chapter 3. WALLY-mmu-sv39 passes.

So I think the issue was a misconfiguration in rv64gc.  Now that imperas.ic and wally-config both have SVADU_SUPPORTED=1, we are matching.

David

eroom1966, 2023-04-11:
@davidharrishmc 
Great catch, so we should now make the following the default settings, for the reference model

--override cpu/updatePTEA=T
--override cpu/updatePTED=T

I will issue a PR

davidharrishmc, 2023-04-11:
I have PR229 in for this, but it breaks some of our regression tests, so I’m waiting to fix those before pulling.

David

davidharrishmc, 2023-04-12:
229 and 231 collectively fix this.  Ross, please close this when you are satisfied.