openhwfoundation/cvw #276: PostProcess Module Error in the cvtshiftcalc.sv file IntToFp Line 105 with solution proposed

Thomas-J-Kidd, 2023-04-24:
# PostProcess Module Error in the cvtshiftcalc.sv file IntToFp Line 105 with proposed solution

We found a error in the ```rv64gc_uncovered_fpu.rpt``` 

On line 105, the variable ```IntToFp``` is never 1 within the covered test. To investigate this error, we needed to find where the ```IntToFp``` variable comes from. After some digging, we found it in the ```postprocess.sv``` file when the ```cvtshiftcalc``` module is initialized. Within the ```postprocess.sv``` file we traced ```InToFP``` down and found it was assigned to the third bit of the ```OpCtrl```. Backtracking ```OpCtrl```, we found it was in the ```fpu.sv``` file initiated using the ```OpCtrlM``` control signal, which function is to select which operation to do in each component. Eventually we found that ```OpCtrlM``` is not assigned anywhere in the fpu. On the other hand, OpCtrlE is used in assigning many times. 

Code issue in the ```rv64gc_uncovered_fpu.rpt``` 
```
Expression Coverage:
    Enabled Coverage              Bins   Covered    Misses  Coverage
    ----------------              ----      ----    ------  --------
    Expressions                      3         2         1    66.66%

================================Expression Details================================

## Expression Coverage for instance /core/fpu/fpu/postprocess/cvtshiftcalc --

  File ../src/fpu/postproc/cvtshiftcalc.sv
----------------Focused Expression View-----------------
Line       105 Item    1  ((XZero ~| IntToFp) & (CvtCe < {{5{1}},ResNegNF}))
Expression totals: 2 of 3 input terms covered = 66.66%

                   Input Term   Covered  Reason for no coverage   Hint
                  -----------  --------  -----------------------  --------------
                        XZero         Y
                      IntToFp         N  '_1' not hit             Hit '_1'
  (CvtCe < {{5{1}},ResNegNF})         Y

```

## Issue in the ```cvtshiftcalc.sv``` file:

```
  assign CvtResUf = ($signed(CvtCe) < $signed({{`NE-$clog2(`NF){1'b1}}, ResNegNF}))&~XZero&~IntToFp;

```

## Our solution

Our idea to fix this issue ended up by removing IntToFp from line 105 in the ```cvtshiftcalc.sv``` file. Our reasoning behind this is from the fact that we could not find how IntToFp was being assigned any other value other than 0. If this is the case then removing IntToFp is a modification we can make. 

## Modified code: 

```  assign CvtResUf = ($signed(CvtCe) < $signed({{`NE-$clog2(`NF){1'b1}}, ResNegNF}))&~XZero&;```

## Fully covered in rvgc64_covered_fpu.rpt

```
================================Expression Details================================

Expression Coverage for instance /core/fpu/fpu/postprocess/cvtshiftcalc --

  File ../src/fpu/postproc/cvtshiftcalc.sv
----------------Focused Expression View-----------------
Line       105 Item    1  ((CvtCe < {{5{1}},ResNegNF}) & ~XZero)
Expression totals: 2 of 2 input terms covered = 100.00%
```

Author: Zach Wilson and Thomas Kidd @Zach-Stealth-Wilson @Thomas-J-Kidd

davidharrishmc, 2023-04-25:
OpCtrlM comes from the flip-flop at line 330 of fctrl.sv.  OpCtrlE comes from another flip-flop in fctrl.sv.    Integer to floating-point conversions shouldn't underflow.

The clue to this one is nearby in the uncovered report.
  Row   4:    ***0***  IntToFp_1                      ((CvtCe < {{5{1}},ResNegNF}) && ~XZero)

You'll need to try to create a suitable test case with IntToFp = 1 (a conversion instruction from integer to floating-point) with CvtCe meeting this inequality, and X not being zero.  This will take understanding CvtCe.  The logic is described in Chapter 13.

davidharrishmc, 2023-06-07:
James Stine working on FP coverage.  Closing this issue.

closed 2023-06-07 by davidharrishmc