openhwfoundation/cvw #301: text.h_32-01 hangs in ImperasDV

davidharrishmc, 2023-05-24:
I’m having an odd hanging behavior on one of the bit manipulation tests (see below).  run-elf.bash hangs at 80000080 on a mv x31, x0 instruction.  All the other B tests that I’ve tried so far are working, and the B tests pass in regression when I’m not using ImperasDV. 

run-elf.bash --elf /home/harris/cvw/tests/riscof/work/riscv-arch-test/rv32i_m/B/src/zext.h_32-01.S/dut/my.elf --verbose

# [NOTE] testbench.idv_trace2log.process_event @ 700: RET,0,32,8000007c,"09c08093 addi    x1,x1,156      ",x1=0000000080003114,,,CSRb00(mcycle)=0000000000000041 CSRb02(minstret)=0000000000000020,
# Info 32: 'refRoot/cpu', 0x000000008000007c(rvtest_code_begin+4): Machine 09c08093 addi    x1,x1,156
# Info   MEMX 0x8000007c 0x8000007c 2 8093
# Info   MEMX 0x8000007e 0x8000007e 2 09c0
# Info   x1 0000000080003078 -> 0000000080003114
# [NOTE] testbench.idv_trace2log.process_event @ 710: RET,0,33,80000080,"00000f93 mv      x31,x0         ",x31=0000000000000000,,,CSRb00(mcycle)=0000000000000042 CSRb02(minstret)=0000000000000021,
# Info 33: 'refRoot/cpu', 0x0000000080000080(inst_0): Machine 00000f93 mv      x31,x0
# Info   MEMX 0x80000080 0x80000080 2 0f93
# Info   MEMX 0x80000082 0x80000082 2 0000

davidharrishmc, 2023-06-08:
Hmm.  Now the mv is at 800000e0, but it still hangs

```
# Info 56: 'refRoot/cpu', 0x00000000800000dc(rvtest_code_begin+34): Machine 00000013 nop
# Info   MEMX 0x800000dc 0x800000dc 2 0013
# Info   MEMX 0x800000de 0x800000de 2 0000
# [NOTE] testbench.idv_trace2log.process_event @ 1050: RET,0,57,800000e0,"00000f93 mv      x31,x0         ",x31=0000000000000000,,,CSRb00(mcycle)=0000000000000064 CSRb02(minstret)=0000000000000039,
# Info 57: 'refRoot/cpu', 0x00000000800000e0(inst_0): Machine 00000f93 mv      x31,x0
# Info   MEMX 0x800000e0 0x800000e0 2 0f93
# Info   MEMX 0x800000e2 0x800000e2 2 0000
```

eroom1966, 2023-06-12:
@davidharrishmc @ross144 

I instrumented the testcase to display a clock tick, and the RVVI .valid signal.
After the RTL reports the retirement of the instruction here, no more retirements are reported over the RVVI interface

```
# [NOTE] testbench.idv_trace2log.process_event @ 710: RET,0,33,80000080,"00000f93 mv      x31,x0         ",x31=0000000000000000,,,CSRb00(mcycle)=0000000000000042 CSRb02(minstret)=0000000000000021,
# tick 710
# valid 710
# Info 33: 'refRoot/cpu', 0x0000000080000080(inst_0): Machine 00000f93 mv      x31,x0
# Info   MEMX 0x80000080 0x80000080 2 0f93
# Info   MEMX 0x80000082 0x80000082 2 0000
# tick 720
# tick 730
# tick 740
# tick 750
# tick 760
...
```

so this sounds like either a tracer issue, or a bug in the RTL

ross144, 2023-06-16:
The problem is Wally generates an illegal instruction access fault on 
800000e4:	080fcfb3          	zext.h	t6,t6
which then goes to a non-existent trap handler at address 0x0.  Since there are no instructions there it repeatedly generates illegal instruction access faults at address 0x0.

ross144, 2023-06-16:
Hmm. Something more interesting is happening here.  This should be a 32bit test, but the configuration is 64 bit.  That's probably why it generates the exception.

davidharrishmc, 2023-06-16:
There are also two 64-bit tests: rv64i_m/B/src/zext.h-01.S and zext.h_64-01.S.  The former looks stale.

The 32-bit test can't run on our 64-bit ImperasDV because it has an illegal opcode in RV64.

So the issue is resolved.  I was running iter-elf in the wrong directory, triggering zext.h_32-01.S, which has an opcode incompatible with rv64gc.

closed 2023-06-16 by davidharrishmc