openhwfoundation/cvw #310: CoreMark performance has dropped from earlier versions

davidharrishmc, 2023-06-05:
CM/MHz = 2.30 and CPI = 1.25.
Earlier, CM/MHz = 2.54 and CPI = 1.16

cd ~/cvw/benchmarks/coremark
make
make run

```
 Read memfile ../benchmarks/coremark/work/coremark.bare.riscv.elf.memfile
#    WALLY CoreMark Results (from get_time)
#     Elapsed MTIME: 4342759
#     Elapsed MINSTRET: 3448364
#     COREMARK/MHz Score: 10,000,000 / 4342759 = 2.30 
#     CPI: 4342759 / 3448364 = 1.25
# 2K performance run parameters for coremark.
# CoreMark Size    : 666
# Total ticks      : 4342759
# Total time (secs): 43
# Iterations/Sec   : 0
# Iterations       : 10
# Compiler version : GCC13.1.1 20230509
# Compiler flags   : -mcmodel=medany -fno-tree-loop-distribute-patterns -fno-common -lm -lgcc -T /home/harris/cvw/benchmarks/coremark/riscv64-baremetal/link.ld -g -mabi=lp64 -march=rv64gc -static -falign-functions=16 -mbranch-cost=1 -DSKIP_DEFAULT_MEMSET -mtune=sifive-3-series -O3 -finline-functions -falign-jumps=4 -fno-delete-null-pointer-checks -fno-rename-registers --param=loop-max-datarefs-for-datadeps=0 -funroll-all-loops --param=uninlined-function-insns=8 -fno-tree-vrp -fwrapv -fipa-pta -nostdlib -nostartfiles -ffreestanding -mstrict-align -DTOTAL_DATA_SIZE=2000 -DMAIN_HAS_NOARGC=1 -DPERFORMANCE_RUN=1 -DITERATIONS=10 -DXLEN=64   -static-libgcc -lgcc
# Memory location  : Code and Data in external RAM
# seedcrc          : 0xe9f5
# [0]crclist       : 0xe714
# [0]crcmatrix     : 0x1fd7
# [0]crcstate      : 0x8e3a
# [0]crcfinal      : 0xfcaf
# Correct operation validated. See README.md for run and reporting rules.
# Load Stalls 633002
# D-Cache Accesses 287810
# D-Cache Misses 32197
# I-Cache Accesses 688034
# I-Cache Misses 83
# Branches 15096
# Branches Miss Predictions 31423
# BTB Misses 271619
# Jump, JAL, JALR 80940
# RAS Wrong 256524
# Returns 0
# BP Class Wrong 679521
# Done printing performance counters
# Benchmark: coremark is done.
# ** Note: $stop    : ../testbench/testbench.sv(306)
#    Time: 48662225 ns  Iteration: 1  Instance: /testbench
# Break at ../testbench/testbench.sv line 306
# Stopped at ../testbench/testbench.sv line 306
# End time: 12:05:30 on May 30,2023, Elapsed time: 0:03:51
# Errors: 0, Warnings: 0
1.03user 0.10system 3:52.03elapsed 0%CPU (0avgtext+0avgdata 85924maxresident)k
464inputs+42248outputs (147major+78110minor)pagefaults 0swaps
python3 coremark-postprocess.py
```

ross144, 2023-06-05:
Hmm.  The branch predictor counts look very wrong.  So to the counts of d-cache accesses relative to load stalls.  I am rerunning the simulation in gui mode and the counters appear to be more reasonable.  This either means the driver (syscall.c) is still wrong or it was not updated when compiled. Looking through the source code it looks valid to me.

ross144, 2023-06-05:
There is definitely something wrong with instruction class prediction.  We are getting way more than we should.

ross144, 2023-06-05:
I think I found the problem.  When we removed the forces from the testbench we missed on.  This was forcing 0 for all entries of the BTB for all time and resulted in significant misprediction rates.  I'm rerunning now.  This should fix the issues.

ross144, 2023-06-06:
This is fixed by pr #313.

closed 2023-06-07 by davidharrishmc