openhwfoundation/cvw #323: embench postprocessing crashes

davidharrishmc, 2023-06-11:
benchmark_speed.py in the embench postprocessing is crashing right now.  Perhaps the format of CSRs printed by embench has changed.

The crash can be reproduced by
cd $WALLY/benchmarks/embench
make clean
make
make run

This takes a long time to run all the benchmarks, then invokes the following command, which can also be invoked directly after the benchmarks have been run once:

../../addins/embench-iot/benchmark_speed.py --builddir=bd_sizeopt_speed --target-module run_wally --cpu-mhz=1 --json-output > wallySizeOpt_speed.json 

The error is shown below:
```
harris@chips:~/cvw/benchmarks/embench$ ../../addins/embench-iot/benchmark_speed.py --builddir=bd_sizeopt_speed --target-module run_wally --cpu-mhz=1 --json-output > wallySizeOpt_speed.json 
/usr/lib/python3/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (1.25.8) or chardet (5.1.0) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
Traceback (most recent call last):
  File "../../addins/embench-iot/benchmark_speed.py", line 389, in <module>
    sys.exit(main())
  File "../../addins/embench-iot/benchmark_speed.py", line 367, in main
    raw_data, rel_data = collect_data(benchmarks, remnant)
  File "../../addins/embench-iot/benchmark_speed.py", line 285, in collect_data
    raw_data[bench] = benchmark_speed(bench, target_args)
  File "../../addins/embench-iot/benchmark_speed.py", line 216, in benchmark_speed
    exec_time = decode_results(
  File "/home/harris/cvw/addins/embench-iot/pylib/run_wally.py", line 67, in decode_results
    pc_trigger = list(map(lambda s: int(s,16), output_signature[0:5]))
  File "/home/harris/cvw/addins/embench-iot/pylib/run_wally.py", line 67, in <lambda>
    pc_trigger = list(map(lambda s: int(s,16), output_signature[0:5]))
ValueError: invalid literal for int() with base 16: ''
```

davidharrishmc, 2023-06-14:
The crash is occurring because embench-iot/bd_sizeopt_speed/src/aha-monte64/aha-monte64.sim.output is empty.

ross144, 2023-06-16:
Is this post processing fixed?

davidharrishmc, 2023-06-18:
I had to enable md5sum, prime count, and stalemate, which aren't part of the Embench 1.0.  Otherwise various things were crashing unable to find them.  We will need to strip them out and recalculate the geometric mean manually.

closed 2023-06-18 by davidharrishmc