openhwfoundation/cvw #354: CPU gets stuck and watchdog triggered during test

jont-imperas, 2023-06-30:
While running IDV with this test (attached) it reports the CPU gets stuck:
# FAILURE: Watch Dog Time Out triggered. PCW stuck at 0000000080002106 for more than     1000000 cycles
The test runs fine on the standalone ISS. I wondered if it was to do with memory map, but the nearest memory access instructions go to an address that has previously been read/written, so I don't think that's the issue.

ELF file attached. [
[a.out.gz](https://github.com/openhwgroup/cvw/files/11918357/a.out.gz)
](url)

The instructions around the problem location look fairly innocuous in the disassembly:
    80002102:   e406                    sd      ra,8(sp)
    80002104:   078a                    slli    a5,a5,0x2
    80002106:   00018713                mv      a4,gp
    8000210a:   97ba                    add     a5,a5,a4
    8000210c:   439c                    lw      a5,0(a5)

It would be great if someone can have a look and investigate why this fails to make forwards progress.

ross144, 2023-06-30:
I'm taking a look now.

ross144, 2023-06-30:
@jont-imperas  When the Watch dog time out occurs it say Wally is hung up. This usually happens when Questa hangs up because of x propagation.  At 3501ns the program attempts to LW from address 0x8000_23a8 which is not initialized or previously written to so it returns x's.  Then cpu then goes on to use that value in a branch instruction causing the processor to hang up indefinitely.  I suspect this is a program with reading from an uninitialized variable.

Looking at the object dump, address 0x8000_23a8 is at the beginning of the .bss segment.

davidharrishmc, 2023-06-30:
The bss segment should be initialized to 0 when the program is loaded. Is
it lacking 0 in the elf?  Maybe an issue about how don’t automatically
initialize bss when rhe test bench reads an elf?

jont-imperas, 2023-06-30:
Thanks for the quick replies. Yes, I think it's an oversight in my startup code not initializing all of the bss. I thought I'd fixed that but I'll revisit.

ross144, 2023-06-30:
The memfile itself has no entries for address 0x8000_23a8.

ross144, 2023-07-05:
@jont-imperas Is this issue resolved?

jont-imperas, 2023-07-06:
Yes, this is resolved.

closed 2023-07-06 by jont-imperas