openhwfoundation/cvw #405: Bug in spill logic for uncachable memory

davidharrishmc, 2023-09-02:
I'm getting a weird fetch behavior on uncacheable memory and I think it is related to a non-multiple-of-4 address.

See tests/coverage/tlbNAPOT.S and .objdump at 8000004a.  The program should fetch the ecall.  PBMT is set to make the memory uncacheable.

done:
    li a0, 4            # argument to finish program    
    80000048:	4511                	li	a0,4
    ecall               # system call to finish program
    8000004a:	00000073          	ecall
    j self_loop         # wait forever (not taken)
    8000004e:	a06d                	j	800000f8 <self_loop>

In Questa at time 373406, PCNextF = 8000004a.  PostSpillInstrRawF gets 00735411.  This has part of the ecall in the upper half, which seems to be a bug in the IFU on this non-multiple-of-2 address.

This instruction ends in a 1, making the Decomp Unit decompress it into something weird, which eventually causes an IllegalInstruction trap.

Following a little more, at 373396, HREADRam = 06d0000000734511 because HADR was 80000048 (the multiple of 8 below 8000004a). Thus InstrRawF = 00734511.

In spill.sv:

  localparam                SPILLTHRESHOLD = P.ICACHE_SUPPORTED ? P.ICACHE_LINELENINBITS/32 : 1; 

  assign SpillF = &PCF[$clog2(SPILLTHRESHOLD)+1:1];

Thus, it appears that the spill threshold is not adjusted for uncacheable memory.

@ross144 can you sort this one out?  It might also be a good idea to set Cacheable to 0 in the memory in general (by changing the PMA) and verifying that regression still works after this fix.

David

ross144, 2023-12-26:
Looks like I already implemented this and it just needs to be tested.

davidharrishmc, 2023-12-28:
rv32gc_arch32c and rv64gc_arch64c hang when uncore memory not cacheable.  See many other timeouts recorded in issue 440.

ross144, 2023-12-29:
I've been having difficulty reproducing this bug.  Before this happens the program goes haywire because the "ill-formed NAPOT pages" test.

The test generates a bogus address for which is there is no valid instruction address translation and intends to generate an instruction page fault.  The trap handler then attempts to read the instruction at this address to detect if the instruction was compressed.  This second operation also traps and the program gets stuck in an infinite trap loop.

The underlying problem my be resolved with the other changes I am about to push.  There were a whole bunch of bugs in the uncached logic.

ross144, 2024-01-07:
I am still unable to reproduce this bug.  The tlbNAPOT program runs to completion without errors.  I can see it is not fetching instructions from the bus but rather from the I$.

As a side note.  00734511 should still decompress the 4511 (li a0, 4) into the full instruction correctly.  I wonder why it was decompressing incorrectly.

davidharrishmc, 2024-01-07:
The test is passing for me when run normally, but failing when ICACHE_SUPPORTED=0 in config/rv32gc/config.vh.  Haven't tried reproducing with uncachable PBMT-based memory tonight.

davidharrishmc, 2024-01-13:
Also fails when DCACHE_SUPPORTED = 0.  rv64gc_arch64a, rv64gc_arch64zifencei, rv64gc_wally64priv.  Maybe zifence should fail without a cache?

ross144, 2024-01-15:
wally64periph fails because timing changes.
arch64a fails because there is a bug in the non-cache implementation which does not support atomics.  This would be easy to fix.  
wally64zifencei fails because the controller disables FlushDCacheD if the ICache is not present. I've fixed this one and I am working on the others.

ross144, 2024-01-16:
Most of these are fixed with my latest pull request.  However the zicclsm spec says that both cacheablity and coherence PMA must be supported. The wally64priv test is running misaligned tests so it will fail without a data cache.

There is still one more issue to solve.  The zifencei test fails with a D$, but without an I$.

davidharrishmc, 2024-01-16:
Sounds good.  We should document the expectation of failure on wally64priv somehow.

davidharrishmc, 2024-01-18:
Almost done...

After incorporating the SPI test fix, rv32gc and rv64gc pass all tests with DCACHE=1,ICACHE=0.  I'm not seeing zifencei failing in this situation.

However, rv64gc_arch64zifencei fails with DCACHE=0, ICACHE=1.

rv64gc_wally64priv fails with DCACHE=0 because of the misaligned issue described above, as expected.  This is not a problem.

rv32gc fails lint with DCACHE=0.

```
harris@chips:~/cvw/sim$ ./lint-wally 
rv32e linting...
rv64gc linting...
rv32gc linting...
%Warning-WIDTHTRUNC: ../src/lsu/lsu.sv:372:62: Input port connection 'ByteMask' expects 4 bits on the pin connection, but pin connection's VARREF 'ByteMaskM' generates 8 bits.
                                             : ... note: In instance 'wallywrapper.dut.core.lsu'
  372 |         .HWSTRB(LSUHWSTRB), .BusRW, .BusAtomic(AtomicM[1]), .ByteMask(ByteMaskM), .WriteData(LSUWriteDataM[P.XLEN-1:0]),
      |                                                              ^~~~~~~~
                     ... For warning description see https://verilator.org/warn/WIDTHTRUNC?v=5.021
                     ... Use "/* verilator lint_off WIDTHTRUNC */" and lint_on around source to disable this message.
%Warning-WIDTHEXPAND: ../src/lsu/lsu.sv:377:36: Operator ASSIGNW expects 64 bits on the Assign RHS, but Assign RHS's SEL generates 32 bits.
                                              : ... note: In instance 'wallywrapper.dut.core.lsu'
  377 |       else assign ReadDataWordMuxM = FetchBuffer[P.XLEN-1:0];
      |                                    ^
%Error: Exiting due to 2 warning(s)
Exiting after rv32gc lint due to errors or warnings
```

ross144, 2024-01-18:
I have the fix for the no d cache with I cache. I just need to push out the
changes.

ross144, 2024-01-18:
All the possible to fix bugs should be fixed with the latest pull request, #580.

davidharrishmc, 2024-01-18:
rv32gc still has a lint problem with DCACHE=0

ross144, 2024-01-18:
Thanks for checking I completely forgot about that one.

closed 2024-02-08 by davidharrishmc