openhwfoundation/cvw #411: Make: Error no such files or directory

Krishnakumarmohanraj, 2023-09-16:
Hi Team,

    When we do make, we are getting below error

creating testfloat sqrt test vectors
creating arch test vectors
Traceback (most recent call last):
  File "./extract_arch_vectors.py", line 294, in <module>
    create_vectors(c)
  File "./extract_arch_vectors.py", line 83, in create_vectors
    src_file2 = open(source_dir2 + vector2,'r')
FileNotFoundError: [Errno 2] No such file or directory: '/Projects/marmik_project/manicka.vinodhini/krishna/cvw/tests/riscof/work/riscv-arch-test/rv32i_m/M/src/div-01.S/ref/Reference-sail_c_simulator.signature'
make[1]: *** [testfloat] Error 1
make[1]: Leaving directory `/Projects/marmik_project/manicka.vinodhini/krishna/cvw'
make: *** [all] Error 2

ross144, 2023-09-18:
Which makefile is this (which directory did you run in)?

Krishnakumarmohanraj, 2023-09-19:
[manicka.vinodhini@localhost cvw]$ make
make install
make[1]: Entering directory `/Projects/marmik_project/manicka.vinodhini/krishna/cvw'
[manicka.vinodhini@localhost cvw]$ pwd
/Projects/marmik_project/manicka.vinodhini/krishna/cvw
(base) [manicka.vinodhini@localhost cvw]$ ls
addins           dvtestplan.md  README.md         synthDC
benchmarks       examples       setup.csh         testbench
bin              fpga           setup.imperas.sh  tests
bugs.txt         gitflow.txt    setup.sh          wallyriscvTopAll.png
config           LICENSE        sim
CONTRIBUTING.md  linux          src
docs             Makefile       studies
(base) [manicka.vinodhini@localhost cvw]$

Krishnakumarmohanraj, 2023-09-29:
Hi Rose Thompson, any update on this issue

ross144, 2023-09-29:
Hi Krishna,

My apologies I completely forgot about this. 

I am not able to reproduce your error.  Did you make sure to source the setup.sh file prior to running make?

Also provide more context? What system are you running? Ubuntu version? redhat? (We currently cannot get sail to install on redhat.) Did you run the tool chain install script? or did your system admin already install all the required packages?

Krishnakumarmohanraj, 2023-10-03:
Hi Rose Thompson, 

We sourced setup.sh and run make, still we facing error

[image]
[image]

ross144, 2023-10-03:
There is a subtle bug with sourcing bash files. You need to run
source ./setup.sh

You may also need to edit setup.sh to make sure the following are set to the correct values for your system.
1. MGLS_LICENSE_FILE should point to your Siemens license server
2. SNPSLMD_LICENSE_FILE should point to your Synopsys license server. This is only required if your going to do synthesis.
3. QUESTA_HOME should point to the directory where questa is installed.

The make will fail if you haven't installed the tools with the tool chain install script.  This takes a while (> 1 hr) to build.
If you haven't installed yet you need to do the following first.
source ./setup.sh # after edtting to setup your licenses
cd bin
./wally-tool-chain-install.sh

Now you can go back and run make

davidharrishmc, 2023-10-07:
PR #417 also reversed the order of making riscof and testfloat, which should help this.  There was a change this spring about test float tests and I think you are the first one to find it broke tool installation.

ross144, 2023-10-23:
@Krishnakumarmohanraj Were you able to get CVW up and running?

Krishnakumarmohanraj, 2023-10-30:
> CAUTION: This email originated from outside the Organization. Do not open Link or Attachments unless you recognize the sender and know the content is safe.

Hi Rose Thompson,

  When i try to run ./wally-tool-chain-install.sh in ubuntu, am getting below error message

g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [Makefile:1157: insn-emit.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/opt/riscv/riscv-gnu-toolchain/gcc-13/gcc/expmed.cc: In function â€˜rtx_def* extract_bit_field_1(rtx, poly_uint64, poly_uint64, int, rtx, machine_mode, machine_mode, bool, bool, rtx_def**)â€™:
/opt/riscv/riscv-gnu-toolchain/gcc-13/gcc/expmed.cc:1838:45: warning: â€˜*(unsigned int*)((char*)&imode + offsetof(scalar_int_mode, scalar_int_mode::m_mode))â€™ may be used uninitialized [-Wmaybe-uninitialized]
 1838 |       rtx sub = extract_bit_field_as_subreg (mode1, op0, imode,
      |                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
 1839 |                                              bitsize, bitnum);
      |                                              ~~~~~~~~~~~~~~~~
/opt/riscv/riscv-gnu-toolchain/gcc-13/gcc/expmed.cc:1798:19: note: â€˜*(unsigned int*)((char*)&imode + offsetof(scalar_int_mode, scalar_int_mode::m_mode))â€™ was declared here
 1798 |   scalar_int_mode imode;
      |                   ^~~~~
rm gfdl.pod gcc.pod gcov-dump.pod gcov-tool.pod fsf-funding.pod gpl.pod cpp.pod gcov.pod lto-dump.pod
make[2]: Leaving directory '/opt/riscv/riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc'
make[1]: *** [Makefile:4615: all-gcc] Error 2
make[1]: Leaving directory '/opt/riscv/riscv-gnu-toolchain/build-gcc-newlib-stage1'
make: *** [Makefile:591: stamps/build-gcc-newlib-stage1] Error 2
(base) ***@***.***:~/cvw/bin#

Thanks,
Krishna

Disclaimer

“This e-mail message (including any attachments) is intended to be received only by person(s) entitled to receive the confidential and/or legally privileged information it may contain. If you are not the intended recipient, you are notified that you have received this message by error, and any review, dissemination, distribution, storage or copying of the contents of this message is strictly prohibited. Please notify the sender immediately by email, and delete the message and its contents completely from your computer system and from all devices where it could have downloaded and all online Email inboxes, without disclosing it to anyone.”

davidharrishmc, 2023-10-30:
> Hi Krishna,
>
> The RISC-V toolchain is a little temperamental to get set up correctly. This seems like an issue with gcc. We were able to install correctly with Ubuntu operating systems. I might suggest cleaning and starting again. Let me know if you are able to get further - happy to help!
>
> All my best,
>
> James

Agreed with James.

While troubleshooting, I find it helpful to run the commands from wally-tool-chain-install.sh one at a time by hand and watch that everything looks right.

If you can find the root cause, we’d love to update wally-tool-chain-install to be more robust.

David

ross144, 2023-10-31:
Hi Krishna,

I am concerned by all the troubles you've had installing Wally and the tool chain.

What system are you running?  Ubuntu? What version?

davidharrishmc, 2023-12-14:
Hi, Krishna.  Are you still working on this?  We would be happy to zoom and help.  I'm closing the issue because it seems to be related to your environment.  Please reply by email if you want to walk through this.

closed 2023-12-14 by davidharrishmc