openhwfoundation/cvw #412: Issue creating file for loading program into memory

duncangraham-Imperas, 2023-09-21:
When I attempt to run one of my tests in the cvw environment I am finding that the infrastructure uses a Python script to convert the binary to a hex file (${RISCV}/lib/python2.7/site-packages/freedom-bin2hex.py) for loading into memory.

My test has pages distributed through the memory space and this causes a huge file to be generated, filling intermediate (unused) address space with 0's.

If I use a comamnd line such as the objcopy shown below; I get a file containing only the code sections required to run but I have no mechanism, I am aware of, to load this file.
`${RISCV}/bin/riscv64-unknown-elf-objcopy  -O verilog  test-S_0.elf outfile`
Before I start trying to modify the testbench to load using a file such as this output format into the verilog memory I wanted to reach out to see if you have observed this issue before and if you have a recommented approach to handle it.

Thanks

ross144, 2023-09-21:
Hi Duncan,

Our testbench only support continuous preload memories. How big is your application?  You may need to also adjust the configuration files to support a larger physical address space.  The default is 128MiB.

Imperas, 2023-09-22:
I think you may have just given me the prompt in the correct direction.
The test was generated for a generic platform, we need to do some configuration for cvw. Thanks. I will let you know how we get on.

duncangraham-Imperas, 2023-09-22:
Unfortunately, that idea did not work. The application I am trying to use has program defined at one address and some at a higher address. This is resulting in the bin2hex creaeting a huge file filling the intermediate addresses with 0x00000000.
What are the required files for loading into the system verilog memory, perhaps there is an alternate approach I could use.

ross144, 2023-09-22:
I wonder, could we divide up the binary into multiple contiguous segments.  Use bin2hex.py to convert each segment into a preloadable memfile and then use separate $readmemh() function calls for each specifying the starting address for each segment.

eroom1966, 2023-09-22:
Objdump already has an output format compatible with creating a sparse memory file for verilog
This same format can be loaded directly into a verilog memory using $readmem

ross144, 2023-09-22:
I wasn't aware $readmemh could directly preload a verilog memory.  Is there a special flag or option to enable this?

ross144, 2023-09-22:
Now that I think about this I recall there was an option for objcopy to output readmemh compatible memfiles, but there was a bug which prevented it from handling endianess correctly and it only output the text segment or some similar issue.  

I'm not sure if this is the same objdump output you're thinking of Lee.

eroom1966, 2023-09-26:
@ross144 
We reproduced what you reported that the bytes are swapped, for the order expected in your memory file.
We have built a small utility that will take the order and reverse.
This means we can create a sparse verilog memory file, and successfully load into your verilog memory
we will commit something to the repo when complete.
This resolves the issue of creating 256Gb memory files containing all zeros !

duncangraham-Imperas, 2023-09-26:
Hi @ross144, this may be the incorrect place to add this, maybe a new issue would be better (happy to move, let me know).

After the changes we were able to load the memory with an MMU access test we are developing. We have found that this is locking up. In the zip file [MMU-test.zip](https://github.com/openhwgroup/cvw/files/12726991/MMU-test.zip) I have included the test, the reference files and the WLF file from Questa v2022.1 that we are using. Hoping that you could take a look at what is happening and give us your input.

@eroom1966 will send an email to suggest a call to discuss this and work out the next step to resolve.

Thanks

duncangraham-Imperas, 2023-09-27:
Hi @ross144 When running one of our MMU tests we are getting a hang in the table walker. This test appears to be different from the MMU tests we see provided with CVW in that it creates a very large NAPOT region. We wonder if there is a limit in the size for CVW?
Thanks

ross144, 2023-09-27:
Hi @duncangraham-Imperas this sounds like possible real bug. Is this the same test as the MMU-test.zip you linked above?  If so I'm happy to take a closer look.

duncangraham-Imperas, 2023-09-27:
Yes the test provided in the zip file configures the same large NAPOT region

eroom1966, 2023-09-27:
@ross144, please remember you will need Duncans approach described above in order to create the sparse verilog memory

ross144, 2023-09-27:
The problem is the HPTW asks the LSU to fetch address 0x8010_0000; however, the bus returns all x's and the HPTW locks up with x propagation.  The memfile contains data at these address so my first guess is the readmemh did not actually put the data at address 0x8010_0000

ross144, 2023-09-27:
Ok. So the problem is annoying.  The Wally memfiles addressing space is different from the byte address space.  For XLEN=64bit the memfile are 8 bytes wide and use double word (8 byte) addressing.  For XLEN=32bit the addressing is word (4 byte).  This means the addresses in the memfile need to be divided by 8 or 4.  I hacked the memfile and got past the hangup.  However, there appears to be a real failure shortly after.
[MMU-test.tar.gz](https://github.com/openhwgroup/cvw/files/12742594/MMU-test.tar.gz)  The tarball has the hacked up memfile.

eroom1966, 2023-09-28:
@davidharrishmc 
the background to this, is that we have been developing a number of MMU tests, and we want to port them to Wally to increase coverage. this is not specifically demand from customers or users.

duncangraham-Imperas, 2023-09-28:
@ross144 Thanks for the insight into what was happening. I have modified the utility to modify the memfile for loading and I now have most of the MMU tests runnnig and reporting passing on the CVW core.

I am investigating the failures, which appear to fall into a common mode, and will let you know anything interesting I may find.

eroom1966, 2023-09-28:
@davidharrishmc 
I think its possible that this is in violation of the privilege spec, please let me know your thoughts here

The Privileged Specification has these two comments about PMP granularity:
```
Although the PMP mechanism supports regions as small as four bytes, platforms may specify
coarser PMP regions.
```
What this says is that you **don't** have to support the smallest sizes. It does not say that you are allowed **to not** support the largest ones.
```
Software may determine the PMP granularity by writing zero to pmp0cfg, then writing all ones
to pmpaddr0, then reading back pmpaddr0. If G is the index of the least-significant bit set, the
PMP granularity is 2^G+2 bytes.
```
  
This shows you how to find the size of the smallest supported PMP entry. There is no similar algorithm for finding whether larger PMP sizes are supported. The implication is that larger sizes are always supported.

I guess the question is whether you are intending to be compliant or not ?

davidharrishmc, 2023-09-28:
> We support all size PMP. Just 64k

We support all size napot PMP. Just 64kb huge pages on svnapot, which is a
different beast related to virtmem not PMP .

eroom1966, 2023-10-04:
@davidharrishmc @ross144 
Could you please clarify the last two comments.
Do we still think there is either a bug in the existing RTL, or a non implemented part of the specification ?

davidharrishmc, 2023-10-04:
Finally back on my computer with internet.  Sorry I had some trouble following this thread from my phone.  

There are two uses of the term NAPOT.  One is in PMP, and the other is in Svnapot.  PMP has been stable for more than a year and has no known bugs. We believe it should handle all sizes. We implemented Svnapot this summer.  It defines some virtual memory huge pages; we presently only support 64 KiB huge pages per the spec.  We've given it basic testing, but it's a much less mature feature.

At first I thought you were talking about Svnapot, but rereading the thread, I think the issue is an interaction of PMP and the page table walker.

So it sounds like this is a real bug in Wally, and the tarball Rose attached should let us reproduce it and troubleshoot.

ross144, 2023-10-09:
There is indeed a hardware bug here.  It happens with a concurrent I$ miss and DTLB miss.  When the HPTW updates the DTLB it then produces a load page fault.  However, the I$ is still waiting on fetch from main memory.  The trap is taken which messed up the instruction fetch.  I'm a bit surprised we haven't run into this bug yet.  It independent of the PMP.

ross144, 2023-10-09:
The good news is the fix is very simple.  We already have CommittedF from the IFU which delays interrupts.  We can repurpose the same signal to delay exceptions.  This fixes the issue and passes the regression test.  My latest pull request #424 fixes the issue.

ross144, 2023-10-10:
@duncangraham-Imperas Can you close this now that the issue is resolved?

duncangraham-Imperas, 2023-10-11:
All generated MMU tests are now passing on CVW. Thanks

closed 2023-10-11 by duncangraham-Imperas