openhwfoundation/cvw #440: CBOZ extension needs to zero non-cacheable memory

ross144, 2023-10-24:
The cbo.zero instruction is valid for non-cacheable memory. However our implementation zeros cacheable memory.

ross144, 2023-12-26:
Looks like I already added support for this. We just need to write a test for it.

davidharrishmc, 2023-12-28:
I'm not sure that it's worth putting multiple configurations of cacheability into the regression suite.

However, an easy test is to change CacheableRegion to 0 at line 61 of pmachecker.sv, or to set DCACHE_SUPPORTED = 0.

CacheableRegion=0 causes the following regression failures.
rv32gc_arch32a          attempts to read from uninitialized RAM at 80003114, fails immediately on amoadd.w-01 result 1 being x
rv32gc_arch32priv     fails on misalign-jar-01.S at test 1 at 80003114, sim 44, sig 47
rv32gc_arch32d         fails because D on RV32 is not supported without caches
rv32gc_wally32priv    fails on WALLY-cbom (expected because there is no cache).  When cbom is commented out, it has a watchdog timeout on cboz
rv64gc_wally64periph watchdog timeout
rv32gc_arch32d_fma  fails because D on RV32 is not supported without caches

Many other tests seem to hang. They timeout with runtime exceeding 600 seconds:
rv32gc_arch32c
many arch64gc tests

ross144, 2023-12-29:
Fixed with pr #544

closed 2023-12-29 by ross144