openhwfoundation/cvw #451: Modelsim SE simulation issue

junambi, 2023-11-03:
Hi,

I am trying to compile and run Wally with Modelsim SE, i am getting the following error.

# ** Error: (vsim-12001) ../testbench/common/shadowmem.sv(46) : Illegal reference to a parameter below an arrayed instance: testbench.dut.core.lsu.bus.dcache.dcache.CacheWays[0].SRAMLEN
#    Time: 0 ps  Iteration: 0  Region: /testbench/DCacheFlushFSM/genblk1 File: ../testbench/common/shadowmem.sv
# ** Error: (vsim-12001) ../testbench/common/shadowmem.sv(47) : Illegal reference to a parameter below an arrayed instance: testbench.dut.core.lsu.bus.dcache.dcache.CacheWays[0].NUMSRAM
#    Time: 0 ps  Iteration: 0  Region: /testbench/DCacheFlushFSM/genblk1 File: ../testbench/common/shadowmem.sv

Could you please let me know if i am missing something.

Thank you.

davidharrishmc, 2023-11-03:
Nambi,

We have the simulation flow running in Questa. We have not yet Wally in ModelSim SE and don’t have easy access. Do you have access to Questa?

We hope to get Wally running on a wider range of simulators eventually, but won’t get to that for some time. 

David Harris

junambi, 2023-11-04:
David,

I do not have access to Questa, the Modelsim that i am using is a free version from the Intel - Altera website.

In fact, i tried to fix this issue, but ended up crashing Modelsim, since, i am on a tight timeline, i was not able to spend time.

I could look at this, and if i get it running, then i will update you.

Regards,
Nambi

junambi, 2023-11-16:
Closing this, works on Questasim.

closed 2023-11-16 by junambi