openhwfoundation/cvw #452: Simulating wally with Verilator

junambi, 2023-11-03:
Hi

I am trying to simulate wally with verilator script provided, but i am getting the following errors.

%Error: ../testbench/testbench.sv:560:3: Unknown verilator lint message code: 'WIDTHEXPAND', in '/*verilator lint_on WIDTHEXPAND*/'
  560 |   /*verilator lint_on WIDTHEXPAND*/
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        ../config/rv64gc/config.vh:171:24: ... note: In file included from config.vh
        ../testbench/testbench.sv:29:1: ... note: In file included from testbench.sv
%Error: Exiting due to 8 error(s)
Exiting after rv64gc lint due to errors or warnings

The Verilator version that i am using is,

Verilator 5.005 devel rev v5.004-58-g5fce23e90

Thank you.

davidharrishmc, 2023-11-03:
The Verilator script is a work in progress and is not running yet either.  Please use Questa for now.

If you are a sophisticated Verilog user and want to help us get that running, I’d be happy to collaborate.

David

junambi, 2023-11-04:
Hi David,

Yes, i would like to help and contribute, please let me know how we could work on this together.

Thank you,

Regards,
Nambi

davidharrishmc, 2023-11-04:
Nambi,

Getting cvw running on Verilator would be terrific, but we have encountered a number of difficulties.  The tool is evolving rapidly and is not as robust as commercial simulators yet.

I dusted off what we’d started on Verilator long ago and patched it up to point to the current file organization.  It’s part of PR453, which I’m waiting for Rose to accept.  When that PR is accepted, you can see where things stand.

I’m using Verilator 5.011
***@***.***:~/cvw/sim$ verilator --version
Verilator 5.011 devel rev v5.010-114-g4b7b185d0

The latest Verilator crashes when linting cvw.  I filed issue 4660 and they seem to be looking at it and have isolated which commit broke things. Until they fix this, you’ll need to be sure you are using a version of Verilator that is fairly recent (to support SystemVerilog testbenches) but old enough to not have this crash.  I would assume that it could be months before that is solved.

When I go to $WALLY/sim and run “verilate”, I’m presently getting 10 errors about dotted scope/variable. It’s not clear to me yet why these errors come up.  For example, on loggers.sv at 220, it appears Predictor should be in the hierarchy and I’m not sure why Verilator is complaining. In testbench.sv at 338, Verilator is complaining about dut.core.ifu.bpred.bpred.PHT but the code in the file refers to dut.core.ifu.bpred.bpred.Predictor.DirPredictor.PHT.mem, which is a different path.

I’m not sure at this point whether there is a real bug in the testbench, or whether this is a bug in Verilator or in Verilator’s error reporting (printing an incorrect path related to line 338).  The testbench does run with Questa.

This might involve running a debugger on Verilator and looking at its understanding of the cvw hierarchy and its error reporting to isolate what is happening, and either find the root cause in cvw or patch the bug in Verilator. That’s beyond my skill level to do quickly. Maybe you have ideas of an easier way to approach this.  If you have skills to address this, we would greatly appreciate your contributions.  

Could you share what organization you are from and what interests you have in CORE-V Wally?

David

v64gc simulating...
%Error: ../testbench/common/loggers.sv:220:103: Can't find definition of 'Predictor' in dotted scope/variable: 'dut.core.ifu.bpred.bpred.Predictor'
  220 |       flopenrc #(1) PCSrcMReg(clk, reset, dut.core.FlushM, ~dut.core.StallM, dut.core.ifu.bpred.bpred.Predictor.DirPredictor.PCSrcE, PCSrcM);
      |                                                                                                       ^~~~~~~~~

%Error: ../testbench/testbench.sv:338:59: Can't find definition of 'PHT' in dotted scope/variable: 'dut.core.ifu.bpred.bpred.PHT'
  338 |           dut.core.ifu.bpred.bpred.Predictor.DirPredictor.PHT.mem[adrindex] = 0;
      |                                                           ^~~
        ../config/rv64gc/config.vh:176:24: ... note: In file included from config.vh
        ../testbench/testbench.sv:29:1: ... note: In file included from testbench.sv
        ... Known scopes under 'dut.core.ifu.bpred.bpred': PCSrcMReg, icpred, pcmux1
%Error: Exiting due to 10 error(s)
Exiting after rv64gc lint due to errors or warnings

davidharrishmc, 2023-12-29:
Closed.  Merged with Issue 335.

closed 2023-12-29 by davidharrishmc