openhwfoundation/cvw #457: Bugs in coverage tests

davidharrishmc, 2023-11-08:
Coverage tests are presently failing on ifu, leu, flu, csrwrites, tlbNAPOT.

cd tests/coverage
iter-elf.bash --search .

ifu fails because Zcb is not enabled in imperas.ic
csrwrites fails because Svadu is not enabled in imperas.ic

fpu fails because write of -1 to stimecmp doesn't change stimecmp
``
# Info (RISCV_IS) CPU 'refRoot/cpu' 0x80000072 14d29073 csrw    stimecmp,x5:  MIP:00000080->00000000
# Info   stimecmp 0000000000000000 -> ffffffffffffffff
# Info (IDV) Instruction executed prior to mismatch '0x80000072(interrupt+10): 14d29073 csrw    stimecmp,x5'
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x0000000080000072 interrupt+10):
# Error (IDV) Mismatch 0> CSR 14d (stimecmp)
# Error (IDV)   . dut:0x0000000000000000 (not updated)
# Error (IDV)   . ref:0xffffffffffffffff
# Error (IDV) testbench.idv_trace2api.state_compare @ 14330: MISMATCH
``

ieu fails because an amoadd at 80003082 produces a load access fault when it should produce a store/AMO access fault
``
 Info 552: 'refRoot/cpu', 0x0000000080003082(main+82): Machine 0000202f amoadd.w x0,x0,(x0)
# Info   MEMX 0x80003082 0x80003082 2 202f
# Info   MEMX 0x80003084 0x80003084 2 0000
# Info (OP_NCH) GlobalTime:0.000000 LocalTime:0.000006 Net:refRoot/coverpoint 0 => 33
# Info (OP_NCH) GlobalTime:0.000000 LocalTime:0.000006 Net:refRoot/coverpoint 33 => 35
# Info (OP_NCH) GlobalTime:0.000000 LocalTime:0.000006 Net:refRoot/coverpoint 35 => 41
# Info (OP_NCH) GlobalTime:0.000000 LocalTime:0.000006 Net:refRoot/coverpoint 41 => 49
# Info (OP_NCH) GlobalTime:0.000000 LocalTime:0.000006 Net:refRoot/coverpoint 49 => 98
# Info (RISCV_PMP) CPU 'refRoot/cpu': PMP PRIV=rwx 0x00000000:0x3fffffffb (mode Machine)
# Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x80003082 0000202f amoadd.w x0,x0,(x0): No access permission for store/atomic memory operation (0x0)
# Info   mstatus 0000000a00000088 -> 0000000a00001880 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0->3 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:1->0 SIE:0]
# Info   mcause 0000000000000002 -> 0000000000000007 [Interrupt:0 Code:2(Illegal instruction)->7(Store/AMO access fault)]
# Info   mtval 000000006030101b -> 0000000000000000
# Info 553: 'refRoot/cpu', 0x0000000080000050(trap_handler): Machine 34021273 csrrw   x4,mscratch,x4
# Info   MEMX 0x80000050 0x80000050 2 1273
# Info   MEMX 0x80000052 0x80000052 2 3402
# Info   x4 0000000000000000 -> 0000000080002400
# Info   mscratch 0000000080002400 -> 0000000000000000
# Info (IDV) Instruction executed prior to mismatch '0x80000050(trap_handler+0): 34021273 csrrw   x4,mscratch,x4'
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x0000000080000050 trap_handler+0):
# Error (IDV) Mismatch 0> CSR 342 (mcause)
# Error (IDV)   . dut:0x0000000000000005 Interrupt:0 Code:5(Load access fault)
# Error (IDV)   . ref:0x0000000000000007 Interrupt:0 Code:7(Store/AMO access fault)
# 
``

tlbNAPOT page faults in ImperasDV but not in Wally.
``
# Warning (RISCV_PTWE) CPU 'refRoot/cpu': Page table entry illegal Svnapot entry [address=0xc0215240 PTEAddress=0x800150a8 access=W]
# Info (OP_NCH) GlobalTime:0.000000 LocalTime:0.000006 Net:refRoot/coverpoint 23 => 1
# Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x80019068 01e2a023 sw      x30,0(x5): Page fault at store/atomic memory operation address (0xc0215240)
# Info   mstatus 0000000a00000088 -> 0000000a00000880 [SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0->1 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:1->0 SIE:0]
# Info   mepc 000000008001901e -> 0000000080019068
# Info   mcause 000000000000000b -> 000000000000000f [Interrupt:0 Code:11(Environment call from M-mode)->15(Store/AMO page fault)]
# Info   mtval 0000000000000000 -> 00000000c0215240
# Info 620: 'refRoot/cpu', 0x0000000080000050(trap_handler): Machine 34021273 csrrw   x4,mscratch,x4
# Info   MEMX 0x80000050 0x80000050 2 1273
# Info   MEMX 0x80000052 0x80000052 2 3402
# Info   x4 0000000000000000 -> 0000000080018400
# Info   mscratch 0000000080018400 -> 0000000000000000
# Info (IDV) Instruction executed prior to mismatch '0x80019064(loop+0): 01c3dd63 bge     x7,x28,8001907e'
# Error (IDV) PC mismatch (HartId:0, PC:0x0000000080000050 trap_handler+0):
# Error (IDV) Mismatch 0>
# Error (IDV)   . dut:0x0000000080019068 loop+4
# Error (IDV)   . ref:0x0000000080000050 trap_handler+0
``

davidharrishmc, 2023-11-09:
ImperasDV does not yet implement the menvcfg.HADE bit of Svadu.  Wait for that feature to complete and then revisit csrwrites.

ifu now passing with cpu/Zcb=T flag in imperas.ic

eroom1966, 2023-12-08:
Hi David,
can you provide a link to the spec you are referring to ?
The ratified spec we think is menvcfg.ADUE bit[61] 
The model had this implemented on 9/Nov

Q. What is the version you are running - this is printed in the log

From this issue/thread we are confused which issues you think are RTL bugs, and which you think are REF bugs.
Could I ask you to split this into one issue per bug, and assign the ones to me which you believe are bugs with the REF model
In each issue describe how to reproduce your findings described above

Thx
Lee

davidharrishmc, 2023-12-14:
Lee,

Yes, I see it is now ADUE.  We had implemented the feature before it was ratified, and it was named HADE at the time. 

I’ve solved and close the issue, and opened a new more specific issue 530 that I’d like your feedback on regarding honoring menvcfg.STCE.

https://github.com/openhwgroup/cvw/issues/530

David

closed 2023-12-21 by davidharrishmc

davidharrishmc, 2023-12-21:
Fixed with PR 540