openhwfoundation/cvw #462: WALLY privileged tests failing in ImperasDV

davidharrishmc, 2023-11-10:
The following tests are failing in ImperasDV:

WALLY-spi
WALLY-mmu-sv39-svadu-svnapot-svpbmt
WALLY-mmu-sv48 (hangs)

I have added an SPI entry to the PMA map, but don't understand memory-mapped I/O checking in ImperasDV well, so perhaps this is a root cause.

Reproduce SPI with
run-elf.bash --elf ~/cvw/tests/riscof/work/wally-riscv-arch-test/rv64i_m/privilege/src/WALLY-spi-01.S/ref/ref.elf --verbose

The SPI failure occurs on a read from x28 = 10040000 (SPI sckdiv).  The DUT returns 3, which is the reset value, while the ref returns 0.  I'm not sure how the ref knows what it should return.

```
# Info 276: 'refRoot/cpu', 0x0000000080000878(read32_test+8): Machine 000e2383 lw      x7,0(x28)
# Info   MEMX 0x80000878 0x80000878 2 2383
# Info   MEMX 0x8000087a 0x8000087a 2 000e
# Info   MEMR 0x10040000 0x10040000 4 00000000
# Info   x7 0000000000000bad -> 0000000000000000
# Info (IDV) Instruction executed prior to mismatch '0x80000878(read32_test+8): 000e2383 lw      x7,0(x28)'
# Error (IDV) GPR register value mismatch (HartId:0, PC:0x0000000080000878 read32_test+8):
# Error (IDV) Mismatch 0> GPR x7
# Error (IDV)   . dut:0x0000000000000003
# Error (IDV)   . ref:0x0000000000000000
# Error (IDV) testbench.idv_trace2api.state_compare @ 5290: MISMATCH
```

eroom1966, 2023-11-15:
@davidharrishmc @ross144 
I am blocked by the issue when trying to run

git clone https://github.com/openhwgroup/cvw --recurse-submodules

resulting in
fatal: remote error: upload-pack: not our ref 2c5675d7a58e98d47bef3a6cf5a8373397b0d0be
fatal: The remote end hung up unexpectedly
Fetched in submodule path 'addins/riscv-arch-test', but it did not contain 2c5675d7a58e98d47bef3a6cf5a8373397b0d0be. Direct fetching of that commit failed.

davidharrishmc, 2023-11-15:
@eroom1966 I removed riscv-arch-test and added it back, and now it downloads for me and compiles.  Maybe the pointer to an old commit had broken.  Let me know if you have issues now.

davidharrishmc, 2023-11-21:
@eroom1966 Lee, did you get to look at this again?

eroom1966, 2023-12-07:
As discussed on call - use rvviRegfMemorySetVolatile

davidharrishmc, 2023-12-07:
WALLY-spi is now working per Lee's explanation of adding SPI to the test bench.

eroom1966, 2023-12-08:
Hi Davide - can we close this issue ?

davidharrishmc, 2023-12-14:
SPI is closed.  Opening new issues for other failing tests.

closed 2023-12-14 by davidharrishmc