openhwfoundation/cvw #507: test float fails on sqrt only when in all mode

davidharrishmc, 2023-11-24:
sim-testfloat-batch-all fails on sqrt but sim-testfloat-batch sqrt passes.  Seems like a testfloat test bench issue.

$ cd $WALLY/sim
$ ./sim-testfloat-batch-all
...
 Running f128_div_rnm.tv vectors
# Running f128_sqrt_rne.tv vectors
# 
# Error in f128_sqrt_rne.tv
# TestNum        105 OpCtrl x
# inputs: 00000000000000000000000000000000 4006ffbffffffffffffeffffffffffff 00000000000000000000000000000000
# SrcA: xxxxxxxxxxxxxxxx
#  Res: 00000000000000000000000000000000 00
#  Expected: 4002ffdffeffeffebfe37d57bd39374a xx
# ** Note: $stop    : ../testbench/testbench-fp.sv(961)
#    Time: 1050014245 ns  Iteration: 1  Instance: /testbenchfp
# Break in Module testbenchfp at ../testbench/testbench-fp.sv line 961
# Stopped at ../testbench/testbench-fp.sv line 961

$ ./sim-testfloat-batch sqrt
...
# Running f128_sqrt_rne.tv vectors 
# Running f128_sqrt_rz.tv vectors
...

stineje, 2023-12-15:
Bug is a messy one and still no solution:  if done separately works correctly separately for DIV and SQRT but fails on all - output below prints out OpCtrl and you can see for "all" it does not distinguish OpCtrl.  The problem is due to if(OpCtrl[0])  (L1112) is not getting evaluated correctly when all is utilized.  Thought had solution by independently telling it DIV or SQRT but then it started failing again.  I am leaning towards thinking that the configuration is causing the issue; however, I am determine to find the bug.   Still working on it - apologies for the delay.  If needed, I will re-write the test bench portion for div/sqrt with testbench-fp.sv

SQRT
-----
# Running f128_sqrt_rne.tv vectors 
# OpCtrl SQRT = 001
# Running f128_sqrt_rz.tv vectors
# OpCtrl SQRT = 001
# Running f128_sqrt_ru.tv vectors
# OpCtrl SQRT = 001
# Running f128_sqrt_rd.tv vectors
# OpCtrl SQRT = 001
# Running f128_sqrt_rnm.tv vectors
# OpCtrl SQRT = 001

DIV
----
# Running f128_div_rne.tv vectors 
# OpCtrl DIV = 000
# Running f128_div_rz.tv vectors
# OpCtrl DIV = 000
# Running f128_div_ru.tv vectors
# OpCtrl DIV = 000
# Running f128_div_rd.tv vectors
# OpCtrl DIV = 000
# Running f128_div_rnm.tv vectors
# OpCtrl DIV = 000

all
----
# Running f128_div_rne.tv vectors
# OpCtrl DIV = 000
# Running f128_div_rz.tv vectors
# OpCtrl DIV = 000
# Running f128_div_ru.tv vectors
# OpCtrl DIV = 000
# Running f128_div_rd.tv vectors
# OpCtrl DIV = 000
# Running f128_div_rnm.tv vectors
# OpCtrl DIV = 000
# Running f128_sqrt_rne.tv vectors
# OpCtrl DIV = 001
# 
# Error in f128_sqrt_rne.tv

stineje, 2023-12-15:
Fixed and complete

closed 2023-12-18 by davidharrishmc