openhwfoundation/cvw #510: Priority of Interrupts and Exceptions

ross144, 2023-11-29:
@eroom1966 @duncangraham-Imperas
ImperasDV mismatches with Wally when there is a pending interrupt and an instruction page fault.  ImperasDV takes the page fault, but Wally takes the interrupt.  I don't see in the specification where it defines the correct priority of interrupts over exceptions or vice versa.

Wally priorities interrupts and ImperasDV priorities exceptions.  I'm not sure if this is actually a real bug.  Is there away to configure ImperasDV to prioritize interrupts? 

```
# Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x7fff85fdc8a8 fffff7b7 lui     x15,0xfffff: Page fault at fetch address (0x7fff85fdc8a8)
# Info (IDV) Instruction executed prior to mismatch '0xffffffff8000308c(): 14021273 csrrw   x4,sscratch,x4'
# Error (IDV) CSR register value mismatch (HartId:0, PC:0xffffffff8000308c ):
# Error (IDV) Mismatch 0> CSR 142 (scause)
# Error (IDV)   . dut:0x8000000000000005 Interrupt:1 Code:5(Supervisor timer interrupt)
# Error (IDV)   . ref:0x000000000000000c Interrupt:0 Code:12(Instruction page fault)
# Error (IDV) Mismatch 1> CSR 143 (stval)
# Error (IDV)   . dut:0x0000000000000000
# Error (IDV)   . ref:0x00007fff85fdc8a8
# Error (IDV) testbench.idv_trace2api.state_compare @ 9896207100: MISMATCH
```

duncangraham-Imperas, 2023-12-07:
Hi, sorry for such a slow response. I put it on my list to look at .. and well it got a bit lost amongst everything else!

The behaviour you are seeing in the RTL is the same as we would expect in the reference model. We think the issue is caused because the reference model is not being told about the interrupt (in time) and so it is not being seen as active and not evaluated. Could you check when the interrupt is being reported over the RVVI interface. Is it possible to run the same test with RVVI tracing turned on (set IDV_TRACE2LOG=1); if it is after many instructions the tracing can be deferred by setting IDV_TRACE2LOG_AFTER=<number of instructions>

eroom1966, 2023-12-08:
Following our call yesterday, the interrupt should be taken before the exception.
The fact that we are not seeing a number of scenarios evaluated, implies that the stimer interrupt has not been delivered to the IDV model prior to the exception occurring.

I think we need the tracing enabled (as Duncan described) - also please evaluate your logic for delivering the stimer interrupt.
Just looking at the path I am naively concerned that the node you are using to pass to IDV is already conditioned in your logic, and is therefore delivered 'after the fact'

Tracing will describe the net changes being passed - please make sure you enable 
--monitornets

ross144, 2023-12-26:
I reran the Linux boot with logging so now we have the wlf file and the ImperasDV trace.  Below is the relevant ImperasDV output.  Wally's implementation follows the old spec and prioritizes interrupts over exceptions.  The new spec is undefined. 

The ImperasDV trace shows STimerInterrupt's net change prior sret's retirement.  If we change the Wally implementation to prioritize exceptions then ImperasDV is happy.  I think one of two things is going on.  1. Either ImperasDV does not support this type of priority or is misconfigured or we are not correctly informing ImperasDV about the processor state.  2.  Looking at the waveform we see the timer interrupt arrives after sret is in the Memory stage.  The processor is stalled on a TLB miss and a branch miss prediction. During the stall the interrupt rises.  A few cycles later the stall finishes and the sret retires.  Is there any chance this timing is screwing with the sequence of events reported to ImperasDV?  I doubt this is the case since the ImperasDV trace shows the STimerInterrupt net change prior sret retirement.

```
 Info (IDV) testbench.idv_trace2log.process_event @ 9896206780: RET,0,491410108,ffffffff800031c4,"7fee     c.ldsp  x31,248(x2)    ",x31=3c00000040001001,,,CSRb00(mcycle)=000000003afc69c1 CSRb02(minstret)=000000001d4a52bc,
# Info (IDV) testbench.idv_trace2log.process_event @ 9896206790: RET,0,491410109,ffffffff800031c6,"6142     c.ldsp  x2,16(x2)      ",x2=00007ffffacb44b0,,,CSRb00(mcycle)=000000003afc69c2 CSRb02(minstret)=000000001d4a52bd,
# Info (IDV) NET,989620699,STimerInterrupt,1
# Info (IDV) Net change 'STimerInterrupt' = 0x1 added to queue
# Info (IDV) testbench.idv_trace2log.process_event @ 9896207020: RET,0,491410110,ffffffff800031c8,"10200073 sret                   ",,,,CSR100(sstatus)=0000000200004022 CSR144(sip)=0000000000000020 CSR300(mstatus)=0000000a000040a2 CSR344(mip)=0000000000000020 CSRb00(mcycle)=000000003afc69d9 CSRb02(minstret)=000000001d4a52be,
# Info (IDV) testbench.idv_trace2log.process_event @ 9896207100: RET,0,491410111,ffffffff8000308c,"14021273 csrrw   x4,sscratch,x4 ",x4=ffffaf8007691300,,,CSR100(sstatus)=0000000200004020 CSR140(sscratch)=00007fff8608c3c0 CSR142(scause)=8000000000000005 CSR143(stval)=0000000000000000 CSR300(mstatus)=0000000a000040a0 CSRb00(mcycle)=000000003afc69e1 CSRb02(minstret)=000000001d4a52bf,
# Info (OP_NWR) GlobalTime:0.000000 LocalTime:4.914327 Net:refRoot/coverpoint 98 => 0
# Info (OP_NWR) GlobalTime:0.000000 LocalTime:4.914327 Net:refRoot/coverpoint 0 => 4
# Info (OP_NWR) GlobalTime:0.000000 LocalTime:4.914327 Net:refRoot/coverpoint 4 => 2
# Info (OP_NWR) GlobalTime:0.000000 LocalTime:4.914327 Net:refRoot/coverpoint 2 => 29
# Info (OP_NWR) GlobalTime:0.000000 LocalTime:4.914327 Net:refRoot/coverpoint 29 => 28
# Info (OP_NWR) GlobalTime:0.000000 LocalTime:4.914327 Net:refRoot/coverpoint 28 => 27
# Info (OP_NWR) GlobalTime:0.000000 LocalTime:4.914327 Net:refRoot/coverpoint 27 => 12
# Info (RISCV_PTWE) CPU 'refRoot/cpu': Page table entry V=0 [address=0x7fff85fdc8a8 PTEAddress=0x876d1178 access=X]
# Info (OP_NWR) GlobalTime:0.000000 LocalTime:4.914327 Net:refRoot/coverpoint 12 => 1
# Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x7fff85fdc8a8 fffff7b7 lui     x15,0xfffff: Page fault at fetch address (0x7fff85fdc8a8)
# Info (OP_NWR) GlobalTime:0.000000 LocalTime:4.914327 Net:refRoot/readcsr 983296 => 262464
# Info (IDV) Instruction executed prior to mismatch '0xffffffff8000308c(): 14021273 csrrw   x4,sscratch,x4'
# Error (IDV) CSR register value mismatch (HartId:0, PC:0xffffffff8000308c ):
# Error (IDV) Mismatch 0> CSR 142 (scause)
# Error (IDV)   . dut:0x8000000000000005 Interrupt:1 Code:5(Supervisor timer interrupt)
# Error (IDV)   . ref:0x000000000000000c Interrupt:0 Code:12(Instruction page fault)
# Error (IDV) Mismatch 1> CSR 143 (stval)
# Error (IDV)   . dut:0x0000000000000000
# Error (IDV)   . ref:0x00007fff85fdc8a8
# Error (IDV) testbench.idv_trace2api.state_compare @ 9896207100: MISMATCH
```

davidharrishmc, 2023-12-26:
I don't understand why ImperasDV is behaving like this.
As I read the trace a supervisor timer interrupt is pending before the sret.
When sret returns and reenable interrupts, the DUT then takes the interrupt.
ImperasDV fetches the next instruction, which causes a page fault.  It takes this page fault rather than the interrupt.
If the next instruction didn't fault, I'm pretty certain an interrupt would be taken, rather than a page fault.  It doesn't seem logical that the page fault should occur rather than the interrupt.
This seems like either a bug in ImperasDV or perhaps something not communicating right in the tracer so that ImperasDV doesn't take the pending interrupt.
@eroom1966 any thoughts?

eroom1966, 2024-01-05:
I think we know why this is happening now.
We are not correctly evaluating the STimer Interrupt 
I know it takes a long time to run so we are going to try and get a branch update, if so can you run this over the weekend ?

ross144, 2024-01-05:
Hi Lee,

I will definitely run over the weekend with the update.

Rose

ross144, 2024-01-05:
> Hi Rose,
> We have uploaded a build on the ImperasDV-OpenHW repository, a new branch
> called 'beta', can you try this ?

Hi Lee,

I have the simulation running the updated ImperasDV-OpenHW repo.  It should
finish sometime on sunday.

Rose

closed 2024-01-08 by ross144