openhwfoundation/cvw #518: Can no longer clone repository

eroom1966, 2023-12-08:
@davidharrishmc @ross144 

Hi David, Rose, 
I have been having an issue for a while trying to clone your repo, initially there was a missing hash (which David fixed)
The issue I now have is related to this submodule
Cloning into '/home/moore/git/eroom1966/openhwgroup/cvw/cvw/addins/ahbsdc'...

I run the following command
git clone https://github.com/openhwgroup/cvw --recurse-submodules

Everything runs fine until it gets to this submodule and I get

> 
> Cloning into '/home/moore/git/eroom1966/openhwgroup/cvw/cvw/addins/ahbsdc'...
> @@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
> @    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
> @@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
> IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
> Someone could be eavesdropping on you right now (man-in-the-middle attack)!
> It is also possible that a host key has just been changed.
> The fingerprint for the RSA key sent by the remote host is
> SHA256:********************************************************.
> Please contact your system administrator.
> Add correct host key in /home/moore/.ssh/known_hosts to get rid of this message.
> Offending RSA key in /home/moore/.ssh/known_hosts:119
>   remove with:
>   ssh-keygen -f "/home/moore/.ssh/known_hosts" -R "github.com"
> RSA host key for github.com has changed and you have requested strict checking.
> Host key verification failed.
> fatal: Could not read from remote repository.
> 
> Please make sure you have the correct access rights
> and the repository exists.
> fatal: clone of 'git@github.com:jacobpease/ahbsdc.git' into submodule path '/home/moore/git/eroom1966/openhwgroup/cvw/cvw/addins/ahbsdc' failed
> Failed to clone 'addins/ahbsdc' a second time, aborting

I am completely blocked running anything associated with cvw

eroom1966, 2023-12-08:
By the way, the same thing happens in my fork version of cvw

eroom1966, 2023-12-08:
I notice that ALL of the gitmodules are referenced using a URL of the same form EXCEPT this one causing the issue
all others 
url = https://github.com/<account>/<project>/

the one causing the issue
url = git@github.com:jacobpease/ahbsdc.git

ross144, 2023-12-11:
Very interesting. I do not have this problem. I will change the url from git@ to https and hopefully it resolves the issue.

ross144, 2023-12-11:
I believe this is fixed with PR #519.

infinitymdm, 2023-12-26:
I certainly didn't have this issue when running the same clone command. I'd guess this is fixed.

davidharrishmc, 2023-12-26:
Closing this.  Please reopen if somebody else has a problem.

closed 2023-12-26 by davidharrishmc