openhwfoundation/cvw #541: testbench-fp does not check CVTINT or CMP

davidharrishmc, 2023-12-25:
At line 952, testbench-fp excludes checking CVTINT or CMP.

assign CheckNow = ((DivDone | ~divsqrtop) | (TEST == "add" | TEST == "fma" | TEST == "sub")) & (UnitVal !== `CVTINTUNIT) & (UnitVal !== `CMPUNIT);

CVTINT code later checks the flags but not the result.  There is no CMP check at all.

As a consequence, it is possible to change a bit in the expected answer (e.g. tests/fp/vectors/ui32_to_f128_rne.tv or f128_eq_rne.tv) and have the mismatch go undetected.

davidharrishmc, 2024-01-09:
I changed fp/vectors/ui32_to_f128_rne.tv to add ones near the right of lines 4 and 5, and sim-testfloat-batch cvtint still passed.

1FEFFFEF_401BFEFFFEF000000000000000000000_00
00000000_00000000000000000000000000000000_00
0000448E_400D1238000000000000000000000000_00
00000001_3FFF00000000000000000000000000**1**0_00
007FFFCD_4015FFFF340000000000000000000000_0**1**

stineje, 2024-01-09:
I did a full run on this and thought was testing everything.  Apologies - checking this now.

closed 2024-01-15 by davidharrishmc