openhwfoundation/cvw #546: ImperasDV does not raise page fault on non-leaf PTE with PBMT field != 0

davidharrishmc, 2024-01-02:
According to the privileged spec, a non-leaf PTE must have the PBMT bits = 0 or else raise a page fault.  ImperasDV does not seem to throw this fault.

"For non-leaf PTEs, bits 62–61 are reserved for future standard use. Until their use is defined by a standard extension, they must be cleared by software for forward compatibility, or else a page-fault exception is raised." RISC-V privileged spec at page 93.

See the attached tlbmisc.elf (remove .txt suffix) and tlbmisc.log.  At PC=80016068, the lw attempts to read from 80600000.  The hardware page table walker first fetches a PTE from 80010000 of 0x00000000200044C1 (pointing to 0x80001100).  It then fetches from 800011010 of 0x00000000200048C1 (pointing to 0x80001200).  It then fetches from 80012018 of 0x4000000020004CC1.  This non leaf PTE has bit 62 = 1, so it should throw a page fault, but it doesn't in ImperasDV. See line 1459 of the log file.

[tlbmisc.log](https://github.com/openhwgroup/cvw/files/13807876/tlbmisc.log)
[tlbmisc.elf.txt](https://github.com/openhwgroup/cvw/files/13807879/tlbmisc.elf.txt)

davidharrishmc, 2024-01-02:
Also the Wally HPTW has the same bug of not checking this.

eroom1966, 2024-01-03:
Hi David
Firstly - Happy New Year

Thanks for the testcase.
Can you please provide a set of instructions how to reproduce ?

I am guessing I simply clone the repo and run a command ?

Thx
Lee

davidharrishmc, 2024-01-03:
Yes, (working from memory because power is off at my campus)
clone the repo
run-elf.bash --elf tlbmisc.elf --verbose
look at the log at PC = 80016068

eroom1966, 2024-01-05:
Hi David,
Can you try the beta branch in ImperasDV-OpenHW, I think this is fixed
Thx

ross144, 2024-01-05:
I confirmed the updated ImperasDV-OpenHW beta branch detects the non-lef PBMT bits and raises a page fault.  Wally still contains the fault.  I will patch Wally today.

ross144, 2024-01-05:
I fixed Wally with PR #549.

closed 2024-01-06 by davidharrishmc