openhwfoundation/cvw #547: HPTW malfunctions during CBO instruction

davidharrishmc, 2024-01-02:
The tlbmisc coverage test caused a HPTW malfunction when a CBO instruction caused a TLB miss.  The CMO instruction CacheCMOpM is asserted during the page table walk, causing the wrong page table entry to be written into the TLB.

Rose solved this in lsu.sv

-        assign CacheCMOpM = CacheableM ? CMOpM : '0;
+        assign CacheCMOpM = (CacheableM & ~SelHPTW) ? CMOpM : '0;

davidharrishmc, 2024-01-02:
Issue just opened for tracking purposes.  Closing it.

This is a real logic bug that was discovered through coverage testing of the InvalidCBOM line of tlbcontrol.sv.

davidharrishmc, 2024-01-03:
There is a new logic bug after this fix.  Now tlbmisc is not producing a Store page fault during the cbo.zero at 0x800170ac. This is causing a mismatch with ImperasDV.

```

run-elf.bash --elf tlbmisc.elf --verbose > tlbmisc.log

# Info (IDV) testbench.idv_trace2log.process_event @ 14460: RET,0,365,800170a2,"000812b7 lui     x5,0x81        ",x5=0000000000081000,,,CSRb00(mcycle)=00000000000005a1 CSRb02(minstret)=000000000000016d,
# Info (IDV) testbench.idv_trace2log.process_event @ 14470: RET,0,366,800170a6,"8012829b addiw   x5,x5,-2047    ",x5=0000000000080801,,,CSRb00(mcycle)=00000000000005a2 CSRb02(minstret)=000000000000016e,
# Info (IDV) testbench.idv_trace2log.process_event @ 14970: RET,0,367,800170aa,"02b2     c.slli  x5,0xc         ",x5=0000000080801000,,,CSRb00(mcycle)=00000000000005d4 CSRb02(minstret)=000000000000016f,
# Info (IDV) testbench.idv_trace2log.process_event @ 14980: RET,0,368,800170ac,"0042a00f cbo.zero x5            ",,,,CSRb00(mcycle)=00000000000005d5 CSRb02(minstret)=0000000000000170,
# Info (RISCV_PTWE) CPU 'refRoot/cpu': Page table entry does not allow access [address=0x80801000 PTEAddress=0x80015008 access=W]
# Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x800170ac 0042a00f cbo.zero x5: Page fault at store/atomic memory operation address (0x80801000)
# Info (IDV) Instruction executed prior to mismatch '0x800170aa(main+aa): 02b2     c.slli  x5,0xc'
# Error (IDV) PC mismatch (HartId:0, PC:0x0000000080000050 trap_handler+0):
# Error (IDV) Mismatch 0>
# Error (IDV)   . dut:0x00000000800170ac main+ac
# Error (IDV)   . ref:0x0000000080000050 trap_handler+0
# Error (IDV) Insn. bit pattern mismatch (HartId:0, PC:0x0000000080000050 trap_handler+0):
# Error (IDV) Mismatch 1>
# Error (IDV)   . dut:0042a00f cbo.zero x5
# Error (IDV)   . ref:34021273 csrrw   x4,mscratch,x4
# Error (IDV) GPR register value mismatch (HartId:0, PC:0x0000000080000050 trap_handler+0):
# Error (IDV) Mismatch 2> GPR x4
# Error (IDV)   . dut:0x0000000000000000 (not updated)
# Error (IDV)   . ref:0x0000000080016400
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x0000000080000050 trap_handler+0):
# Error (IDV) Mismatch 3> CSR 300 (mstatus)
# Error (IDV)   . dut:0x0000000a00000088 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:1 SIE:0 (not updated)
# Error (IDV)   . ref:0x0000000a00000880 SD:0 MBE:0 SBE:0 SXL:2 UXL:2 TSR:0 TW:0 TVM:0 MXR:0 SUM:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:1 VS:0(Off) SPP:0 MPIE:1 UBE:0 SPIE:0 MIE:0 SIE:0
# Error (IDV) Mismatch 4> CSR 340 (mscratch)
# Error (IDV)   . dut:0x0000000080016400 (not updated)
# Error (IDV)   . ref:0x0000000000000000
# Error (IDV) Mismatch 5> CSR 341 (mepc)
# Error (IDV)   . dut:0x00000000800171b4 (not updated)
# Error (IDV)   . ref:0x00000000800170ac
# Error (IDV) Mismatch 6> CSR 342 (mcause)
# Error (IDV)   . dut:0x000000000000000b Interrupt:0 Code:11(Environment call from M-mode) (not updated)
# Error (IDV)   . ref:0x000000000000000f Interrupt:0 Code:15(Store/AMO page fault)
# Error (IDV) Mismatch 7> CSR 343 (mtval)
```

ross144, 2024-01-07:
This is fixed with a combination of mine and David's changes.  I believe we can close now.

closed 2024-01-07 by davidharrishmc