openhwfoundation/cvw #569: Python package not found for debian while building docker image from dockerfile

Sanket-0510, 2024-01-13:
@ross144  I was trying to setup the project, I tried doing native setup but it failed on my PC, hence tried building it using dockerfile. While building the image there  seems to be a small fix needed in dockerfile, as the source is debian the docker build is failing cause of missing python package. The python needs to be replaced with python3 it seems.
[image]

How to reproduce the error:
    1. clone the repo.
    2. `cd cvw/docs`
    3. try to build the docker image `docker build -t <image_name> .`
    4. eg. `docker build -t cvw .`
   
The image build will fail and will throw python package missing error while building the image.

How to fix:  

[image]

Replace `python` with `python3` instead.

Sanket-0510, 2024-01-13:
@ross144 I have opened a pull request for this small fix. Please review the issue and PR and guide me if I am at right track or not.

davidharrishmc, 2024-01-15:
PR merged.

closed 2024-01-15 by davidharrishmc

ross144, 2024-01-15:
@Sanket-0510 Thanks for updating the dockerfile. What was the failure you had on native install?  And what platform are you running, ubuntu? version?

Sanket-0510, 2024-01-16:
> @Sanket-0510 Thanks for updating the dockerfile. What was the failure you had on native install? And what platform are you running, ubuntu? version?

While installing EDA tools I faced the python package issue cause the .sh file tries to install python packages on the same system and my ubuntu does not support externally managed packages, I tried adding virtual env in .sh file but still getting error, ubuntu version is 23.04 [
[image]
](url)

got this error whie trying to install EDA tools with command `sudo $WALLY/bin/wally-tool-chain-install.sh`

stineje, 2024-01-16:
Hi Sanket,

Before installing, I make sure I have a version of Ubuntu that is upgradable by updates.  Since Ubuntu changes frequently, sometimes it goes out of date and I have to do an upgrade.  

However, 23.04 sounds like a fairly good version.  I am not sure it helps, but I try “sudo ./wally-tool-chain-install.sh” within the bin directory.  This shouldn’t affect the command you typed, but can you 1.) update all packages and then 2.) run the install script again.  

To update my Ubuntu version, I type “sudo apt update && sudo apt upgrade -y”.  Let us know and keep us updated.  Take care.

All my best,

James

Sanket-0510, 2024-01-17:
@ross144 This is Sanket, 3rd year undergraduate from IIIT NAGPUR India I am interested in applying for LFX mentorship program & project. I have started preparing the proposal for the same. I am pursuing my engineering in Electronics Communication, hence I have worked with HDL and VLSI during my course period and I also have great hands on experience with Python, C and shell. All this scenario I think perfectly matches with the CVW'S project. I would like to connect with the mentors for further guidance regarding proposal and understanding of the project. It would be great if someone would share the communication platform for CVW's repository and project. Also it would be great if you can share the resources to get a better understanding for the same. Thank you so much.

Regards,
Sanket.