openhwfoundation/cvw #582: Imperas error when setting FIOM bit of menvcfg

jordancarlin, 2024-01-19:
When running `run-elf.bash --elf csrwrites.elf` the error message below occurs. It appears that the Imperas version is not properly setting the FIOM bit while Wally is.

```
# Info (IDV) testbench.idv_trace2log.process_event @ 1140: RET,0,29,80003020,"30a0e073 csrsi   menvcfg,1      ",,,,CSR30a(menvcfg)=e0000000000000f1 CSRb00(mcycle)=000000000000006d CSRb02(minstret)=000000000000001d,
# Info (IDV) Instruction executed prior to mismatch '0x80003020(main+20): 30a0e073 csrsi   menvcfg,1'
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x0000000080003020 main+20):
# Error (IDV) Mismatch 0> CSR 30a (menvcfg)
# Error (IDV)   . dut:0xe0000000000000f1
# Error (IDV)   . ref:0xe0000000000000f0 STCE:1 PBMTE:1 PME:0 CBZE:1 CBCFE:1 CBIE:3 FIOM:0
# Error (IDV) testbench.idv_trace2api.state_compare @ 1140: MISMATCH
```

[csrwrites.elf.txt](https://github.com/openhwgroup/cvw/files/13983841/csrwrites.elf.txt)
[vsim.log](https://github.com/openhwgroup/cvw/files/13983842/vsim.log)

davidharrishmc, 2024-01-19:
@eroom1966 , I can't find information about the menvcfg.FIOM bitfield in the OVP_Model_Specific_information manual.  Is there anything we need to do to enable setting that bit?  Presently, ImperasDV appears to be treating it as a read-only 0.

eroom1966, 2024-01-24:
there is an override affecting the write mask of this register
if you add the following to your imperas.ic file you will get the full listing of verrides, and there current value
> --showoverrides

I am running a slightly different platform to you, but mine shows
>     --override iss/cpu0/envcfg_mask=0 (Uns64) (default=0) (override) Specify hardware-enforced mask of writable bits in envcfg registers

If I add 
> --override iss/cpu0/envcfg_mask=1

this overcomes your issue and the instruction gives
> Info 29: 'iss/cpu0', 0x0000000080003020(main+20): Machine 30a0e073 csrsi   menvcfg,1
> Info 29: 'iss/cpu0',   menvcfg e0000000000000f0 -> e0000000000000f1

I am checking with my colleague if this is a legal thing to override, otherwise I am surprised it is not the default

eroom1966, 2024-01-24:
We are going to change the default writability of this bit

davidharrishmc, 2024-01-24:
Sounds good.  Thank you for the fix.

closed 2024-01-24 by davidharrishmc