openhwfoundation/cvw #593: ImperasDV decodes 04000043 as fmadd.h but there is no such instruction

davidharrishmc, 2024-01-24:
ImperasDV decodes instruction 04000043 as fmadd.h.  This is incorrect because Zfh does not define half-precision fmadd, fmsub, fnmsub, or fnmadd.  (This caught me by surprise too.)  ImperasDV should produce an illegal instruction exception on instructions of these forms.

This is presently causing a mismatch between ImperasDV and cvw because cvw correctly throws the exception in the fpu.S coverage test.

# Error (IDV)   . ref:04000043 fmadd.h f0,f0,f0,f0,rne

eroom1966, 2024-01-24:
@davidharrishmc - could you please attach the elf file you are running to this issue

davidharrishmc, 2024-01-24:
Test case attached.

[8.tar.gz](https://github.com/openhwgroup/cvw/files/14039606/8.tar.gz)

davidharrishmc, 2024-01-24:
Thanks!  My mistake.  I had thought they existed, then a riscv-arch-test developer pointed me to something saying these instructions do not exist.  It must have been an outdated spec that we were considering.

closed 2024-01-24 by davidharrishmc