openhwfoundation/cvw #595: ImperasDV does not throw instruction page fault on non-leaf page with PBMT != 0

davidharrishmc, 2024-01-24:
I'm getting a mismatch between CVW and ImperasDV when the program attempts to fetch an instruction from a page with a non-leaf PTE with PBMT != 0.  CVW throws an instruction page fault, while ImperasDV accepts the TLB entry.  I think there is either a bug in ImperasDV or CVW imperas.ic.

See attached 3.tar.gz for an example, with the interesting part listed below.

At 800170fe, the program attempts to jump to x5 = 0x80806000 with virtual memory enabled and menvcfg.PBMT = 0.  This SV48 virtual address is missing from the page table entry, so the page table walker fetches first fetches a root PTE of 0x00000000200044C1 from 0x8001000.  That points to the  0x80011010, which contains the next level PTE of 0x00000000200048C1.  This points to 0x80012018, which contains the next level (non-leaf) PTE of 0x4000000020004CC1. This entry has bit 62 = 1.  

According to the RISC-V Privileged spec at Chapter 6 (PBMT) page 93, 
"For non-leaf PTEs, bits 62–61 are reserved for future standard use. Until their use is defined by a standard extension, they must be cleared by software for forward compatibility, or else a page-fault exception is raised."

CVW throws the page-fault exception.  ImperasDV creates a TLB entry.   I believe this is incorrect behavior in ImperasDV, or a configuration setting we are missing.

```
 Info (IDV) testbench.idv_trace2log.process_event @ 20780: RET,0,643,800170fe,"000280e7 jalr    x1,x5          ",x1=0000000080017102,,,CSRb00(mcycle)=0000000000000819 CSRb02(minstret)=0000000000000283,
# Info (RISCV_TLB) CPU 'refRoot/cpu': CREATE TLB ENTRY:
# Info (RISCV_TLB) CPU 'refRoot/cpu': VA 0x80806000:0x80806fff PA 0x80000000:0x80000fff priv=[raw:rwx U:--- S:rwx] U=0 G=0 A=1 D=1 ASID=0
# Info (RISCV_PMP) CPU 'refRoot/cpu': PMP PRIV=rwx 0x00000000:0x3fffffffb (mode Supervisor)
# Info (IDV) testbench.idv_trace2log.process_event @ 20870: RET,0,644,80017230,"34021273 csrrw   x4,mscratch,x4 ",x4=0000000080016400,,,CSR300(mstatus)=0000000a00000880 CSR340(mscratch)=0000000000000000 CSR341(mepc)=0000000080806000 CSR342(mcause)=0000000000000002 CSR343(mtval)=000000002000004f CSRb00(mcycle)=0000000000000822 CSRb02(minstret)=0000000000000284,
# Info (IDV) Instruction executed prior to mismatch '0x800170fe(main+fe): 000280e7 jalr    x1,x5'
# Error (IDV) PC mismatch (HartId:0, PC:0x0000000080806000 _end+7eed90):
# Error (IDV) Mismatch 0>
# Error (IDV)   . dut:0x0000000080017230 ipf_handler+0
# Error (IDV)   . ref:0x0000000080806000 _end+7eed90
```

[3.tar.gz](https://github.com/openhwgroup/cvw/files/14034011/3.tar.gz)

eroom1966, 2024-01-24:
@davidharrishmc 
in the imperas.ic file you have specified
> --override cpu0/Svpbmt=T

In which case why would you expect this to fault ?

eroom1966, 2024-01-24:
I should add, in an earlier instruction you have set menvcfg.PBMTE=1, at instruction 23
that enables bits in the PTE for PBMT to be non-zero

davidharrishmc, 2024-01-24:
Agreed with all of this, but this is a non-leaf PTE:  0x4000000020004CC1.  The specification does not allow bits 62 or 61 to be set in a non-leaf PTE, even though these bits have meaning for a leaf PTE when menvcfg.PBMTE=1 and Svpbmt=T.

"For non-leaf PTEs, bits 62–61 are reserved for future standard use. Until their use is defined by a standard extension, they must be cleared by software for forward compatibility, or else a page-fault exception is raised." - Privileged Spec Chapter 6 page 93.

eroom1966, 2024-01-24:
Hi David,
This is what we are seeing
and the PBMT bits are only set in the leaf

```
Info 665: 'iss/cpu0', 0x00000000800170fe(main+fe): Supervisor 000280e7 jalr    ra,t0
Info 665: 'iss/cpu0',   MEMX 0x800170fe 0x800170fe 2 80e7
Info 665: 'iss/cpu0',   MEMX 0x80017100 0x80017100 2 0002
Info 665: 'iss/cpu0',   ra 00000000800170e0 -> 0000000080017102
Info 666: 'iss/cpu0', 0x0000000080806000(rvtest_entry_point): Supervisor 8082     c.jr    ra
Info 666: 'iss/cpu0',   MEMRM 0x80010000 0x80010000 8 00000000200044c1 L3 (iss/cpu0/64-bit Supervisor Physical unified)
Info 666: 'iss/cpu0',   MEMRM 0x80011010 0x80011010 8 00000000200048c1 L2 (iss/cpu0/64-bit Supervisor Physical unified)
Info 666: 'iss/cpu0',   MEMRM 0x80012020 0x80012020 8 00000000200054c1 L1 (iss/cpu0/64-bit Supervisor Physical unified)
Info 666: 'iss/cpu0',   MEMRM 0x80015030 0x80015030 8 20000000200000cf L0 (iss/cpu0/64-bit Supervisor Physical unified)

```.

This does not seem to agree with your earlier data ?