openhwfoundation/cvw #596: Why does ImperasDV give access fault on misaligned AMO access?

davidharrishmc, 2024-01-24:
I've found a situation where cvw throws an Store/AMO address misaligned exception and ImperasDV throws Store/AMO access fault.  This seems to be either a configuration problem or bug in ImperasDV.

The lsu.S test at 0x8000300c performs ampadu.w on address 80000001. This is a misaligned address, so I would expect a misaligned exception.

I believe imperas.ic is configured correctly to give this region AMO access.
--callcommand refRoot/cpu/setPMA -lo 0x0080000000 -hi 0x008FFFFFFF -attributes " rwxaA- 1248 " # UNCORE_RAM

reproduce with

cd ~/cvw/tests/coverage
make
run-elf.bash --elf lsu.elf

```
# Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x8000300c 00a522af amoadd.w x5,x10,(x10): No access permission for store/atomic memory operation (0x80000001)
# Info (IDV) Instruction executed prior to mismatch '0x80000050(trap_handler+0): 34021273 csrrw   x4,mscratch,x4'
# Error (IDV) CSR register value mismatch (HartId:0, PC:0x0000000080000050 trap_handler+0):
# Error (IDV) Mismatch 0> CSR 342 (mcause)
# Error (IDV)   . dut:0x0000000000000006 Interrupt:0 Code:6(Store/AMO address misaligned)
# Error (IDV)   . ref:0x0000000000000007 Interrupt:0 Code:7(Store/AMO access fault)
```

eroom1966, 2024-01-24:
Hi @davidharrishmc @ross144 
Do you know if the work to port wally to VCS is in place ?
in particular could I reproduce this issue running under VCS

Thx
Lee

eroom1966, 2024-01-24:
@davidharrishmc I may be able to reproduce this with our simulator standalone, could you attach the elf file, log etc please ?

davidharrishmc, 2024-01-24:
Attached

[22.tar.gz](https://github.com/openhwgroup/cvw/files/14040579/22.tar.gz)

eroom1966, 2024-01-24:
OK this is an awful spec definition, please take a look at 

> 3.6.3.3 Alignment
> Memory regions that support aligned LR/SC or aligned AMOs might also support misaligned
> LR/SC or misaligned AMOs for some addresses and access widths. If, for a given address and
> access width, a misaligned LR/SC or AMO generates an address-misaligned exception, **then all
> loads, stores, LRs/SCs, and AMOs using that address and access width must generate address 
> misaligned exceptions.**

So because you have set
> --override cpu0/unaligned=T

Which means a load/store would NOT generate a misaligned exception, then the AMO cannot generate a misaligned exception either, and therefore must generate an access fault

If you were to disable support for unaligned accesses
> --override cpu0/unaligned=F

Then the AMO access WOULD generate a misaligned exception

you are not the first to stumble into this

davidharrishmc, 2024-01-24:
Can you refresh me on what the little a and big A mean in the PMA line? Maybe that will help.

I’m having trouble parsing all this. This configuration supports Zicclsm:

Zicclsm Misaligned loads and stores to main memory regions with both the cacheability and coherence PMAs must be supported.

Note
This introduces a new extension name for this feature. This requires misaligned support for all regular load and store instructions (including scalar and vector) but not AMOs or other specialized forms of memory access. Even though mandated, misaligned loads and stores might execute extremely slowly. Standard software distributions should assume their existence only for correctness, not for performance.
https://github.com/riscv/riscv-profiles/blob/main/profiles.adoc

Consistent with this spec, our rv64gc config supports misaligned regular loads and stores, but not misaligned AMO.

What do we need to put in imperas.ic to describe this configuration?  Sounds like unaligned=T is too strong, and is telling the reference model to support AMO as well as load/store.

ross144, 2024-01-24:
Two things.
1. I got VCS working with our standard regression test about 1.5 weeks ago,
but I ran into a few issues.  First It does like undefined modules even if
those modules don't exist.  We have several of these for SRAMs we cannot
legally put into the repo.  Second I had issues setting top-level
parameters.  I followed the instructions to compile with
-pvalue+PARAM=VALUE and it does not seem to take effect.  I am able to hack
around the first issue, but the second one is blocking me from getting
ImperasDV running with VCS.
2. As David mentioned I'm confused by this spec you mentioned.  The Zicclsm
extension appears to conflict with 3.6.3.3.

Rose

eroom1966, 2024-01-25:
it looks like Zicclsm is not yet a ratified spec ?
Can you point me at a reference to the section which specifically conflicts with 3.6.33 ?

eroom1966, 2024-01-25:
Apologies, I see @davidharrishmc posted earlier in the thread

I don't see a contradiction here, unless I am missing something
This is not a question regarding support of misaligned Load/Store, this is a question about the type of exception raised when an AMO attempts to perform a misaligned access.

eroom1966, 2024-01-25:
OK, I think I understand what it is you are trying to configure, I am not sure it is legal, but I want to just put down the thought process for future reference.

# Assertion 1
Setting unaligned=T, means that regular loads and stores can perform misaligned accesses **without raising an** exception.

# Assertion 2
Extension Zam controls whether an AMO operation can be performed on a misaligned access **without raising an** exception

# Assertion 3
An LR/SC performed on a misaligned access will **always raise an exception**

In your current configuration you have unaligned=T and Zam is not defined
this means that misaligned AMO will still raise exceptions, so the question is, what exception should it raise ?
From the earlier section of 3.6.3.3, I believe this is as follows assuming Zam is not defined

# Assertion 4 (from section 3.6.3.3)
```
if (unaligned==T) then
  misaligned(AMO) raises EXCEPTION=ACCESS_FAULT
else
  misaligned(AMO) raises EXCEPTION=MISALIGNED
endif
```

If you enable Zam, then misaligned load stores on AMO will not raise exceptions

So as far as I can see, none of these setting combinations can satisfy your implementation.
In your implementation, it is as follows
- Regular Loads and Stores to misaligned addresses do NOT raise exceptions - this matches Assertion 1
- AMO to misaligned addresses raises MISALIGNED exception - this **violates** Assertion 4

This issue is, if you wish misaligned AMO to raise a MISALIGNED exception(not ACCESS_FAULT), then the regular loads and stores must ALSO raise a MISALIGNED exception
These two conditions are in contradiction with each other.

I cannot see how Zicclsm  affects this in any way
>  **Zicclsm Misaligned loads and stores to main memory regions with both the cacheability and coherence PMAs must be supported.**

> This introduces a new extension name for this feature. This requires misaligned support for all regular load and store instructions (including scalar and vector) but not AMOs or other specialized [...]

Happy to have a call to discuss this if you wish.

ross144, 2024-01-25:
Hi Lee,

Your explanation makes sense to me now. I was conflating misaligned and
access fault exceptions.  After rereading the last paragraph of section
3.6.3.3 I agree with your analysis.  For convenience I've copied the
paragraph below.

Rose

Implementations may raise access-fault exceptions instead of
address-misaligned exceptions for
some misaligned accesses, indicating the instruction should not be emulated
by a trap handler.
If, for a given address and access width, all misaligned LRs/SCs and AMOs
generate access-fault
exceptions, then regular misaligned loads and stores using the same address
and access width are
not required to execute atomically.

davidharrishmc, 2024-01-26:
Lee, thank you for your patience explaining this to us.  This was quite subtle, but I agree with what you and Rose are saying.

We need to change misaligned atomics to cause access-fault exceptions.

David

closed 2024-01-26 by davidharrishmc