openhwfoundation/cvw #610: RAS return after return bug

ross144, 2024-02-01:
RAS does not correctly handle consecutive returns.

davidharrishmc, 2024-09-12:
Performance bug, not logic bug.

davidharrishmc, 2026-11-01:
@rosethompson, I have a fix for this in draft PR #2041 and would appreciate your review.

The root cause is that the RAS pops in F, repairs in D, and pushes in E, but turned them into a single ±1 pointer step. When two updates landed in the same cycle (for example, a `ret` popping in F while a call pushes in E, or a class repair in D at the same time as a pop), one was lost. The push also wrote at the updated pointer, so a call in E overwrote the live entry its callee's `ret` had just popped. Separately, stacks whose size isn't a power of 2 wrapped from 0 to 0 instead of to `RAS_SIZE-1`.

The PR sums the updates (−2..+2), pushes at `Ptr + 1 + BPReturnD`, and forwards the link address of a call in E or D to a return fetched right behind it.

- Return mispredictions caused by the RAS go to zero on CoreMark (rv64gc: 11 → 0) and Embench (674 → 0), but total cycles change by only about ±0.001%, so this is not a performance win.
- The RAS gets slightly smaller in TSMC28, and the existing −64 ps path StallD → PopF → pointer → write decode goes away.
- `tests/coverage64gc/rasupdates.S` exercises each pair of simultaneous updates.

closed 2026-11-04 by rosethompson