openhwfoundation/cvw #635: riscof errors

solomatnikov, 2024-02-17:
```
    INFO | Selecting Tests.
    INFO | Running Tests on DUT.
   ERROR | make[2]: Entering directory '.../cvw/tests/riscof/riscof_work'
make[2]: Leaving directory '.../cvw/tests/riscof/riscof_work'
   ERROR | .../cvw/addins/riscv-arch-test/riscv-test-suite/rv32i_m/I/src/jalr-01.S: Assembler messages:
.../cvw/addins/riscv-arch-test/riscv-test-suite/rv32i_m/I/src/jalr-01.S:72: Error: illegal operands `la x0,5b'
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not open my.elf; searched paths:
	. (current directory)
	.../riscv/v20180629-492-g59ab58e/riscv64-unknown-elf/bin/ (based on configured --prefix and --with-target)
make[2]: *** [.../cvw/tests/riscof/riscof_work/Makefile.DUT-spike:1517: TARGET378] Error 255
    INFO | Running Tests on Reference Model.
   ERROR | make[2]: Entering directory '.../cvw/tests/riscof/riscof_work'
make[2]: Leaving directory '.../cvw/tests/riscof/riscof_work'
   ERROR | .../cvw/addins/riscv-arch-test/riscv-test-suite/rv32i_m/I/src/jalr-01.S: Assembler messages:
.../cvw/addins/riscv-arch-test/riscv-test-suite/rv32i_m/I/src/jalr-01.S:72: Error: illegal operands `la x0,5b'
riscv64-unknown-elf-objdump: 'ref.elf': No such file
make[2]: *** [.../cvw/tests/riscof/riscof_work/Makefile.Reference-sail_c_simulator:1517: TARGET378] Error 1
    INFO | Initiating signature checking.
   ERROR | Signature file : /work/sols/misc/cvw/tests/riscof/riscof_work/rv32i_m/I/src/jalr-01.S/dut/DUT-spike.signature does not exist
make[1]: *** [Makefile:28: arch32] Error 1
make[1]: Leaving directory '/work/sols/misc/cvw/tests/riscof'
make: *** [Makefile:52: riscoftests] Error 2
```

Karl-Han, 2024-02-17:
You may encounter the same problem as I have yesterday, and I learned that something is broken in the master branch. 

https://github.com/openhwgroup/cvw/releases/tag/CVW_v0.9 is supposed to be the stable branch that works. Therefore, if your goal is to get cvw running, I suggest you to checkout that branch and cherry-pick the branch https://github.com/openhwgroup/cvw/issues/520#issuecomment-1889909857 as you have done before. 

FYI: the test at my end has been running 4 RISCOF jobs from 16 hours so far, and I don't know how long it will take to run all of the tests.

P.S. make the following patch to run 4 jobs at the same time.
```diff
diff --git a/tests/riscof/config.ini b/tests/riscof/config.ini
index ae5a2f13b..d750f023d 100644
--- a/tests/riscof/config.ini
+++ b/tests/riscof/config.ini
@@ -9,6 +9,7 @@ pluginpath={0}/spike
 ispec={0}/spike/spike_rv{1}_isa.yaml
 pspec={0}/spike/spike_platform.yaml
 target_run=1
+jobs=4

 [sail_cSim]
 pluginpath={0}/sail_cSim
```

Karl-Han, 2024-02-17:
I am currently running at Branch: https://github.com/openhwgroup/cvw/tree/CVW_v0.9

In addition to the problems mentioned above, I think there may be some problems with the `spike` DUT, because a single test, TARGET0, in `cvw/tests/riscof/riscof_work` runs for more than 17 hours as shown below. 

```text
user    3095129 99.9  0.0  16532  6536 pts/9    R+   Feb16 1062:52 spike --isa=rv32imfdc_Zba_Zbb_Zbc_Zbs +signature=.../cvw/tests/riscof/riscof_work/rv32i_m/A/src/amoadd.w-01.S/dut/DUT-spike.signature +signature-granularity=4 my.elf
```

I temporarily skip the test and continue with `./regression-wally`.

ross144, 2024-02-17:
I think this specific bug is related to the latest version of gcc.  Try rolling back to version tag: 2023.12.20.

solomatnikov, 2024-02-18:
> I think this specific bug is related to the latest version of gcc. Try rolling back to version tag: 2023.12.20.

When I tried it I got:

```
    INFO | Env path set to .../cvw/addins/riscv-arch-test/riscv-test-suite/env
    INFO | Running Build for DUT
    INFO | Running Build for Reference
   ERROR | riscv_sim_RV32: executable not found. Please check environment setup.
make[1]: *** [Makefile:28: arch32] Error 1
make[1]: Leaving directory '.../cvw/tests/riscof'
make: *** [Makefile:52: riscoftests] Error 2
```

`spike` is installed:

```
which spike
.../riscv/v2023.12.20-g8c969a9/bin/spike
```

ross144, 2024-02-18:
riscv_sim_RV32 is sail_riscv.   Can you try pulling the latest sail_riscv commit and compiling with the following commands.
```
ARCH=64 make c_emulator/riscv_sim_RV64
ARCH=32 make c_emulator/riscv_sim_RV32
```

You'll need to make sure these are in the path somewhere.  The wally-tool-chain-install.sh does the following
```
sudo ln -sf $RISCV/sail-riscv/c_emulator/riscv_sim_RV64 /usr/bin/riscv_sim_RV64
sudo ln -sf $RISCV/sail-riscv/c_emulator/riscv_sim_RV32 /usr/bin/riscv_sim_RV32
```

solomatnikov, 2024-02-18:
Yes, I had to compile `riscv_sim_RV32/64` in `sail_riscv` and I got https://github.com/riscv/sail-riscv/issues/399 but it doesn't matter.

ross144, 2024-02-18:
To clarify are you still getting this error when running make?

 ERROR | riscv_sim_RV32: executable not found. Please check environment setup.

solomatnikov, 2024-02-19:
> To clarify are you still getting this error when running make?
> 
> ERROR | riscv_sim_RV32: executable not found. Please check environment setup.

`make` works now but it is annoying that one has to spend so much time b/c `riscof` doesn't work with the latest `v2024.02.02-g59ab58e` version of `riscv-gnu-toolchain`.

closed 2024-02-19 by solomatnikov

solomatnikov, 2024-02-19:
BTW, if you want many people to really use this design, you should document requirements and setup a lot better than it is now, i.e. no references to anything HMC specific, and full list of reqs explained in the docs instead of reqs buried in the HMC specific scripts, which nobody is going to run.

Using docker and specifying everything in `Dockerfile` is also a good idea. It's not 100% reliable but it helps one to set up the env.

Also, using a standard Linux distro with long term maintenance like RHEL 9 is a good idea.

davidharrishmc, 2024-02-20:
Alex,

Synopsys doesn’t list RHEL9 as supported until the 09.24 release.  We’ve had a problem moving from Ubuntu 2204LTS with glibc 2.31 to Ubuntu 2204LTS with glibc 2.34 because the new library isn’t compatible with Synopsys Design Compiler either; DC depends on some private symbols in glibc 2.31.

We’re about to migrate to RHEL8.9, though I would have preferred 9.  Are you able to run Synopsys tools on RHEL9?

David

solomatnikov, 2024-02-24:
I haven't run Design Compiler on RHEL 9, surprised Synopsys didn't support it so far.