openhwfoundation/cvw #647: Remove Hardcoded Paths

ross144, 2024-03-05:
There are still several locations in the scripts with hardcoded paths specific to either HMC or OSU servers.  I'm going to use this issue to track these locations until they are all found.

Synthesis:
```
synthDC/.synopsys_dc.setup
    set pdk /proj/models/tsmc28/libraries/28nmtsmc/tcbn28hpcplusbwp30p140_190a/                                                                                                                                   
    set osupdk /import/yukari1/pdk/TSMC/28/CMOS/HPC+/stclib/9-track/tcbn28hpcplusbwp30p140-set/

    set memory /home/jstine/WallyMem/rv64gc/                                                                                                                                                                      
    set osumemory /import/yukari1/pdk/TSMC/WallyMem/rv64gc/   
```

davidharrishmc, 2024-06-13:
@stineje  create site-specific prefix in the site-setup.sh.  Move files around so the prefix is common to everything on a site.

jordancarlin, 2024-09-18:
Additional hardcoded paths that I found in the repo. Some of these can likely just be removed instead of updating them to use generic versions sourced from setup files.
  - bin/fparchtest.sh
      - [ ] `RISCVCTG=/home/harris/repos/riscv-ctg`
  - fpga/generator/wave_config.wcfg
      - [ ] `<db_ref path="/home/ross/repos/active/cvw/fpga/generator/WallyFPGA.hw/hw_1/wave/hw_ila_data_1/hw_ila_data_1.wdb"`
  - linux/buildroot-packages/fpga-axi-sdc/fpga-axi-sdc.mk~
      - [ ] `FPGA_AXI_SDC_SITE = /home/jpease/repos/fpga-axi-sdc`

stineje, 2024-09-18:
Thanks @jordancarlin - fixed some of them but not sure about wave_config.wcfg - asking @ross144 - seems Vivado generated

ross144, 2024-09-18:
This is a temp file and should be removed. linux/buildroot-packages/fpga-axi-sdc/fpga-axi-sdc.mk~

ross144, 2024-09-18:
fpga/generator/wave_config.wcfg is very out of date.  It is used for arranging signals in the ILA waveform viewer.  It was generated by vivado and I can't see why it would need to include that hard coded path.  It is ignored when used in other directories. I think we can ignore this one.

jordancarlin, 2024-09-18:
Closed with PRs #956 and #954

closed 2024-09-18 by jordancarlin